luxonis/depthai #5: Video pause in between every time while running test.py

kapoorshivam100, 2020-01-17:
[image]

here is the image that shows while running test.py , the video pause in between and the code is showing error like

Device get data failed: 7
E: [         0] [python] XLinkReadDataWithTimeOut:1468	/home/felistrs/projects/libs/intel_mdk/R9.8/mdk/common/components/XLink/shared/XLink.c:1468
	 Assertion Failed: link != NULL

I am using Raspbian buster and Pi4 for the same

Luxonis-Brandon, 2020-01-21:
Sorry about the delay on this.  Which board is being used again?  Would you mind posting a picture of the unit as used so we can be sure.

kapoorshivam100, 2020-01-22:
In the Latest version 0.07a (Thanks For that) , i am able to get the depth perception and now wants to calibrate the camera to get better results.
Also if we want to obtain distance data from it (That we further need to feed to the ardupilot for rover/drone for vision navigation) ,please do suggest a pipeline for that .

Luxonis-Brandon, 2020-01-28:
Hrm... this comment is showing up in email, but not on Github.  Maybe was
it deleted on Github?  So are you trying to get the distance map overall?
Or distance to a specific object.

Also, sorry about the delay - I had the flu last week and was down all
week.  Just starting to catch back up now.

itsderek23, 2020-02-24:
Hi @kapoorshivam100 - we've released a number of updates to DepthAI since this issue was created. Mind [following our upgrade steps](https://docs.luxonis.com/api/#upgrade) and re-trying?

If this fails to help, mind sharing the log output and your board version (or a photo of the board if unknown)?

kapoorshivam100, 2020-02-28:
[image]

If we can get a better depth analysis ,it would be great

closed 2020-02-28 by kapoorshivam100