luxonis/depthai #9: default.calib file in version control

itsderek23, 2020-02-04:
The `resources/default.calib` is currently under version control. I believe this is problematic as it should be overwritten after performing calibration and it's easy to push this board-specific `calib` file on mistake.

Possible alternative:

* Change `default.calib` to `default.calib.example`. Keep this in version control.
* Ignore all files w/the `calib` extension
* When running our example scripts, copy `default.calib.example` => `default.calib` if it doesn't exist.

/cc @Luxonis-Brandon @Luxonis-Brian

Luxonis-Brian, 2020-02-04:
I agree this is a potential problem. Your plan sounds good, though!

itsderek23, 2020-02-05:
@Luxonis-Brian implemented an alternative in c66462d40ad64345d4bcf3d7cb56d05e990bc0ed that attempts to load the `depthai.calib` first, then drops down to the default. 

`depthai.calib` is already ignored by version control so I think this solves the root problem.

closed 2020-02-05 by itsderek23