luxonis/depthai #25: Crash when moving out of frame

itsderek23, 2020-02-18:
@Luxonis-Brandon reported that the [reduce_rgb_latency](https://github.com/luxonis/depthai-python-extras/compare/reduce_rgb_latency) branch crashed when he moved out of the frame on the BW1097 when running `python3 test.py`.

Host exception:

```py
Traceback (most recent call last):
  File "test.py", line 120, in <module>
    y1 = int(e[0]['top'] * img_h)
ValueError: cannot convert float NaN to integer
```

My guess is `e[0]['top']` is NaN. This didn't happen every time he left the frame.

itsderek23, 2020-02-18:
I was not able to replicate this. Idea:

Wrap the coordinate math inside a `try` block and and `continue` on `ValueError` inside an `except` block. Print info on the error inside the `except` block. This will ensure the script doesn't crash if we receive a `NaN` value.

I'm thinking a `try` block versus checking if every number is not `NaN` as most of the time these numbers look fine and doing the check every time adds overhead. Instead, we just handle the few times where things are correct.

/cc @Luxonis-Brandon

Luxonis-Brandon, 2020-02-18:
Yes, that sounds like an ideal solution to me.  Thanks!

itsderek23, 2020-02-24:
Closing as it might have ran via old code.

closed 2020-02-24 by itsderek23