luxonis/depthai #26: Crash w/invalid label index

itsderek23, 2020-02-18:
See:

```
Traceback (most recent call last):  File "test.py", line 132, in <module>    cv2.putText(frame, labels[int(e[0]['label'])], pt_t1, cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 0, 255), 2)IndexError: list index out of range
```

System info:

```
./log_system_information.sh 
Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
depthai.__version__ == 0.0.9a
depthai.__dev_version__ == c1082d5a67f724398b0617485076f0e34da1054f
numpy.__version__ == 1.16.2
Python 3.7.3
```

itsderek23, 2020-02-18:
I was not able to replicate this. I'm a bit at a loss how an invalid index could be passed unless `e[0]['label'] < 0` as we check for a valid label index via:

```py
if e[0]['label'] > len(labels):
  print("Label index=",e[0]['label'], "is out of range. Not applying text to rectangle.")
```

Luxonis-Brandon, 2020-02-18:
I'm guessing I must have just been running code that was prior to this fix.  When I re-ran after a git pull I didn't see this again either.

itsderek23, 2020-02-24:
Closing as it might have ran via old code.

closed 2020-02-24 by itsderek23