luxonis/depthai #28: Add colorized depth map

itsderek23, 2020-02-21:
The current depth map is grayscale, but we can support a better-looking colorized depth map based on https://github.com/luxonis/depthai-python-extras/commit/0fd2cc61c6be1b75ad5eed07424a2ad509f48074.

Add this back to the `test.py` script.

/cc @GergelySzabolcs

itsderek23, 2020-02-21:
Ensure frame rate for doing colorization on host is acceptable as well.

itsderek23, 2020-02-25:
Believed this was reverted in 6109ed87948bdaeefacfdf8221c7ce483304054f

chris-piekarski, 2020-02-27:
BTW @itsderek23, I checked out the commit you reference and it changed from black & white to black & red. Regardless of the objects in the frame, I couldn't get anything but red.

itsderek23, 2020-02-27:
Thanks @chris-piekarski - we're working on this in #37.

szabi-luxonis, 2020-05-04:
This was added on master a while ago.
python3 test.py -s depth_sipp
or
colorized disparity
python3 test.py -s depth_color_h

closed 2020-05-04 by szabi-luxonis