luxonis/depthai #29: Add full frame color stream option

chris-piekarski, 2020-02-21:
Linux os can't see the cameras so I'll need to get full-color 720/1080/2160p frames from the api (like left right streams).

itsderek23, 2020-02-24:
@chris-piekarski - Thanks for the issue. 

Want to clarify: are you able to see the color output at all? Or requesting the larger original pixel size because the current preview display is too small?

chris-piekarski, 2020-02-27:
Yes, I can see the preview color output @itsderek23. For my use case, I need to show a larger original pixel size to the user.

itsderek23, 2020-02-27:
Thanks @chris-piekarski. 

For a quick hack, would scaling the 300x300 size be workable or do you need the higher-res output?

chris-piekarski, 2020-02-27:
Yeah, that is basically what we will be doing until we can get higher resolution frames. We're sending the preview stream to a remote 1080p viewer and I'd like to save 2160p video locally at the same time. At the moment I don't actually _need_ to run inference on the color stream so even just direct access to the camera would work for POC (but would like to enable inference detection in v2).

Luxonis-Brandon, 2020-05-07:
So we now have h.264 color video output, see [here](https://docs.luxonis.com/faq/#how-do-i-record-video-with-depthai).

So @chris-piekarski is this sufficient for the use-case you were intending?

closed 2020-06-23 by Luxonis-Brandon

Luxonis-Brandon, 2021-01-23:
FWIW also see here:
https://github.com/luxonis/depthai-python/pull/94
https://www.youtube.com/watch?v=Ww-vLSxOdW8&feature=youtu.be