luxonis/depthai #31: Example scripts have blurry video output

itsderek23, 2020-02-24:
The video output for emotion / landmarks is very blurry and the text overlay is large compared to the video output size:

[image]

From @yuriilaba-luxonis via Slack:

> The image in previewout has the same size as input tensor for neural network (in case of emotion recognition it is 64x64, for landmark recognition it is 48x48).
We use it, because it's cheap (resources) to get it.
So we just rescale previewout to 300x300, that's why image is the blurry. We can reduce text size.

itsderek23, 2020-02-26:
@yuriilaba-luxonis - do you have any ideas on how to adjust the input image so we can provide decent-looking output for a reasonable range of image sizes?

itsderek23, 2020-02-27:
From a Slack convo w/Yurii:

> The one way is to add another stream which will output the video which has same size as input video.
> And apply NN results on this video.
> There isn't any fast way to improve video, at least I don't see it.

Luxonis-Brandon, 2020-05-04:
Just as an update on this, this network is intended to be used after a face detector.  

We will be starting soon an a pipeline builder which allows one to build series and parallel pipelines of multiple neural networks.

And with this one will be able to select the face detector of interest, and then run networks like this emotion detector on the outputs of the face detector (and then say show emotion on the detected face bounding box).

And similarly than this could be used w/ other networks which are intended to be run on detected faces, like facial landmarks, face pose estimation, etc.

Luxonis-Brandon, 2020-07-07:
This is fixed in https://github.com/luxonis/depthai/pull/94#pullrequestreview-444155280 and will be merging to master soon.

closed 2020-07-07 by Luxonis-Brandon