luxonis/depthai #44: Hang/Stall if No Color Camera Present

Luxonis-Brandon, 2020-03-09:
When debugging https://github.com/luxonis/depthai-python-extras/issues/40 @PTS93 and I circled to the hypothesis that actually the lack of a color camera in his setup (see [here](https://github.com/luxonis/depthai-python-extras/issues/40#issuecomment-596225752)) is actually what was causing his setup to hang even when used with USB2.

(When using USB3, the USB3-with-Pi hangup noted in that issue was the problem.)

So to confirm that the hang/stall does happen when no color camera is connected, I temporarily disconnected the camera on a BW1097 (which is only USB2) and confirmed that this same hang does occur.  

To do this I just popped the MIPI connector for the color camera off the back of the board, as below:
[image]

After this I booted the BW1097 (RPi Compute Module Edition) and rand the calibration script.

And here's a video of the lockup/hangup:

[https://photos.app.goo.gl/qoRNxocqBPUsvxHi9](https://photos.app.goo.gl/qoRNxocqBPUsvxHi9)

@PTS93 - is this what you are seeing?

Thanks,
Brandon

PTS93, 2020-03-09:
Yes! thats the exact same behaviour I'm getting

Luxonis-Brandon, 2020-03-10:
Excellent, thanks for confirming @PTS93 !  

We think we have a handle on how to fix this and should be able to attack soon.  We were discussing/debugging today.  Should have an update on this later this week.

Luxonis-Brandon, 2020-03-10:
So we do now have an initial prototype version that lets the calibration and stereo depth without the color camera being present.

Our plan is to figure out how to integrate this properly, which means probably releasing such functionality in the coming weeks.

But if you @PTS93 or anyone else wants an initial 'hack' version to play with we can build a release of that with this functionality.

Since you're wanting to do object localization as well @PTS93 (and we're sending you the color camera today), I'm assuming just using the color camera to prevent this bug will be good enough.  But let me know if that's not the case.

Thoughts?

Thanks,
Brandon

PTS93, 2020-03-10:
Awesome! Though no hurry from my side, as I need the color camera either way.

Luxonis-Brandon, 2020-03-10:
Perfect, thanks!

Luxonis-Brandon, 2020-07-07:
This is implemented in https://github.com/luxonis/depthai/pull/94 which will be merged to master soon.

closed 2020-07-07 by Luxonis-Brandon