luxonis/depthai #52: Calibration.py writes to same image directories without notifying user

Luxonis-Brandon, 2020-04-02:
When running Calibration.py back-to-back from one board to the next, the default script writes the images to the same directory - so you end up with images from different boards - and therefore the calibration fails.

TDTRT We should at the least warn the user of this, and ideally warn the user and make a new folder path for the other board, and work out of that.

Luxonis-Brandon, 2020-04-15:
In some cases folks will want to use images from previous calibration attempts (say if they're using the same board, but just want more images for a better cal) and in others they will not (say if they're now calibrating a different board on the same system).  

So to allow this I'm thinking that the calibration process should automatically detect if an images folder already exists, if it does, it prompts user if they want to use that or make a new one, and then go from there.

So a command line question at the beginning of the calibration if an images folder already exists: 
`Do you want to use images from previous calibration runs?`

Talked to @squeakus offline and he's going to implement this flow.

Luxonis-Brian, 2020-04-16:
Great! Thank you @squeakus! So great to have your input on this. It's actually given rise to a number of deep dives in which we've found some ways to improve things. We've been able to dig through code to find some knobs to improve calibration across physical board variation, improvements in shutter sync between the two cameras, and also metadata output to allow proper acquisition of L/R stereo fame pairs during the calibration procedure. This is all a bit "behind the scenes", but I plan to roll all this up into a nice user experience that is also providing accurate results.

Luxonis-Brandon, 2020-09-11:
This is now resolved with capability to over-write or not in calibrate.py.

closed 2020-09-11 by Luxonis-Brandon