luxonis/depthai #63: Mirror the mono camera videos during calibration

Luxonis-Brandon, 2020-04-16:
The 'why' of this is if the video is not mirrored, you're instincts are to move the calibration checkerboard in the wrong direction.  It's like riding a bike with the handlebars reversed (it's really hard, see [here](https://www.youtube.com/watch?v=MFzDaBzBlL0)).

So this is why w/ FaceTime and other tools that show yourself, they mirror the video.  It's way more intuitive (in the FaceTime, they mirror it when they show it back to you, but then don't mirror it going out to the other party).

So we should do the same thing, mirror on display only, and pass the original frames-is to the calibration routine.

So I think the host (even the Pi) should be fast enough to do the flipping, but if not we can probably flip the streams in the Myriad X easily.

closed 2020-04-17 by VanDavv

Luxonis-Brandon, 2020-04-17:
So @VanDavv I'm thinking we should make inverting on the horizontal axis default.  

So there are two ways to calibrate:
1. The board is stationary, the calibration target moves.
2. The calibration target is stationary, the board moves.

For both 1 and 2 having the imaged flipped horizontally is desirable.  For '2' flipping vertically is also desirable.

But since flipping horizontally is beneficial for both I'm thinking we have that as the default.

Thoughts?

Thanks,
Brandon

VanDavv, 2020-04-17:
On 1097 it works well with the default non-flipped image, and my gut feeling tells me to keep flags defaulting to false (wanted to refactor the `--no-swap-lr` to `--swap-lr`). But anyway, it doesn't really matter to me, can make them as default. We need to decide which model is "base" one

Luxonis-Brandon, 2020-04-17:
Got it.  Maybe I should try again now that it's not latent.  Yes I'd say ignore my comment for now until I get a chance to play with it more.

Thanks for the quick response @VanDavv !

PR #71 (2020-04-17): Add ability to invert the horizontal and vertical axis
Resolves #63 

Adds `-iv` and `-ih` (with long-names `--invert-vertical` and `--invert-horizontal` respectively) that flips the displayed image - for calibration, still the "un-flipped" frame is passed