luxonis/depthai #64: Check if calibration is being run on Pi, Set Framerate to 10FPS

Luxonis-Brandon, 2020-04-16:
The `why` of this is to reduce the latency on the Pi.  Since the Pi is doing some work in the background on calibration (reducing the stream resolution from 2x 1280x720 so you can see both easily on a 1920x1080 display), the Pi struggles to keep up and so the streams getting latent with high jitter in the latency.  This is especially painful when the streams aren't mirrored (see enhancement https://github.com/luxonis/depthai/issues/63), as you move the checkerboard in the wrong direction - and it takes a while for this to update on the screen - so by the time you realize you've moved the boards WAY in the wrong direction.

On faster hosts, they can easily keep up so this isn't necessary, but on the Pi it does start to struggle w/ 2x 1280x720 video feeds at 30FPS each and resizing them smaller for display fitting purposes.

So what would be great is to detect if calibration is being run on a Pi and automatically reduce the framerate to 10FPS.

To do this, it looks like (from [here](https://raspberrypi.stackexchange.com/questions/5100/detect-that-a-python-program-is-running-on-the-pi)) that you can just use `import os` and `os.uname()` and then pull out the nodename result and check if it is `raspberrypi`.

Then we could use the following specify the framerate of the streams coming from DepthAI (but change it for `left` and `right`:
https://github.com/luxonis/depthai/blob/91907b6854081e0e2998f5add74941c41209f1b3/depthai.py#L126

VanDavv, 2020-04-17:
`os.uname().nodename` is not a good discriminator, take a look at the example below

`pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.uname()
posix.uname_result(sysname='Linux', nodename='raspberrypi', release='4.19.97-v7+', version='#1294 SMP Thu Jan 30 13:15:58 GMT 2020', machine='armv7l')
pi@raspberrypi:~ $ hostname
raspberrypi
pi@raspberrypi:~ $ sudo hostname test
pi@raspberrypi:~ $ hostname
test
pi@raspberrypi:~ $ python3
Python 3.7.3 (default, Dec 20 2019, 18:57:59) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import os
>>> os.uname()
posix.uname_result(sysname='Linux', nodename='test', release='4.19.97-v7+', version='#1294 SMP Thu Jan 30 13:15:58 GMT 2020', machine='armv7l')`

I'll use `os.uname().machine` and fit for all ARM processors

Luxonis-Brandon, 2020-04-17:
Thanks for catching that @VanDavv and going with a better solution!  Much appreciated.  I just tested your PR and it works great!

PR #70 (2020-04-17): limit FPS on embedded platforms
This PR resolves #64 

I assume that machine specs starting with `arm` and `aarch64` are embedded systems, as listed [here](https://stackoverflow.com/questions/45125516/possible-values-for-uname-m)
closed 2020-04-17 by VanDavv