luxonis/depthai #78: Calibration Fails on Mac OS X with Segmentation Fault

Luxonis-Brandon, 2020-04-20:
When running calibration on Mac OS X it seems to SegFault on or around saving the 13th image.  

Tested on https://github.com/luxonis/depthai/pull/66/ and also Master (49aad6662d18850a0544a209eba9fcf3ab624da3).

```
git submodule update --init
./depthai-api/build_py_module.sh
grep: /proc/meminfo: No such file or directory
MemAvailable:  MB
Unable to get MemAvailable
[BUILD depthai-api/host/py_module/build] make -j4
[ 14%] Built target nlohmann_json_schema_validator
[100%] Built target depthai
[BUILD depthai-api/host/py_module/build] cp depthai.cpython-37m-darwin.so ../../../../

python3 calibrate.py -ih
No calibration file. Using Calibration Defaults.
Using Arguments= {'count': 1, 'square_size_cm': 2.5, 'image_op': 'modify', 'mode': ['capture', 'process'], 'config_overwrite': None, 'field_of_view': 71.86, 'baseline': 9.0, 'swap_lr': True, 'dev_debug': None, 'invert_v': False, 'invert_h': True}
Starting image capture. Press the [ESC] key to abort.
Will take 13 total images, 1 per each polygon.
XLink initialized.
Sending device firmware "cmd_file": /Users/leeroy/depthai-python-extras/depthai.cmd
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
watchdog started 1000
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
WARNING: Version (dev) does not match (unknown & 62b4db70b327f51f6066af9620187ee4ae3878b8)
CNN configurations read: /Users/leeroy/depthai-python-extras/resources/nn/object_detection_4shave/object_detection.json
depthai: Calibration file is not specified, will use default setting;
config_h2d json:
{"_board":{"_homography_right_to_left":[0.988068163394928,0.0029474012553691864,5.067617416381836,-0.008765067905187607,0.9921473264694214,-8.795275688171387,-8.449587767245248e-06,-3.603489403758431e-06,1.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"left"},{"name":"right"},{"name":"meta_d2h"}]},"ai":{"calc_dist_to_bb":false},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.0,"left_to_right_distance_m":0.09000000357627869,"override-eeprom":false,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"depth":{"padding_factor":0.30000001192092896}}
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1000 bytes to config_h2d
!!! XLink write successful: config_h2d (1000)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Read: 14514560
Attempting to open stream inBlob
Successfully opened stream inBlob with ID #1!
Writing 14514560 bytes to inBlob
!!! XLink write successful: inBlob (14514560)
Closing stream inBlob: ...
Closing stream inBlob: DONE.
depthai: done sending Blob file /Users/leeroy/depthai-python-extras/resources/nn/object_detection_4shave/mobilenet_ssd.blob
Attempting to open stream outBlob
Successfully opened stream outBlob with ID #0!
Closing stream outBlob: ...
Closing stream outBlob: DONE.
CNN input width: 300
CNN input height: 300
CNN input num channels: 3
Host stream start:meta_d2h
Opening stream for read: meta_d2h
Attempting to open stream meta_d2h
Successfully opened stream meta_d2h with ID #1!
Starting thread for stream: meta_d2h
Host stream start:left
Opening stream for read: left
Attempting to open stream left
Started thread for stream: meta_d2h
Successfully opened stream left with ID #2!
Starting thread for stream: left
Started thread for stream: left
Host stream start:right
Opening stream for read: right
Attempting to open stream right
Successfully opened stream right with ID #3!
Starting thread for stream: right
depthai: INIT OK!
Started thread for stream: right
py: Saved image as: left_p0_0.png
py: Capture failed, unable to find chessboard! Fix position and press spacebar again
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full right:
Data queue is full left:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
py: Saved image as: left_p0_0.png
py: Saved image as: right_p0_0.png
py: Saved image as: left_p1_1.png
py: Capture failed, unable to find chessboard! Fix position and press spacebar again
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
Data queue is full left:
Data queue is full right:
py: Saved image as: left_p1_1.png
py: Saved image as: right_p1_1.png
py: Saved image as: right_p2_2.png
py: Saved image as: left_p2_2.png
py: Saved image as: left_p3_3.png
py: Saved image as: right_p3_3.png
py: Saved image as: left_p4_4.png
py: Saved image as: right_p4_4.png
py: Saved image as: left_p5_5.png
py: Saved image as: right_p5_5.png
py: Saved image as: left_p6_6.png
py: Saved image as: right_p6_6.png
py: Saved image as: right_p7_7.png
py: Saved image as: left_p7_7.png
py: Saved image as: left_p8_8.png
py: Saved image as: right_p8_8.png
py: Saved image as: left_p9_9.png
py: Saved image as: right_p9_9.png
py: Saved image as: left_p10_10.png
py: Saved image as: right_p10_10.png
py: Saved image as: left_p11_11.png
py: Saved image as: right_p11_11.png
py: Saved image as: right_p12_12.png
py: Saved image as: left_p12_12.png
Segmentation fault: 11
```

Luxonis-Brandon, 2020-04-20:
I think I had an old branch of the depthai-api submodule checked out, and that was the actual issue.  Will circle back after experimenting.

Luxonis-Brandon, 2020-04-20:
Nope... just checked out everything with the proper respective branches and it still behaves the same.  So this seems to be a bug with calibration on Mac OS X.  

[image]

```
brew --version
Homebrew 2.2.13
Homebrew/homebrew-core (git revision 8a418; last commit 2020-04-15)
Homebrew/homebrew-cask (git revision 9f589; last commit 2020-04-15)
```

```
./log_system_information.sh 
Darwin DESKTOP-OGSI4CI.localdomain 19.4.0 Darwin Kernel Version 19.4.0: Wed Mar  4 22:28:40 PST 2020; root:xnu-6153.101.6~15/RELEASE_X86_64 x86_64
depthai.__version__ == 0.0.10a
depthai.__dev_version__ == unknown
numpy.__version__ == 1.16.2
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: thread::join failed: Invalid argument
Python 3.7.7
```

Luxonis-Brandon, 2020-04-20:
And can confirm that both versions do not have a problem on the BW1097.

Luxonis-Brandon, 2020-04-20:
[image]

[image]

after I comment the del pipeline, no more segfault

Luxonis-Brandon, 2020-04-27:
So an interesting datapoint here... simply running `python3 calibrate.py -h` results in:
```
libc++abi.dylib: terminating with uncaught exception of type std::__1::system_error: thread::join failed: Invalid argument
Abort trap: 6
```

szabi-luxonis, 2020-07-08:
Was fixed in #86

closed 2020-07-08 by szabi-luxonis