luxonis/depthai #87: Reduce significant figures in X, Y, Z outputs

Luxonis-Brian, 2020-04-29:
The precision isn't real, so we should eliminate it. I think it will also look and feel like it's less noisy as the user won't see the numbers changing as much, even though the apparent precision is lower. 

This could be set as a function of distance from the camera that follows the rise in stereo error at increased Z, or it could be set to a fixed value, like 10cm. [Here's an example](https://files.readme.io/09c93c5-b7.png) of RMS error increasing with distance from some [Intel Real Sense devices](https://dev.intelrealsense.com/docs/tuning-depth-cameras-for-best-performance). 

Suggestion:

* <2m use cm precision
* 2m to 5m use dm precision
* >5m use half-meter precision

Luxonis-Brandon, 2020-05-06:
This is partially addressed in https://github.com/luxonis/depthai/pull/95

Luxonis-Brandon, 2020-05-07:
It's just fixed decimal for now... but closing the issue.  For other applications where more precision is required, it's simple enough to just change the print statement https://github.com/luxonis/depthai/blob/0611eb701fffbcd743644e4f0d2c7b92e60b7cd9/depthai.py#L198 (I think).

closed 2020-05-07 by Luxonis-Brandon