luxonis/depthai #89: Creating a pipeline with a blob file config that doesn't exist results in obscure error

ConnorChristie, 2020-05-01:
I switched to a new detection network and upon doing so made a typo in the `blob_file_config` which presented me with a very obscure error seen below. It would be helpful if an error was thrown if the blob or blob config file does not exist.

```
Using Custom Calibration File: depthai.calib
XLink initialized.
Sending device firmware "cmd_file": /home/pi/depthai/depthai.cmd
Successfully connected to device.
Loading config file
watchdog started Attempting to open stream config_d2h
3000
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
WARNING: Version (dev) does not match (c722ebde932d6627463321816a5654b5be6069e1 & 30295b558351cd030408e12a220cdd55b5fb450e)
EEPROM data: invalid / unprogrammed
watchdog triggered
Stopping threads: ...
Stopping threads: DONE 0.000s.
Closing all observer streams: ...
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    215801] [Scheduler00Thr] dispatcherEventSend:53        Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    215801] [Scheduler00Thr] sendEvents:1036        Event sending failed
Reseting: DONE.
XLink initialized.
Sending device firmware "cmd_file": /home/pi/depthai/depthai.cmd
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
WARNING: Version (dev) does not match (c722ebde932d6627463321816a5654b5be6069e1 & 30295b558351cd030408e12a220cdd55b5fb450e)
EEPROM data: invalid / unprogrammed
terminate called after throwing an instance of 'nlohmann::detail::parse_error'
  what():  [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
Aborted
```

Example code:

```python
depthai.init_device(consts.resource_paths.device_cmd_fpath)

config = {
    'streams': ['metaout', 'previewout'],
    'depth':
    {
        'calibration_file': consts.resource_paths.custom_calib_fpath,
        'padding_factor': 0.3
    },
    'ai':
    {
        'blob_file': "networks/person-2/face-detection-adas-0001.blob",
        # Notice the typo here with networks-2 instead of person-2
        'blob_file_config': "networks-2/person/face-detection-adas-0001_depth.json",
        'calc_dist_to_bb': True
    }
}

pipeline = depthai.create_pipeline(config)
```

Luxonis-Brandon, 2020-05-01:
Thanks @ConnorChristie for taking the time to write this up!  

Good point, the error reporting here should be better (it's not clear at all now) so that it's debuggable what's actually going on.

Will circle w/ the team and figure out what to do such that the code does the right thing (DTRT) for such a situation.

Thanks again,
Brandon

Luxonis-Brandon, 2020-05-06:
Hi @ConnorChristie ,

I accidentally discovered that this appears to be fixed in Master now.  See here:
https://photos.app.goo.gl/hzu7LYo2AxTpcBYEA

[image]

When testing out a social-distancing app we built off of DepthAI I accidentally deleted that model, and then saw a nice proper error message.

I don't remember the PR that snuck in on (I think I missed it), so I'm not 100% sure this will fix your scenario as well.

Thoughts?

ConnorChristie, 2020-05-07:
@Luxonis-Brandon The error I was seeing was when the json config file is missing. With the latest code, it still errors out but now after a couple of seconds instead of straight away. However, it does give you a log entry that says it can't find the file although I would've expected it to exit right after the log error instead of trying to continue initializing.

```
XLink initialized.
Sending device firmware "cmd_file": /home/pi/depthai/depthai.cmd
Successfully connected to device.
Loading config file
watchdog started Attempting to open stream config_d2h
3000
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
WARNING: Version (dev) does not match (30295b558351cd030408e12a220cdd55b5fb450e & c4c87dfdf39f619cacf74df94b98a20808e2c995)
EEPROM data: invalid / unprogrammed
Error opening json file: /home/pi/drone-depthai/networks/person-2/face-detection-adas-0001_depth.jsons
There is no cnn configuration file or error in it's parsing: /home/pi/drone-depthai/networks/person-2/face-detection-adas-0001_depth.jsons
config_h2d json:
{"_board":{"_homography_right_to_left":[1.0012089014053345,0.006868398282676935,-5.654909610748291,-0.003709208220243454,1.0076549053192139,-8.794636726379395,-4.944150077790255e-06,8.152581358444877e-06,1.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"}]},"ai":{"calc_dist_to_bb":true,"keep_aspect_ratio":true},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.03750000149011612,"left_to_right_distance_m":0.07500000298023224,"name":"BW1098OBC","override-eeprom":false,"revision":"R0M0E0","rgb_fov_deg":68.7938003540039,"stereo_center_crop":false,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"depth":{"depth_limit_mm":10000,"padding_factor":0.30000001192092896}}
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1000 bytes to config_h2d
!!! XLink write successful: config_h2d (1000)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
Successfully opened stream host_capture with ID #0!
Creating observer stream host_capture: DONE.
Read: 4244352
Attempting to open stream inBlob
Successfully opened stream inBlob with ID #1!
Writing 4244352 bytes to inBlob
!!! XLink write successful: inBlob (4244352)
Closing stream inBlob: ...
Closing stream inBlob: DONE.
depthai: done sending Blob file /home/pi/drone-depthai/networks/person-2/face-detection-adas-0001.blob
Attempting to open stream outBlob
Successfully opened stream outBlob with ID #2!
Closing stream outBlob: ...
Closing stream outBlob: DONE.
CNN input width: 672
CNN input height: 384
CNN input num channels: 3
CNN to depth bounding-box mapping: start(77, 78), max_size(1126, 643)
Host stream start:metaout
Opening stream for read: metaout
Attempting to open stream metaout
Successfully opened stream metaout with ID #1!
Starting thread for stream: metaout
Started thread for stream: metaout
Host stream start:previewout
Opening stream for read: previewout
Attempting to open stream previewout
Successfully opened stream previewout with ID #2!
Starting thread for stream: previewout
Started thread for stream: previewout
depthai: INIT OK!

cpp_drone_depthai: /root/build/3rdparty/depthai-api/host/core/../core/pipeline/../nnet/tensor_entry_container.hpp:28: unsigned int TensorEntryContainer::size() const: Assertion `!_tensors_info->empty()' failed.
Aborted
```

Luxonis-Brandon, 2020-05-07:
Got it, thanks for checking.  So this actually was just a different check that was already there...  We'll get this detected as well.

Thanks again!

Luxonis-Steven, 2020-07-01:
Hi @ConnorChristie , I did a test today by removing a JSON config under an nn folder and ran the app. The latest app has the feature to notify the user "NN JSON not found" and exit right after the warning (example shown as below). Feel free to let us know if has any thoughts. 
[image]

Thanks!

closed 2020-07-02 by Luxonis-Brandon