luxonis/depthai #90: depth_color_h visualization Improvement

Luxonis-Brandon, 2020-05-04:
Following notes from Alex on Slack:
The depth_color_h could be enhanced actually, it only uses 64 disparity levels, lines 7 to 70 here: https://github.com/luxonis/depthai-api/blob/master/shared/disparity_luts.hpp

The rest are zero, so mapped to black color for objects in close proximity. Gives a false impression of reaching the disparity limit. (edited) 

We should just construct a better LUT there.

Luxonis-Steven, 2020-06-29:
added colormap on disparity, on python frontend. Now disparity output shows in more blue range instead of mapped to black color. Thoughts @Luxonis-Brandon ?
[image]

Luxonis-Brandon, 2020-07-08:
This is implemented in https://github.com/luxonis/depthai/pull/148

Luxonis-Brandon, 2020-07-27:
Was merged this weekend.  :-)

closed 2020-07-27 by Luxonis-Brandon