luxonis/depthai #97: Disparity Confidence Threshold

Luxonis-Brandon, 2020-05-07:
Background:
Currently we hard-code in firmware the disparity confidence threshold to be 200 out of a configurable range of 0 to 255.  

Improvement:
Allow this to be set from the host as a parameter in the `depth` entry in `config`: https://github.com/luxonis/depthai/blob/0611eb701fffbcd743644e4f0d2c7b92e60b7cd9/depthai.py#L382

ConnorChristie, 2020-05-27:
@Luxonis-Brandon I'm not quite sure where to ask this but is there any plan on open sourcing the firmware for the boards? More specifically, the contents of the `depthai.cmd` binaries.

Luxonis-Brandon, 2020-05-27:
Hi @ConnorChristie ,

Great question.  So unfortunately this is out of our control.  So anything that has the Myriad X directly on a board can't be open sourced (hence why we made a module, so the main boards can be open source) and also the code that runs on the Myriad X cannot be open sourced (hence why we made a fully-open-source API on the host and will be doing the same for I2C, SPI, and UART interface).  

So just like when using the NCS2, the DepthAI and megaAI (see [here](https://www.crowdsupply.com/luxonis/megaai)) firmware is closed source.

Our plan though is to make everything useful one could do w/ the Myriad X through the API.  This saves everyone a TON of time speeding up on how to properly time and semaphore hardware blocks with 1mb of memory handling 15 gbps of raw image data.  :-)  We do it so everyone else doesn't need to.

What are you hoping to accomplish (the threshold above?  If it's a priority for you we can crank it out faster...).

Thoughts?

Thanks,
Brandon

ConnorChristie, 2020-05-27:
@Luxonis-Brandon It was more about curiosity and that I am also looking for other ways to communicate with the depthai board instead of using USB. Mainly so I can transition to utilizing a microcontroller instead of a raspberry PI for example.

When you say "hence why we made a fully-open-source API on the host and will be doing the same for I2C, SPI, and UART interface" does this mean you are adding those additional communication protocols in addition to USB? Or is this something that is already functional?

Also, this doesn't really have anything to do with the Github issue so if you want, we can take this elsewhere but I just didn't know where to start the convo.

Luxonis-Brandon, 2020-05-29:
> @Luxonis-Brandon It was more about curiosity and that I am also looking for other ways to communicate with the depthai board instead of using USB. Mainly so I can transition to utilizing a microcontroller instead of a raspberry PI for example.

Totally makes sense.  So we think a huge value of DepthAI (and associate products) will be to use them with microcontrollers instead of hosts... since DepthAI itself is effectively a microcontroller w/ CV/AI super-powers.  

> When you say "hence why we made a fully-open-source API on the host and will be doing the same for I2C, SPI, and UART interface" does this mean you are adding those additional communication protocols in addition to USB? Or is this something that is already functional?

SPI is now, but it's super rigid and more for a super proof-of-concept sort of application.  We're planning on making it more generally useful but first we're working on a pipeline.  So it does work, though.  I could see if it's usable enough that we could bring you into an Alpha group to try it or something (we would be making the Alpha group, and you'd be the first member!).  If interested shoot me an email and we'll figure out what can be done here.

> Also, this doesn't really have anything to do with the Github issue so if you want, we can take this elsewhere but I just didn't know where to start the convo.

At this point this is a fairly not-busy Github so I think it's totally fine to throw this convo in here.  :-)

ConnorChristie, 2020-06-01:
@Luxonis-Brandon Email sent, thanks in advance!

Luxonis-Brandon, 2020-06-01:
Got it thanks!

Luxonis-Brandon, 2020-07-07:
This is initially implemented in https://github.com/luxonis/depthai/pull/144

It's quite entertaining to play with, particularly given that it's real-time:
[image]

Luxonis-Brandon, 2020-07-08:
Merged to master today.

closed 2020-07-08 by Luxonis-Brandon