luxonis/depthai #105: Running just previewout stream (without AI) breaks DepthAI

VanDavv, 2020-05-12:
I want to do display frames from previewout stream, without any AI context at the moment - and I'm unable to achieve it, since depthAI throws errors.

Code that fails (cannot reproduce on depthai.py since it always adds some nnet by default):
```
import consts.resource_paths
import cv2
import depthai

if not depthai.init_device(consts.resource_paths.device_cmd_fpath):
    raise RuntimeError("Error initializing device. Try to reset it.")

p = depthai.create_pipeline(config={
    "streams": ["previewout"]
})

while True:
    _, data_packets = p.get_available_nnet_and_data_packets()

    for packet in data_packets:
        if packet.stream_name == 'previewout':
            data = packet.getData()
            data0 = data[0, :, :]
            data1 = data[1, :, :]
            data2 = data[2, :, :]
            frame = cv2.merge([data0, data1, data2])
            cv2.imshow('previewout', frame)

    if cv2.waitKey(1) == ord('q'):
        break

del p
depthai.deinit_device()
```

Error log from run
```
Stopping threads: ...
E: [global] [    581628] [python] XLinkReadData:165	addEventWithPerf(&event, &opTime) failed with error: 3
Device get data failed: 3
Closing stream previewout: ...
Closing stream previewout: DONE.
E: [global] [    581629] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_CLOSE_STREAM_REQ

Thread for previewout finished.
E: [xLink] [    581629] [Scheduler00Thr] sendEvents:1036	Event sending failed
Stopping threads: DONE 3.749s.
Closing all observer streams: ...
Closing stream host_capture: ...
E: [global] [    581629] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    581629] [Scheduler00Thr] sendEvents:1036	Event sending failed
Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    581629] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    581629] [Scheduler00Thr] sendEvents:1036	Event sending failed
Reseting: DONE.
```

Luxonis-Brandon, 2020-06-23:
This seems to not be a problem for me anymore.  
[image]

Can you confirm @VanDavv ?

Luxonis-Brandon, 2020-06-23:
Oh sorry I just re-read the comment about not being able to reproduce this w/out using custom code.  So likely still a problem?

VanDavv, 2020-06-23:
Yes @Luxonis-Brandon , just re-ran the code and it's still a problem. 

This happens because the previewout resolution is determined by the nnet input shape. So if we don't specify one, just like I did in this code sample, the desired previewout stream shape is unknown

Luxonis-Brandon, 2020-06-23:
OK perfect thanks for confirming @VanDavv .

Luxonis-Brandon, 2020-12-21:
This is now possible in Gen2.  See the Python example here: https://github.com/luxonis/depthai-experiments/blob/f08e06c/gen2-camera-demo/main.py#L49

And there's also a C++ example in depthai-core: https://github.com/luxonis/depthai-core/blob/84618ea5beeee4548b1592b30764aef74ab5160b/example/main.cpp#L198

closed 2020-12-21 by Luxonis-Brandon