luxonis/depthai #111: Add parameter to specify whether get_available_nnet_and_data_packets() is blocking or not

Luxonis-Brandon, 2020-05-18:
Start with the `why`:
Currently pipeline.get_available_nnet_and_data_packets() is non-blocking (returns even if no data is available), causing a high polling rate in the python script, that also increases the CPU usage. So on depthai.py, running the script with default options, I measured a 330 times/s rate.

The 'what':
So it would be great to pass a parameter to get_available_nnet_and_data_packets(), whether to block waiting for new data, or not. The non-blocking behavior may still be useful if there is additional custom processing in the python script main loop (single-threaded model).

themarpe, 2020-06-04:
Added the capability (set by default) in develop branch (Commit: 09aa811ba0f6a239fa615e1b213d82f0428c3f1f)

Running metadata, previewout and video, It runs the loop at around 70 times/s. This number mostly reflects number of packets per second coming into host from device.

Luxonis-Brandon, 2020-06-04:
Sweet thanks!

Luxonis-Brandon, 2020-09-08:
This is now in the `main` (default) branch.

closed 2020-09-08 by Luxonis-Brandon