luxonis/depthai #124: OCR Support on DepthAI (and megaAI)

Luxonis-Brandon, 2020-06-01:
So talking offline (email) with Marx WRT this effort ([here](https://discuss.luxonis.com/d/40-questions-re-depthai-usb3-ffc-edition-cameras/20)) we're discussing options for one thing we haven't really investigated on DepthAI/megaAI:

- Optical Character Recognition (OCR)

> Hi Brandon,
> 
> Cool!  :)
> 
> Yeah, regarding OCR, I'm thinking of testing and implementing the
> following options:
> 
> 1.  OpenVINO's Text Detection (
> https://docs.openvinotoolkit.org/2019_R1/_text_detection_0002_description_text_detection_0002.html
> ) and Text Recognition (
> https://docs.openvinotoolkit.org/2019_R1/_text_recognition_0012_description_text_recognition_0012.html
> ) pre-trained models in tandem ;
> 
> 2.  Any pre-trained TF or TF Lite text recognition model that can be
> converted into OpenVINO format ;
> 
> 3.  Tesseract OCR, its built-in text detection functions and
> PyTesseract, though I think this'll use the RPi CM3+'s resources,
> since I'm not sure at the moment if it'll be possible for DepthAI to
> do this at the moment ;
> 
> 4.  A custom-trained (more of a fine-tuned) Tesseract model for text
> recognition, though I'll first need to understand how something like
> this could possibly be done for OpenVINO ; and
> 
> 5.  Any of the most suitable on-device text detection and text
> recognition options above, along with Microsoft or Google's Cloud OCR
> (as a user option, or as a default option when an Internet
> connection's available ) ...
> 
> Thanks!  :)
> 
> Best Regards,
> Marx

So a long time ago (January 24th) we had very briefly tried `text-detection-0001` and got a lack of memory response and haven't circled back since.  Recently, a customer saw the same error (presumably the same error) when running tiny-yolo-v2 on megaAI (see Github issue [here](https://github.com/luxonis/depthai/issues/123) and [here](https://github.com/luxonis/depthai/commit/0aaea0b5f8fbafc8a4c6f8b165a61f89b3cf77b2)).

So this branch increases the maximum blob size from 50MB to 150MB, which itself might be enough for `text-detection-0001` ([here](https://docs.openvinotoolkit.org/2018_R5/_docs_Retail_object_detection_text_pixel_link_mobilenet_v2_0001_tf_desc_text_detection_0001.html)).

Luxonis-Brandon, 2020-06-01:
This looks of interest too:
https://docs.openvinotoolkit.org/latest/_demos_text_detection_demo_README.html

MXGray, 2020-06-02:
Hi @Luxonis-Brandon,

Looks really interesting.  :)
And I hope we can think of something to go around this issue:

QUOTE:
NOTE: On VPU devices (Intel® Movidius™ Neural Compute Stick, Intel® Neural Compute Stick 2, and Intel® Vision Accelerator Design with Intel® Movidius™ VPUs) this demo is not supported with any of the Model Downloader available topologies.
/QUOTE
SOURCE:  https://docs.openvinotoolkit.org/latest/_demos_text_detection_demo_README.html

Luxonis-Brandon, 2020-06-02:
Thanks.  I noticed that after I posted it and meant to circle back.  I think `text-detection-0001` did say that it was VPU compatible.  But I'm not finding details on it right now.  

https://docs.openvinotoolkit.org/2018_R5/_docs_Retail_object_detection_text_pixel_link_mobilenet_v2_0001_tf_desc_text_detection_0001.html

Luxonis-Brandon, 2020-06-29:
So we were able to get the text detector working even though the documentation says it doesn't work.  It seems to be pretty good, although it runs at about 1FPS.

MXGray, 2020-06-30:
@Luxonis-Brandon 

Cool!  :)
Just a quick question though:

Can models that work in OpenCV DNN be used, just like how NCS1 and NCS2 are used with RPi for these OpenCV DNN models?
•  My purpose in asking is that there are a few other pre-trained text detector models, like the EAST text detector that Adrian tackled in this [URL="https://www.pyimagesearch.com/2018/08/20/opencv-text-detection-east-text-detector/"]blog post[/URL] ...

Please advise. Thanks!  :)

Luxonis-Brandon, 2020-06-30:
Hi @MXGray ,

So most any network that works with the NCS2 will work on DepthAI.  Only very large models might struggle (as on the NCS2 it only has to do neural processing, and on DepthAI we do all the other work of dealing w/ the image sensors/etc. so if the model uses all 512MB of RAM on the NCS2, it might not fit in DepthAI).  Most models are not so big though so will work.

We will see if it's possible to convert that model using OpenVINO to run on the NCS2/DepthAI... starting probably by Googling to see if anyone has already converted this to run on the NCS2.

Thanks for pointing us to this!

Best,
Brandon

Luxonis-Brandon, 2020-12-17:
We now have EAST text detection running on DepthAI:
[image]

Luxonis-Brandon, 2020-12-21:
@alex-luxonis is working on the crop+warp necessary to take the EAST text detection output and feed into the OCR second stage network.  CC: @rt-luxonis

Luxonis-Brandon, 2021-01-25:
This is now implemented here: https://github.com/luxonis/depthai-experiments/tree/master/gen2-ocr

closed 2021-01-25 by Luxonis-Brandon

schreiterjp, 2022-05-28:
Hi @Luxonis-Brandon tried your implementation mentioned in last comment. Right now it is failing because there isn’t a model for east/ocr. Where can i find the .blob file?

Luxonis-Brandon, 2022-05-28:
Asking.  Sorry about the trouble.

vladvyskocil, 2022-05-28:
@schreiterjp Can you please try it again now?

schreiterjp, 2022-05-29:
Thanks for the insane fast support @Luxonis-Brandon and @vladvyskocil now the model was downloaded an is finally working. 🎉