luxonis/depthai #129: Application crashes when running for a long period of time

VanDavv, 2020-06-08:
I've been able to reproduce and verify a couple times that running the test.py for a long time, and using more streams than just `previewout` and `metaout`, results in the application crashes (mostly segfaults, but there are some other errors too, including Python type mappings errors).

Most recent configurations ran:

- `python3.7 depthai.py --streams previewout metaout --video /dev/null` crashed with the following backtrace:

    ```
    Error while getting packet data: Unable to convert function return value to a Python type! The signature was
        (self: depthai.DataPacket) -> array
    Thread 1 "python3.7" received signal SIGFPE, Arithmetic exception.
    raise (sig=8) at ../sysdeps/unix/sysv/linux/raise.c:50
    50  ../sysdeps/unix/sysv/linux/raise.c: No such file or directory.
    (gdb) backtrace
    #0  0x76f378ec in raise (sig=8) at ../sysdeps/unix/sysv/linux/raise.c:50
    #1  0x76ca51f8 in __aeabi_ldiv0 ()
        at /lib/arm-linux-gnueabihf/libgcc_s.so.1
    #2  0x71d84c40 in HostDataPacket::getPythonNumpyArray() ()
        at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #3  0x71db8640 in pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}::operator()(HostDataPacket*) const ()
        at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #4  0x71e085ec in pybind11::array* pybind11::detail::argument_loader<HostDataPacket*>::call_impl<pybind11::array*, pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&, 0u, pybind11::detail::void_type>(pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_meth--Type <RET> for more, q to quit, c to continue without paging--c
    od, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&, pybind11::detail::index_sequence<0u>, pybind11::detail::void_type&&) () at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #5  0x71defaa0 in std::enable_if<!std::is_void<pybind11::array*>::value, pybind11::array*>::type pybind11::detail::argument_loader<HostDataPacket*>::call<pybind11::array*, pybind11::detail::void_type, pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&>(pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&) && () at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #6  0x71dd4e4c in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}, pybind11::array*, HostDataPacket*, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&&, pybind11::array* (*)(HostDataPacket*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(pybind11::detail::function_call&)#3}::operator()(pybind11::detail::function_call) const () at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #7  0x71dd4e9c in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}, pybind11::array*, HostDataPacket*, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&&, pybind11::array* (*)(HostDataPacket*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call) () at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #8  0x71d7e33c in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () at /home/pi/Desktop/depthai/depthai.cpython-37m-arm-linux-gnueabihf.so
    #9  0x001bd678 in _PyMethodDef_RawFastCallKeywords ()
    #10 0x001bf4e8 in _PyObject_FastCallKeywords ()
    ```
- `python3.7 depthai.py --streams previewout metaout meta_d2h` crashed with the following backtrace:

    ```
    Thread 1 "python3.7" received signal SIGSEGV, Segmentation fault.
    0x76fb94e8 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
    (gdb) backtrace
    #0  0x76fb94e8 in memcmp () from /usr/lib/arm-linux-gnueabihf/libarmmem-v7l.so
    #1  0x7577766c in std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >::compare(char const*) const ()
       from /usr/lib/arm-linux-gnueabihf/libstdc++.so.6
    #2  0x724f3450 in bool std::operator==<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*) ()
       from /home/pi/Desktop/depthai-python-extras/depthai.cpython-37m-arm-linux-gnueabihf.so
    #3  0x724f9088 in bool std::operator!=<char, std::char_traits<char>, std::allocator<char> >(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, char const*) ()
       from /home/pi/Desktop/depthai-python-extras/depthai.cpython-37m-arm-linux-gnueabihf.so
    #4  0x725b1388 in HostPipeline::getConsumedDataPackets[abi:cxx11]() ()
       from /home/pi/Desktop/depthai-python-extras/depthai.cpython-37m-arm-linux-gnueabihf.so
    #5  0x725b8564 in CNNHostPipeline::getAvailableNNetAndDataPackets[abi:cxx11]()
        ()
       from /home/pi/Desktop/depthai-python-extras/depthai.cpython-37m-arm-linux-gnueabihf.so
    #6  0x72517db4 in pybind11::cpp_function::cpp_function<std::tuple<std::__cxx11::--Type <RET> for more, q to quit, c to continue without paging--c
    list<std::shared_ptr<NNetPacket>, std::allocator<std::shared_ptr<NNetPacket> > >, std::__cxx11::list<std::shared_ptr<HostDataPacket>, std::allocator<std::shared_ptr<HostDataPacket> > > >, CNNHostPipeline, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(std::tuple<std::__cxx11::list<std::shared_ptr<NNetPacket>, std::allocator<std::shared_ptr<NNetPacket> > >, std::__cxx11::list<std::shared_ptr<HostDataPacket>, std::allocator<std::shared_ptr<HostDataPacket> > > > (CNNHostPipeline::*)(), pybind11::name const&, pybin [... cut ...] on_call) () from /home/pi/Desktop/depthai-python-extras/depthai.cpython-37m-arm-linux-gnueabihf.so
    #11 0x724da33c in pybind11::cpp_function::dispatcher(_object*, _object*, _object*) () from /home/pi/Desktop/depthai-python-extras/depthai.cpython-37m-arm-linux-gnueabihf.so
    #12 0x001beb34 in _PyMethodDef_RawFastCallKeywords ()
    #13 0x001c09a0 in _PyObject_FastCallKeywords ()
    #14 0x0011c138 in ?? ()
    Backtrace stopped: previous frame identical to this frame (corrupt stack?)
    ```

Luxonis-Brandon, 2020-06-10:
OK, so to summarize my understanding on this, which the team is quite likely to correct and say the correct thing.  :-)

Using [AddressSanitizer](https://en.wikipedia.org/wiki/AddressSanitizer) from Google we were able to see there were `heap-use-after-free` ERROR which lead to the discovery that [spsc_queue](https://www.boost.org/doc/libs/1_53_0/doc/html/boost/lockfree/spsc_queue.html) (single producer, single consumer) was being misused as an mpmc_queue (mutiple producer multiple consumer) non-locking queue, which it was not designed for (hence the name, single producer, single consumer).

See https://github.com/luxonis/depthai/pull/130 for more details and a potential short-term fix.  

I'm thinking that ultimately we want to go with the mpmc_queue, maybe [this one](https://github.com/rigtorp/MPMCQueue)?

Luxonis-Brandon, 2020-06-10:
Szabi brought up that we could just be using multiple spsc_queue for each stream as a simpler fix.

szabi-luxonis, 2020-06-11:
Fixed in #130

closed 2020-06-11 by szabi-luxonis