luxonis/depthai #135: Digital PTZ Support

Luxonis-Brandon, 2020-06-24:
## Start with the `why`:

Luxonis-Brandon, 2020-12-28:
As an update, the first version of this will be a fixed-zoom level, with the capability to pan/tilt around the whole frame (so a moving window).

Luxonis-Brandon, 2021-01-11:
And here is the example of how to use it in C++:
https://github.com/luxonis/depthai-core/blob/gen2_develop/examples/src/color_camera_control_example.cpp

Luxonis-Brandon, 2021-01-21:
And as a note on the example above:

It crops out a 640x360 region from RGB camera and resizes that to 300x300 (preview).
These parameters can be changed to fit your needs.  For example to output 1920x1080 crop from a 4K image sensor setting, set `colorCam.setVideoSize(1920, 1080)` and set camera resolution to 4k:
`colorCam.setResolution(dai.ColorCameraProperties.SensorResolution.THE_4_K)`

And to set the sensor to 12 MP, use `SensorResolution.THE_12_MP` instead of `SensorResolution.THE_4_K`.

bharath5673, 2021-05-23:
@Luxonis-Brandon how to use colorCam.setVideoSize(1920, 1080) on yolo or mnssd  ?? any examples ?
i want the preview size to be bigger with all detections happenning .. plz help :

Luxonis-Brandon, 2021-12-31:
Forgot to close this.  Demo videos [here](https://www.youtube.com/watch?v=H-FjrbWsaKg) and [here](https://www.youtube.com/watch?v=uylZgG3yLiU)

closed 2021-12-31 by Luxonis-Brandon

fitzzzzzzzzz, 2023-11-09:
both links are not working! could you please share if possible