luxonis/depthai #136: DepthAI Pipeline Builder Gen2

Luxonis-Brandon, 2020-06-26:
### Start with the ‘why’:
Several of the real-world applications that are desired of the DepthAI platform are actually series or parallel (or both) combinations of neural networks with regions of interest (ROI) passed from one network to one or more subsequent networks.  

### The ‘how’: 

1. Check through the flows below and see what functions are needed to connect networks/etc.
2. Propose blocks that need to implemented and API structure necessary to allocate and appropriately limit resources dynamically.
3. Propose definition system (e.g. graph builder) for defining which models go where - i.e. the end-user interface for placing blocks in the pipeline.

### The ‘what’:
If we support the following with our pipeline builder it seems it would be sufficiently flexible.
So implement a pipeline builder which can be used to implement the flows below.  

#### To support:
 - The OpenVINO security barrier demo ([here](https://docs.google.com/document/d/1SfQsLaxgnzETGHBMGaSVDNh713trCMRScVtCnuPWs9A/edit)). 
    - This does vehicle detection, followed by two parallel networks that operate on the ROI of the vehicle:
    - 1 x NN for vehicle color and vehicle type
    - 1 x NN for license plate detection
    - Then the ROI from the plate detection is passed to another NN which outputs region and OCRs the plate.
Interactive Face Detection Demo (here)
Does face detection and allows running the following as secondary networks run on the ROI of the face
Age + Gender recognition
Facial Expression estimation (incorrectly called ‘emotion’ in their doc)
Facial Landmarks
This demo runs them all in parallel… we don’t have to do that, selecting only one is fine.
Interactive Face Recognition Demo (here)
Detects faces and runs landmarks and face-reidentification to recognize the people.
Cross Road Camera Demo (here)
Detects people, vehicle, bikes, and then runs person attributes and person re-idenfitication on the ROI of detected people.
Pedestrian Tracker (here)
Detects people, re-identification runs on the ROI from person detection
Text Detection and Recognition (OCR) (here)
Detects regions that have text, and then OCR these regions.

Of the examples on the OpenVINO repository, the following seems like it should not be implemented, as it’s the only one that does series, parallel, and output of parallel back to a single model.  So it seems much more specialized.

This will then cover the following items which were previously independently on the DepthAI roadmap:
Get two-stage face detection and following age-gender or emotion working [ALEX]
Person detection, tracking, and reidentification. 
Add capability to run multiple neural networks in parallel
Integrate face detection and identification AP with Python API
First step: without depth
Second step: with depth.
Most common compliment to object detection
Be able to run multiple models in sequence (e.g. facial detection -> facial landmark -> landmark tracking)
Different than multiple-output tensor. (PR)

To not support:
Gaze Estimation (here)
Does face detection, ROI of which goes to both head pose estimation and facial landmarks.
The outputs of head pose estimation and facial landmarks are passed to the gaze estimation model

To keep in mind, but maybe not support initially:
This `smart motion` sort of pipeline, here, which is using motion detection to determine what subset of an scene to pass into object detection, followed by object tracking on the detected object detection

MXGray, 2020-06-26:
@Luxonis-Brandon,

A pipeline builder can make things quicker and more straightforward to piece up!  :)
Some things I'm about to try:

Default mobilenet SSD (Coco) with depth:

> If person
>> Run face recognition and face reidentification

>>> If stranger
>>>> Run age / gender estimator
>>>>> Run facial expression estimator
>>>>>> Run action classifier
Output:  09:00. Person. Male. 20 to 25 years old. Looking happy. Standing. 5 meters away.

>>> If not stranger
>>>> Run facial expression estimator
>>>>> Run action classifier
Output:  09:00. Marx. Looking happy. Standing. 5 meters away.

> If not person
>> Run OCR detection

>>> If text detected
>>>> Run text recognition
Output:  Dead center. Monitor. 3 meters away. Text reads, " Warning: Aliens Spotted Near You ".

>>> If no text
>>>> Pass
Output:  Dead center. Monitor. 3 meters away.

:D

Luxonis-Brandon, 2020-06-26:
Great feedback @MXGray !  Discussing internally now how difficult such results-based dynamic pipelines would be to implement.  I definitely see how useful this would be... not to investigate the relative difficulty/feasibility.

Luxonis-Brandon, 2020-10-21:
The initial Gaze estimation example is implemented here: https://github.com/luxonis/depthai-experiments/pull/8

Luxonis-Brandon, 2021-03-29:
This is now implemented and mainlined.  Most things that were possible in Gen1 API are now possible in Gen2.  See below for resources:

- The official documentation here: [https://docs.luxonis.com/en/latest/](https://docs.luxonis.com/en/latest/)
- Some examples of using the Gen2 API here: [https://github.com/luxonis/depthai-experiments](https://github.com/luxonis/depthai-experiments)
- API Documentation: [https://docs.luxonis.com/projects/api/en/latest/references/python/](https://docs.luxonis.com/projects/api/en/latest/references/python/)
- Code Samples [here](https://docs.luxonis.com/projects/api/en/latest/samples/01_rgb_preview/)

closed 2021-03-29 by Luxonis-Brandon