luxonis/depthai #146: Feature Tracking Stream Support

Luxonis-Brandon, 2020-07-07:
Start with the `why`:

Feature tracking when combined with, stereo depth, and IMU support (https://github.com/luxonis/depthai-hardware/issues/8) and/or AprilTags support (https://github.com/luxonis/depthai-hardware/issues/8) is quite useful in terms of providing self-localization capability on robotic platforms/etc.

We also have initial prototypes of feature tracking working (like [here](https://youtu.be/NVUW1nCgOLk)) so adding this to the DepthAI API is not high technical risk.

Move to the `how`:

Move our prototype Feature Tracking support into DepthAI API and formalize support, likely integrating with IMU results for improved movement/orientation accuracy.

Move to the `what`:

Support feature tracking through DepthAI to enable visual odometry applications.

Luxonis-Brandon, 2020-10-02:
@Soar-RTD here is our issue for Feature Tracking.  This is planned for December 2020 release, but likely will be earlier.

botmayank, 2021-06-02:
Hey, is this a node about to be mainlined in gen2? @Luxonis-Brandon

Luxonis-Brandon, 2021-06-02:
Unfortunately we haven't started on it yet.  We've been heads-down on some stability things and other pressing implementations (like improved IMU support https://github.com/luxonis/depthai-shared/pull/28).

We're getting close though: https://github.com/orgs/luxonis/projects/2

So you can see 107 done, 10 in progress, and 1 bug in that series of deliveries.  We had intended for all of those to be done in December.  

We have 9 left, and Feature Tracking is fifth on the list, which means it should be soon-ish.  I'm hoping in the next 3 months.

botmayank, 2021-06-08:
Great! thanks, the rough timeline helps too.

Luxonis-Brandon, 2021-07-03:
@szabi-luxonis is on this now.  :-)

Luxonis-Brandon, 2021-07-06:
This is now initially working in the Gen2 system.  Demo video to come shortly.

Luxonis-Brandon, 2021-07-06:
Here's a quick demo video:
[image]

Another here:
https://www.youtube.com/watch?v=f0tTW37E7fA

And to try this pre-alpha you can run (in `depthai-python`):
```
git pull 
git checkout feature_tracker
python3 examples/install_requirements.py
python3 examples/feature_tracker.py
```

And then ignore the `Use WASD keys to move ROI!` note in the terminal, this is a carryover from the spatial location calculator which was used to quickly prototype the output.

jlecoeur, 2021-07-07:
That looks great !

Can you give more information on the underlying algorithm (feature type, tracking algorithm), and performance (image resolution, number of features, refresh rate)? I suspect this is tracking corners, and I wonder if you have any plans for supporting ORB?

Luxonis-Brandon, 2021-07-07:
This node is based on a bunch of work that engineers at Intel did building the underlying algorithm and combinations of algorithms, and tuning thereof to make this.  Our work is integrating it into our Gen2 pipeline builder system and easy-to-use in our API, and then also tying in our resource-allocation system. All the algorithm credit/performance/etc. goes to the engineers at Intel.  And so with that said, here's how it works:

The feature detector will be selectable between:
 - Shi-Tomasi
 - Harris

The feature tracker is a modified version of pyrLK, which outperforms standard pyrLK because of the use of a Sharr filter.  And filtering is more strict than OpenCV.

We will produce some more rigorous performance information, but here's how it compares at the top level to other algorithms/etc.

Average tracking length [frames]:
Shi-Tomasi: 23 frames (with 359 features average)
Harris: 10 frames (with 365 features average)
GFTT + pyrLK: 8 frames (with 260 features average)
FAST + pyrLK: 8 frames (with 360 features average)

Tracker performance; Average tracked / frame:
Shi-Tomasi: 344 features
Harris: 329 features
GFTT + pyrLK: 319 features
FAST + pyrLK: 323 features

And for temporal performance, it depends on resolution and configuration parameters, but for 1280x720 from a single grayscale camera, it looks to be range from 19ms/frame to 10ms/frame.  And this depends on how the hardware blocks are configured internally.  

So as far as I can tell, this means at least 30FPS at 128x720p, and in many cases likely 20FPS is possible.  

I'm actually unfamiliar with ORB (which isn't a surprise; I'm still relatively new to computer vision) - could you share a good reading link on it?  Or like a short primer?  My gut is since Intel already did all this optimization for us, it is unlikely that we'd implement ORB unless it's insanely better.

Thoughts?

Thanks,
Brandon

Luxonis-Brandon, 2021-07-07:
Here's the pre-alpha (not sure if is Harris or Shi-Thomasi) running at 1280x800 with the sensor set to 30FPS.  I set the sensor to 60FPS as well and didn't notice a difference - but likely hard to see visually as above 30FPS it's tricky to tell without the aid of some measurement device.

[image]

jlecoeur, 2021-07-08:
Thank you for the detailed answer. These numbers sound really good 🚀

Something to keep in mind for VIO/VSLAM is that it is preferable to have fewer, better quality, and evenly spaced features, rather than detecting all features on the image. I can point you to this exemple of GPU accelerated feature detection and tracking, where features are pruned and forced to be distributed on the whole image https://arxiv.org/abs/2003.13493

As for ORB, I will link the opencv doc :-) https://docs.opencv.org/4.5.3/db/d27/tutorial_py_table_of_contents_feature2d.html
The main difference with Harris or Shi-Tomasi is that it not only detects interesting points in the image, but also describes it, allowing to track the features for extended time and even recover features that were occluded or went out of the field of view. See ORB-SLAM3 https://arxiv.org/abs/2007.11898 that achieves state of the art performance using ORB features.
That said, I now understand better your approach of packaging and providing easier access to the work done by intel engineers on the Myriad chip, and would understand if this falls outside of your bucket list.

Luxonis-Brandon, 2021-07-08:
Thanks @jlecoeur ,

> Something to keep in mind for VIO/VSLAM is that it is preferable to have fewer, better quality, and evenly spaced features, rather than detecting all features on the image. I can point you to this example of GPU accelerated feature detection and tracking, where features are pruned and forced to be distributed on the whole image https://arxiv.org/abs/2003.13493

That makes sense, thanks.  So yes I think it comes down to filtering, but of course there will always be a limit to how well-separated the points are, as the scene has such a strong impact on this (e.g. a big perfectly-white wall with a grid on the corner).  So for example actually in that paper it's also the case that there is a heavy concentration in locations where there are more features to grab:

[image]

I may be missing something though as I just looked for figures to see the actual performance on an image WRT the clustering of features.

> As for ORB, I will link the opencv doc :-) https://docs.opencv.org/4.5.3/db/d27/tutorial_py_table_of_contents_feature2d.html
> The main difference with Harris or Shi-Tomasi is that it not only detects [...]

Thanks, that all makes sense!  Appreciate the explanation.  And this gives me an idea:  

It is probably possible to make ORB run on the SHAVEs, actually with something like [OpenCL](https://docs.openvinotoolkit.org/2021.3/openvino_docs_IE_DG_Extensibility_DG_VPU_Kernel.html) or using neural-model vector-math in PyTorch (or TensorFlow/Keras) as in [here](https://rahulrav.com/blog/depthai_camera.html) to port the algorithm to run as a node in our pipeline builder.

For example @kunaltyagi ported an edge detector using the vector-math approach, [here](https://github.com/kunaltyagi/pytorch_harris).  I don't know ORB, but I would suspect for the determined individual it would probably be possible to get it to run through this framework (i.e. without Luxonis having to port it internally).

All that said, given that feature descriptors are the main additional use here, what could be super fascinating actually is to use a neural network to generate the features and to maintain the tracking of them.  As it would likely outperform any existing algorithm, and have great descriptors.  I think this is something that @saching13 is keenly interested in (and he can talk intelligently about this stuff whereas I'm mostly guessing based on hallway conversations I've overheard).

Thanks again,
Brandon

saching13, 2021-07-09:
> All that said, given that feature descriptors are the main additional use here, what could be super fascinating actually is to use a neural network to generate the features and to maintain the [...]

Yes. I have recently seen most of the Researchers try to use Learning-based feature descriptors to overcome these traditional methods. And using them might be the fastest way for your case to get it working on OAK-D. 
Here is [link](https://link.springer.com/article/10.1007/s11263-020-01359-2) on the survey of algos one of our users shared in the Discord. And recently Vision Transformers seems to be playing a prominent role. Hoping to see some new things using that.  

And Implementing ORB as NN might be hard I think it needs lot of customization to the layers in the network which might not effectively use the Processers. OpelCL might be the better approach here I think incase of trying to test it out. I am not 100% sure on this though.

asa, 2021-07-09:
Worth taking a look at [pyslam](https://github.com/luigifreda/pyslam) which has pretty wide support for algos, both learned and classic in this area.
+1 for getting ORB features or something like SuperPoint running on device!

eric-schleicher, 2021-07-12:
posted because another potentially interesting approach to features using openCL.  

https://github.com/BuaaCompile/Opencl-OpenSurf

Luxonis-Brandon, 2021-07-12:
Update to visualization:
[image]

nickjrz, 2021-07-13:
@Luxonis-Brandon that looks great! 

Can you give some details on the algorithm / network you are using? I would like to implement a dense optical flow estimation application similar to what you showed  with an OAK-D where network is able to estimate velocity as well.

Nick

Luxonis-Brandon, 2021-07-13:
Does https://github.com/luxonis/depthai/issues/146#issuecomment-875915575 help @nickjrz ?

nickjrz, 2021-07-13:
Yes, sorry just saw that comment. That definitely helps, thanks for sharing that.  @Luxonis-Brandon

dhruvsheth-ai, 2021-07-14:
@nickjrz , if you're interested I can implement an initial example of Optical Flow for Velocity estimation Visualisation on OAK-D(host-side). I'll work on this.

nickjrz, 2021-07-14:
@dhruvsheth-ai Thanks for your reply and help, I am definitely  interested on this. I have been looking at how to do that and was thinking about feeding it to a neural network to estimate velocity by the rate of change for each tracked feature sort of similar to this [https://github.com/antoninodimaggio/Voof] but actually knowing position and depth for tracked features.

Luxonis-Brandon, 2021-07-15:
Multi instance support.  Also reduced shave/CMX usage from 4shaves/4 slices to 2 shaves 2 slices, without performance penalty.

[image]

Luxonis-Brandon, 2021-07-16:
Great to know, thanks @kunaltyagi .  So one thing we will be experimenting soon is providing access through our API to more CV kernels at a low-ish level.  We have no idea if it will work (and if it works, if it will be stable), but the idea would be that if we succeed at this it would be reasonable to port anything that exists in OpenCV to run hardware-accelerated on the SHAVES through these exposed, hardware-accelerated kernels.  So we have some pressing things we need to get out first, but we're expecting in let's say 2 months or so to start investigating if this is possible.  IF it is possible, porting ORB may actually be done at the community level (among other useful ports).  CC: @szabi-luxonis

Soar-RTD, 2021-07-16:
> Shi-Tomasi is nice, but having ORB would be a huge benefit, because it's not easy to make a NN for ORB:
>
> *	FAST is nearly impossible to write on NN (that was what I tried, before working on the Shi-Tomasi)
> *	Lack of atan2 (for oFAST)
> *	Lack of support for index-lookup layers (needed for rBRIEF)

I have NO idea what all these acronyms mean. This is not an easy thing to learn as it is. And this only makes it worse.

Is there a primer for this somewhere?

BTW, I never did get my Luxonis MegaAI board to do anything.

Is there a complete procedure written somewhere that shows how to do that with Windows 10?

 

Dean

Luxonis-Brandon, 2021-07-16:
Yes, the Windows 10 steps are here: https://docs.luxonis.com/projects/api/en/latest/install/#windows-10

If you run into any trouble please feel free to join our Discord and we can help you real-ish-time: [https://discord.gg/EPsZHkg9Nx](https://discord.gg/EPsZHkg9Nx)

Thanks,
Brandon

Erol444, 2021-07-17:
> I have NO idea what all these acronyms mean. This is not an easy thing to learn as it is. And this only makes it worse. Is there a primer for this somewhere? BTW, I never did get my Luxonis MegaAI [...]

Hello Dean, as Brandon suggested you can either write to Discord or email us at support@luxonis.com, we will be happy to help you out.
Thanks, Erik

szabi-luxonis, 2021-08-10:
Implemented in : https://github.com/luxonis/depthai-python/pull/317

closed 2021-08-10 by szabi-luxonis

adammpolak, 2021-09-02:
I know this issue is closed but in case anyone else is looking at this issue regarding features and which might be best this challenge seems to compare feature approaches for which work best for relocalization: https://www.cs.ubc.ca/research/image-matching-challenge/current/

Luxonis-Brandon, 2021-09-03:
Thanks, good to know!