luxonis/depthai #149: Full YOLOv3 Support

Luxonis-Brandon, 2020-07-09:
### Start with the `why`:
Although DepthAI already supports tiny-YOLOv3, it can be useful in some cases it is advantages to use full YOLOv3 instead. Example include as when higher accuracy, more classes, or smaller objects need to be identified take precedent over the faster framerate of tinyYOLOv3.

### Move to the `how`:
We already support the model itself, so likely the only additional items needed is the tensor output processing on the host.

Ideally, we should do this on the host to start, and then later support it on the DepthAI standalone as we are doing for tinyYOLOv3 in https://github.com/luxonis/depthai/issues/137.

### Move to the `what`:
Implement support for YOLOv3 on DepthAI.

zhijietan94, 2020-08-26:
Hi Brandon, I am curious on what's the estimated FPS YoloV3 full?

Luxonis-Brandon, 2020-08-26:
Hi @zhijietan94 ,

Great question.  So full YOLOv3 seems to run about at 3.8FPS on DepthAI.  

And actually thanks for the prod because this now does work in DepthAI in master.  

Here's an example of running full YOLO:
Copy into `/resources/nn/yolo-v3/yolo-v3.blob`
Then change default blob in `depthai.py` to:
```
if default_blob:
    #default
    shave_nr = 14
    cmx_slices = 14
    NCE_nr = 2
```
https://drive.google.com/file/d/1uSnB_g0AL-zYPQjgBXJ7bkp-_CP5dpMu/view?usp=sharing

And then you can run it with `./depthai.py -cnn yolo-v3 -dd -sync -rgbf 3.8`

[image]

The sync works fine up to 3.8 fps, and the lag is about 570ms (about 260ms -- 1 frame, is because of NN delay).  This is the standard 416x416 YOLOv3 full model.

So in `develop` branch that will be `main` (and the default branch) this will all be done automatically.  The model will be downloaded and compiled to the desired number of NCE and SHAVE/CMX.  Many other niceties (like a smaller repo) will come with `develop` as well.

But for now you can run it with the above.  

Thoughts?

Thanks,
Brandon

Luxonis-Brandon, 2020-08-26:
Also the IOU and NMS are processed on DepthAI/megaAI directly in the https://github.com/luxonis/depthai/tree/openvino_tensor_representation branch as requested in https://github.com/luxonis/depthai/issues/137.  This will be going mainline as well.  And will allow https://github.com/luxonis/depthai/issues/140 .

zhijietan94, 2020-09-23:
Looks great! Would like to try it but the google drive link for YOLOv3 blob is restricted access. @Luxonis-Brandon Can you help to adjust the access rights for download? Thanks!

Luxonis-Brandon, 2020-09-23:
Yes, sorry about that.  One of my teammates owns the file so I'm re-uploading now and will make the permissions so that anyone can read it.  I'll change the link above and circle back shortly when that's done... uploading now.

Luxonis-Brandon, 2020-09-23:
Just updated it above @zhijietan94

szabi-luxonis, 2020-10-06:
Support added in #202

closed 2020-10-06 by szabi-luxonis