luxonis/depthai #165: `test.py -s depth_raw` error return code -11

billythedummy, 2020-07-29:
**Host OS:** Arch Linux 5.7.10-arch1-1

**Device/Board:** USB3 with onboard cameras / BW1098OBC

**depthai Installation and Version:** `git clone 63f7cc2853ae0f9921cef1128b5e0c69ddae94cb`, then followed instructions on [README](https://github.com/luxonis/depthai/blob/master/README.md#python-modules) to install. Pip modules installed in a virtualenv. 

**Log:**
```
python3 depthai.py '-s' 'depth_raw' 
No calibration file. Using Calibration Defaults.
Using Arguments= {'config_overwrite': None, 'board': None, 'shaves': None, 'cmx_slices': None, 'NN_engines': None, 'rgb_resolution': 1080, 'rgb_fps': 30.0, 'mono_resolution': 720, 'mono_fps': 30.0, 'disparity_confidence_threshold': 200, 'field_of_view': 71.86, 'rgb_field_of_view': 68.7938, 'baseline': 9.0, 'rgb_baseline': 2.0, 'swap_lr': True, 'store_eeprom': False, 'clear_eeprom': False, 'override_eeprom': False, 'device_id': '', 'dev_debug': None, 'force_usb2': None, 'cnn_model': 'mobilenet-ssd', 'cnn_model2': '', 'cnn_camera': 'rgb', 'disable_depth': False, 'draw_bb_depth': False, 'full_fov_nn': False, 'streams': [{'name': 'depth_raw'}], 'video': None}
depthai.__version__ == 0.1.0
depthai.__dev_version__ == def8761fc66e0fc15b9a93e835dfb0009c6f3039
XLink initialized.
Sending device firmware "cmd_file": /home/user/depthai/depthai.cmd
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
watchdog started 6000
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
WARNING: Version (dev) does not match (9d2c233550e9b731e4ea370de0e44fd7dbb1606a & def8761fc66e0fc15b9a93e835dfb0009c6f3039)
EEPROM data: valid (v3)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Calibration homography:
    1.017313,    0.003575,   -7.324748,
    0.007830,    1.010968,  -17.413794,
    0.000015,    0.000009,    1.000000,
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth_raw', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker']
CNN configurations read: /home/user/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd_depth.json
depthai: Calibration file is not specified, will use default setting;
config_h2d json:
{"_board":{"_homography_right_to_left":[0.988068163394928,0.0029474012553691864,5.067617416381836,-0.008765067905187607,0.9921473264694214,-8.795275688171387,-8.449587767245248e-06,-3.603489403758431e-06,1.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"depth_raw"}]},"ai":{"NCEs":1,"blob0_size":14489664,"blob1_size":0,"calc_dist_to_bb":true,"camera_input":"rgb","cmx_slices":7,"keep_aspect_ratio":true,"num_stages":1,"shaves":7},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":false,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"confidence_threshold":0.5,"depth_limit_mm":10000,"padding_factor":0.30000001192092896},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 5000 bytes to config_h2d
!!! XLink write successful: config_h2d (5000)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
Successfully opened stream host_capture with ID #0!
Creating observer stream host_capture: DONE.
Read: 14489664
Attempting to open stream inBlob
Successfully opened stream inBlob with ID #1!
Writing 14489664 bytes to inBlob
!!! XLink write successful: inBlob (14489664)
Closing stream inBlob: ...
Closing stream inBlob: DONE.
depthai: done sending Blob file /home/user/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.blob
Attempting to open stream outBlob
Successfully opened stream outBlob with ID #2!
Closing stream outBlob: ...
Closing stream outBlob: DONE.
CNN input width: 300
CNN input height: 300
CNN input num channels: 3
CNN to depth bounding-box mapping: start(318, 38), max_size(643, 643)
Host stream start:depth_raw
Opening stream for read: (null)
Return code:-11
```
Same thing happens using a custom `depthai.calib` after calibration, and with force usb 2.

VanDavv, 2020-07-30:
Hi @billythedummy !
Thanks for reporting the issue,
I'm working on `5.6.16-1-MANJARO` so it's also Arch-based distro, but couldn't reproduce the issue.

I've folowed the same path as you and my output looks a bit different

```diff
CNN input width: 300
CNN input height: 300
CNN input num channels: 3
CNN to depth bounding-box mapping: start(318, 78), max_size(643, 643)
Host stream start:depth_raw
-Opening stream for read: (null)
-Return code:-11
+Opening stream for read: depth_raw
+Attempting to open stream depth_raw
+Successfully opened stream depth_raw with ID #1!
+Starting thread for stream: depth_raw
+depthai: INIT OK!
+Started thread for stream: depth_raw
+=== New data in observer stream host_capture, size: 1
+Writing 1 bytes to host_capture
+!!! XLink write successful: host_capture (1)
```

I'll check why you could receive `null` instead of `depth_raw` and circle back to you.

In the mean time, could you try again, but reset the device before? For this, you have to power on the device holding the reset button on the board (shown on the image below)

[image]

billythedummy, 2020-07-30:
Just tried it. I held the reset button and plugged in the power, held my finger down on the reset button for 5 seconds, during which the RUN LED was off, and then let go, after which the RUN LED turned on. Still had the same issue.

iamsg95, 2020-07-30:
Hello @billythedummy . I see that you are building from sources, please update to latest master (both depthai and depthai-api). The issue is not present on master. Commit : c4f8ceaeb5e811a81bb0fa1b63bcd2d9b14252fd

Luxonis-Brandon, 2020-08-05:
@billythedummy - did this help?

billythedummy, 2020-08-06:
SFLR, yeah it worked when I recompiled and ran with the current master branch. Closing the issue.

closed 2020-08-06 by billythedummy