luxonis/depthai #167: SIGFPE thrown when running a simple script to access left/right camera

VanDavv, 2020-07-30:
## Overview

I've tried to prepare a minimal working code example accesing the left and right mono cameras. The script throws however the SIGFPE error while running.
This however does not occur when running `python3 test.py -s left right`, so the issue must be related to some configuration missing and therefore causing the error

## Source code

```python
import consts.resource_paths
import cv2
import depthai

if not depthai.init_device(consts.resource_paths.device_cmd_fpath):
    raise RuntimeError("Error initializing device. Try to reset it.")

p = depthai.create_pipeline(config={
    "streams": ["left", "right"],
    "ai": {
        "blob_file": consts.resource_paths.blob_fpath,
        "blob_file_config": consts.resource_paths.blob_config_fpath
    }
})

if p is None:
    raise RuntimeError("Error initializing pipelne")

while True:
    data_packets = p.get_available_data_packets()

    for packet in data_packets:
        if packet.stream_name == 'left' or packet.stream_name == 'right':
            frame = packet.getData()
            cv2.imshow(packet.stream_name, frame)

    if cv2.waitKey(1) == ord('q'):
        break

del p
depthai.deinit_device()
```

## Errors

Running this script produces the following error:

```
Thread 1 "python3" received signal SIGFPE, Arithmetic exception.
0x00007fffd99e1b07 in HostDataPacket::getPythonNumpyArray (this=0x7fffc4000b80) at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/../core/host_data_packet.hpp:94
94	            size_div /= dimensions[i];
```

And the backtrace of the error is below

```
#0  0x00007fffd99e1b07 in HostDataPacket::getPythonNumpyArray (this=0x7fffc4000b80) at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/../core/host_data_packet.hpp:94
#1  0x00007fffd9a0a567 in pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}::operator()(HostDataPacket*) const (this=0x5555559ede98, c=0x7fffc4000b80) at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/3rdparty/pybind11/include/pybind11/pybind11.h:78
#2  0x00007fffd9a52497 in pybind11::detail::argument_loader<HostDataPacket*>::call_impl<pybind11::array*, pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&, 0ul, pybind11::detail::void_type>(pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&, std::integer_sequence<unsigned long, 0ul>, pybind11::detail::void_type&&) (this=0x7fffffffd090, f=...)
    at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/3rdparty/pybind11/include/pybind11/cast.h:1935
#3  0x00007fffd9a3ca76 in pybind11::detail::argument_loader<HostDataPacket*>::call<pybind11::array*, pybind11::detail::void_type, pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&>(pybind11::cpp_function::cpp_function<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&) && (this=0x7fffffffd090, f=...)
    at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/3rdparty/pybind11/include/pybind11/cast.h:1912
#4  0x00007fffd9a249ef in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}, pybind11::array*, HostDataPacket*, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&&, pybind11::array* (*)(HostDataPacket*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(pybind11::detail::function_call&)#3}::operator()(pybind11::detail::function_call) const (this=0x0, 
    call=...) at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/3rdparty/pybind11/include/pybind11/pybind11.h:159
#5  0x00007fffd9a24a56 in pybind11::cpp_function::initialize<pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}, pybind11::array*, HostDataPacket*, pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::cpp_function::initialize<pybind11::array*, HostDataPacket, , pybind11::name, pybind11::is_method, pybind11::sibling, pybind11::return_value_policy>(pybind11::array* (HostDataPacket::*)(), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(HostDataPacket*)#1}&&, pybind11::array* (*)(HostDataPacket*), pybind11::name const&, pybind11::is_method const&, pybind11::sibling const&, pybind11::return_value_policy const&)::{lambda(pybind11::detail::function_call&)#3}::_FUN(pybind11::detail::function_call) ()
    at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/3rdparty/pybind11/include/pybind11/pybind11.h:137
#6  0x00007fffd99db7ee in pybind11::cpp_function::dispatcher (self=0x7fffd9e2d210, args_in=0x7ffff7227070, kwargs_in=0x0)
    at /home/vandavv/dev/luxonis/depthai/depthai-api/host/py_module/3rdparty/pybind11/include/pybind11/pybind11.h:624
#7  0x00007ffff7d328ce in PyCFunction_Call () from /usr/lib/libpython3.8.so.1.0
#8  0x00007ffff7d2518c in _PyObject_MakeTpCall () from /usr/lib/libpython3.8.so.1.0
#9  0x00007ffff7d3c75f in ?? () from /usr/lib/libpython3.8.so.1.0
#10 0x00007ffff7d20635 in _PyEval_EvalFrameDefault () from /usr/lib/libpython3.8.so.1.0
#11 0x00007ffff7d1a654 in _PyEval_EvalCodeWithName () from /usr/lib/libpython3.8.so.1.0
#12 0x00007ffff7dcab03 in PyEval_EvalCode () from /usr/lib/libpython3.8.so.1.0
#13 0x00007ffff7dd6248 in ?? () from /usr/lib/libpython3.8.so.1.0
#14 0x00007ffff7dd0483 in ?? () from /usr/lib/libpython3.8.so.1.0
#15 0x00007ffff7c903f8 in PyRun_FileExFlags () from /usr/lib/libpython3.8.so.1.0
#16 0x00007ffff7c8fe8a in PyRun_SimpleFileExFlags () from /usr/lib/libpython3.8.so.1.0
#17 0x00007ffff7de31e2 in Py_RunMain () from /usr/lib/libpython3.8.so.1.0
#18 0x00007ffff7dbf3d9 in Py_BytesMain () from /usr/lib/libpython3.8.so.1.0
#19 0x00007ffff7a5e002 in __libc_start_main () from /usr/lib/libc.so.6
#20 0x000055555555504e in _start ()
```

## Environment

Tested on two devices, with same error:

- 5.6.16-1-MANJARO x86_64 with compiled `.so` file. depthai repo commit: `c4f8ceaeb5e811a81bb0fa1b63bcd2d9b14252fd`, depthai-api repo commit: `6ea8de5ebccc05ea3d8648d730aa322d252d338e` using Python 3.8.3

- 5.4.51-v7+ armv7l Raspberry Pi 2B with precompiled `.so` file in the depthai repo, commit: `0f467cf862d4fcdd21ddba89e5b118a867f48120` using python 3.7.3

Both devices were using it's own BW1098OBC.

VanDavv, 2020-07-30:
Errors are not occuring when added mono camera config to `create_pipeline` invocation

```diff
p = depthai.create_pipeline(config={
    "streams": ["left", "right"],
    "ai": {
        "blob_file": consts.resource_paths.blob_fpath,
        "blob_file_config": consts.resource_paths.blob_config_fpath
    },
+    'camera': {'mono': {'resolution_h': 720, 'fps': 30}},
})
```

Luxonis-Brandon, 2020-07-30:
Good find.  Looks like a default mono config tucked somewhere is probably all that is needed.  Thoughts @iamsg95 ?

szabi-luxonis, 2020-10-06:
Fixed in #202

closed 2020-10-06 by szabi-luxonis