luxonis/depthai #168: Python dependnecies inconsistency on Raspberry Pi

asmorkalov, 2020-07-30:
Platform: Raspberry Pi 4
OS: Raspbian Buster Full (2020-02-13)

1. Instruction has inconsistency with OpenCV versions: `sudo apt install python3-opencv libcurl4` installs OpenCV 3.2, but requirements.txt contains `opencv-python==4.1.1.26; platform_machine == "armv7l"`
2. The pip package in requirements.txt contains extra dependencies that are missing in the setup instruction. I installed `sudo apt-get install -y libatlas-base-dev libhdf5-dev libhdf5-serial-dev libatlas-base-dev libjasper-dev  libqtgui4  libqt4-test`, but most probable -dev part is redundant.
3. Finally python test fails with error:
```
python3 ./test.py 
python3 depthai.py 
Traceback (most recent call last):
  File "depthai.py", line 10, in <module>
    import cv2
  File "/home/pi/.local/lib/python3.7/site-packages/cv2/__init__.py", line 3, in <module>
    from .cv2 import *
ImportError: /home/pi/.local/lib/python3.7/site-packages/cv2/cv2.cpython-37m-arm-linux-gnueabihf.so: undefined symbol: __atomic_fetch_add_8
Return code:1
```

Looks like the pip package is build with another runtime than Raspberian. I made `apt-get update` && `apt-get upgrade` beforehand and pretty sure that I use the latest and the best OS.

Luxonis-Brandon, 2020-07-30:
Thanks @asmorkalov .  We'll get this sorted and circle back.  @VanDavv I'm sure will be on it tomorrow.  :-)

VanDavv, 2020-07-31:
Hi @asmorkalov!

As I've seen [this issue](https://github.com/piwheels/packages/issues/59), there is a way to do a workaround and use our version - but ended up downgrading the OpenCV version

Also, thank you for pointing out the missing APT dependencies! In fact, we install `python3-opencv` not for usage but to have it's dependencies installed.

asmorkalov, 2020-07-31:
Workaround for pre-build OpenCV from Pypi: https://github.com/opencv/opencv/issues/15278#issuecomment-520380265

PR #169 (2020-08-01): Fix requirements on RPi 
This PR resolves #168 

As specified [here](https://github.com/piwheels/packages/issues/59) the opencv-python version we're using requires a workaround due to an error

```
undefined symbol: __atomic_fetch_add_8
```

And the proposed workaround consists of adding an environment variable

```
LD_PRELOAD=/usr/lib/arm-linux-gnueabihf/libatomic.so.1
```

However, if we downgrade the `opencv-python` version on RPi from `4.1.1.26` to `4.1.0.25`, the error does not occur and no workaround is needed
closed 2020-08-01 by iamsg95