luxonis/depthai #170: Requested CNN resources overlaps with RGB camera error on Raspberry + Oak D

asmorkalov, 2020-07-31:
Platform: Raspberry Pi 4, 4Gb RAM
OS: Raspbian Buster Full (2020-02-13)
Board: Oak-D connected to USB 3 (blue) port, extra power is connected.
`test.py` reports the error in subject, but Oak-1 works without issues. Log is attached.
[oak-d-failure.log](https://github.com/luxonis/depthai/files/5007208/oak-d-failure.log)

Luxonis-Brandon, 2020-07-31:
Thanks @asmorkalov for the error report.  Very interesting and we'll check it out soon.  

CC: @iamsg95 and @themarpe

iamsg95, 2020-08-01:
Hello @asmorkalov .
Can you check if one of your stereo cameras are disconnected on the OAK-D?
I can reproduce this issue by disconnecting one of the stereo cams.
I will add check and error feedback for this kind of issue.

iamsg95, 2020-08-02:
Can you check #171 ? If camera is disconnected it will print error message. @asmorkalov

iamsg95, 2020-08-03:
@asmorkalov can you switch to a shorter USB cable? Long usb cable can be an issue with some hosts.

Luxonis-Brandon, 2020-08-03:
And also worth mentioning we were able to reproduce this error by running a 2 meter USB cable with BW1098OBC and a specific host. 

So the moral of the story is cables no longer than 1 meter should be used, as some hosts have problems with 2 meter cables.  And when we shipped your unit back in January we included 2 meter cable as we hadn’t yet discovered this.

So using a 1 meter cable is recommended.  Alternatively, arbitrarily long cables can be used with the -fusb2 option to force USB2 communication.

Luxonis-Brandon, 2020-08-03:
Oh and we include 1 meter cables now for this reason.

asmorkalov, 2020-08-03:
@Luxonis-Brandon I made several experiments with devices and cables and out is the following:

Luxonis-Brandon, 2020-08-03:
(Looks like the results didn’t get attached.)

asmorkalov, 2020-08-03:
I found that root cause: left camera socket was ejected. I reconnect camera and the first device works good now.

Luxonis-Brandon, 2020-08-03:
Thanks @asmorkalov for circling back.  So I'm reviewing/testing https://github.com/luxonis/depthai/pull/171 now to make the debug cycle here more intuitive.  And thanks @iamsg95 for making that addition.

nullbyte91, 2020-08-12:
Hi @Luxonis-Brandon,

I'm getting a similar error with ssd_mobilenet_v2_coco. But, I don't see any error when I tried with mobilenet-ssd.blob from depthai repo.
Please find the blob conversion steps to reproduce the issue,

```bash
# Convert pb file into bin and xml
python3 mo_tf.py --input_model=/home/nullbyte/Desktop/openvino_models/model/ssd_mobilenet_v2_coco/ssd_mobilenet_v2_coco_2018_03_29/frozen_inference_graph.pb --transformations_config extensions/front/tf/ssd_v2_support.json --tensorflow_object_detection_api_pipeline_config /home/nullbyte/Desktop/openvino_models/model/ssd_mobilenet_v2_coco/ssd_mobilenet_v2_coco_2018_03_29/pipeline.config --reverse_input_channels -o ~/Desktop/mo
_models/mobilenet-ssd-v2/

# xml and bin to blob
 ./bin/aarch64/Release/myriad_compile -m ~/Desktop/mo_models/mobilenet-ssd-v2/frozen_inference_graph.xml -o  ~/Desktop/mo_models/
mobilenet-ssd-v2/FP32/mobilenet-ssd-v2.blob -ip FP32 -VPU_MYRIAD_PLATFORM VPU_MYRIAD_2480 -VPU_NUMBER_OF_SHAVES 4 -VPU_NUMBER_OF_CMX_SLICES 4
```

Since the mobilenet SSD v1 configuration is similar to v2, so I used the same JSON file. 

**System Information:**
Jetson Nano + jetpack 4.4 
OpenVino Native compilation - Master Branch

Regards,
Jegathesan S

iamsg-luxonis, 2020-08-12:
@nullbyte91  it needs to be `-ip U8` instead `-ip FP32`. That is the input format the model receives. In the first layer is converted to FP16, which is the only format supported by Myriad.

nullbyte91, 2020-08-12:
@iamsg-luxonis Thanks for the reply. I tried all the precision including U8 format I forgot to mention that. Please find the attached error log. 
[ssd-v2_error.log](https://github.com/luxonis/depthai/files/5060238/ssd-v2_error.log)

iamsg-luxonis, 2020-08-12:
@nullbyte91 which openvino version are you using? If it's compiled with other than 2020.1 or 2020.2 it won't work, since they changed some internal structure, and we will need to update the firmware to support it. 2020.1 and 2020.2 will work.
You may use this cloud compiler: http://luxonis.com:8080/

nullbyte91, 2020-08-12:
@iamsg-luxonis Yes. I'm using the latest version. I will try with 2020.1/2020.2 and update you. Thanks again for the support. One quick question about precision support, Can we able to use both FP16 and U8 format?

iamsg-luxonis, 2020-08-12:
So `-ip U8` means that the input (image from camera) is in U8 format. That is promoted to FP16 because all the operations are in FP16 (the weights are in FP16). From what I know myriad doesn't support U8 weight quantization, but I will double check.

nullbyte91, 2020-08-13:
@iamsg-luxonis Thanks for the details. I tried with OpenVino 2020.2.117, and I don't see any error except **Device get data failed** but the display window freezes and exit. Have anyone tried mobilenet SSD v2[Tensorflow]?

rt-luxonis, 2020-08-13:
Mobilenet SSD v2 works nicely. Please see this notebook:
https://colab.research.google.com/github/luxonis/depthai-ml-training/blob/master/colab-notebooks/Easy_Object_Detection_With_Custom_Data_Demo_Training.ipynb

nullbyte91, 2020-08-14:
@rt-luxonis Thanks for the notebook. Now, It's working. I made a mistake on model conversion. 

There are **16 SHAVES** on the ship, and **4 SHAVES** are use for Pre-processing. Why are we parsing only 4 SHAVES and 4 CMX for blob conversion?

I tried changing the size of the SHAVES to 12, and I got an error "Error: Blob is compiled for 12 cmx slices but device is configured to calculate on 7". I got a better FPS when I change to 7.

What is the optimal number you would suggest to get the best performance for both Single NN and double NN inference?

iamsg-luxonis, 2020-08-14:
Best performance can be achieved w/ 14 shaves/14cmx and 2 NCEs.
For that the model needs to be compiled for 7 shaves (7 shave for each NCE = 7x2=14)
In `depthai.py` the following config needs to be changed to 14, 14, 2.
```
if default_blob:
    #default
    shave_nr = 7
    cmx_slices = 7
    NCE_nr = 1
```
It will likely exceed 30 fps, so the rgb camera fps needs to be adjusted to 40 fps (or more):
`-rgbf 40`
If inference is requested on both left and right cameras 2 NCEs are mandatory, otherwise they won't be balanced.

@nullbyte91

nullbyte91, 2020-08-14:
@iamsg-luxonis Thanks, That was very detailed information.

closed 2020-10-02 by Luxonis-Brandon