luxonis/depthai #173: Neural-Assisted Depth Estimation

Luxonis-Brandon, 2020-08-07:
## Start with the `why`:

The `why` of this effort (and initial research) is that any many applications depth cameras (and even sometimes LIDAR) are not sufficient to successfully detect objects in varied conditions.  Specifically, for Luxonis’ potential customers, this is directly limiting their business success:

- Autonomous wheelchairs.  The functionality above it would be HUGE for this application as existing solutions are struggling with the output of D435 depth.  It gets tricked too easily and misses objects even w/ aggressive host-side filtering and other detection techniques.

- Autonomous lawn mowing.  This use-case is also struggling with object detection using D435.  The system can't identify soccer-ball sized things reliably even with significant host-side post-processing and then need to be able to identify down to baseball sized things.

- Volumetric estimation of low-visual-interest objects.  Disparity depth struggles significantly with objects (particularly large objects) of low visual interest as it lacks features to match.  Neural networks can leverage latent information from training that overcomes this limitation - allowing volumetric estimation where traditional algorithmic-based disparity-depth solutions cannot adequately perform.

The original idea of DepthAI is to not solve this sort of problem, but it is well suited to solving it.

### Background: 

As of now, the core use of DepthAI is to run 2D Object Detectors (e.g. MobileNetSSDv2) and fuse them with stereo depth to be able to get real-time 3D position of objects that the neural network identifies.  See here for it finding my son's XYZ position for example.  This solution is not applicable to the above two customers because the type of object must be known to the neural network.  Their needs are to avoid any object, not just known ones, and specifically objects which are hard to pick up, which are lost/missed by traditional stereo depth vision.

### New Modality of Use
So one idea we had recently was to leverage the neural compute engines (and SHAVES) of the Myriad X to make better depth - so that such difficult objects which traditional stereo depth misses - could be detected with the depth that’s improved by the neural network.  

Implementing this capability, the capability to run neural inference to produce the depth map directly, or to improve the results of the disparity-produced depth map, is hugely enabling for the use-cases mentioned above, and likely many others.

## Move to the `how`:

The majority of the work of how to make this happen will be in researching what research has been done, and what techniques are sufficiently light-weight to be run on DepthAI directly.  Below is some initial research to that end:

### Google Mannequin Challenge:
Blog Explaining it: https://ai.googleblog.com/2019/05/moving-camera-moving-people-deep.html
Dataset: https://google.github.io/mannequinchallenge/www/index.html
Github: https://github.com/google/mannequinchallenge
Notice in a lot of caes this is actually quite good looking depth just from a single camera.  Imagine how amazing it could look with 2 or 3 cameras.  
 
Could produce just insanely good depth maps.
 
### KITTI DataSet:
 
http://www.cvlibs.net/datasets/kitti/eval_scene_flow.php?benchmark=stereo
 
So check this out.  A whole bunch of ground truth data, with calibration pictures, etc.  So this could be used to train a neural network for sure on this sort of processing.
 
And then there's a leaderboard downbelow of those who have.  
 
### PapersWithCode:
 
PapersWithCode is generally awesome.  They have a slack even.
 
https://paperswithcode.com/task/stereo-depth-estimation
 
So have a dig through there.  This one from there seems pretty neat:
https://github.com/CVLAB-Unibo/Real-time-self-adaptive-deep-stereo
 
These guys seem like they're getting decent results too:
https://arxiv.org/pdf/1803.09719v3.pdf
 
So on a lot of these it's a matter of figuring out which ones are light enough weight and so on to see about porting.  

Notice this one uses KITTI dataset as well:
https://www.cs.toronto.edu/~urtasun/publications/luo_etal_cvpr16.pdf

- From Intel R&D directly: https://arxiv.org/pdf/2001.04552.pdf  Apparently this was never implemented.
- Google’s StereoNet looks really fast/lightweight: https://arxiv.org/pdf/1807.08865.pdf
- Github summarizing depth quality enhancements using CNNs: https://github.com/mdcnn/Depth-Image-Quality-Enhancement
- This one looks pretty interesting: https://arxiv.org/pdf/1910.00541.pdf

### Others and Random Notes:
- https://web.stanford.edu/class/ee368/Project_Autumn_1516/Reports/Jordan_Shridhar.pdf Seems like the Myriad X 2x NCE + SHAVES are plenty fast enough to real-time make a super-great disparity depth output.  
- https://arxiv.org/pdf/1910.13708.pdf
- DDRNet: Depth Map Denoising and Refinement for Consumer Depth Cameras Using Cascaded CNNs: 
  - http://openaccess.thecvf.com/content_ECCV_2018/papers/Shi_Yan_DDRNet_Depth_Map_ECCV_2018_paper.pdf
  - https://github.com/neycyanshi/DDRNet
- AMNet: Deep Atrous Multiscale Stereo Disparity Estimation Networks: https://arxiv.org/pdf/1904.09099.pdf
- Stereo Matching by Training a Convolutional Neural Network to Compare Image Patches: https://github.com/jzbontar/mc-cnn/blob/master/README.md
- Siamese network.  Probably way too big ass it shows multi-second run-times: https://www.ncbi.nlm.nih.gov/pmc/articles/PMC6472548/
- The middlebury stereo dataset seems incredibly useful 
https://github.com/kelkelcheng/GC-Net-Tensorflow/blob/master/README.md
- DispNetC shows 0.06 runtime, which is encouraging.
- Real-time self-adaptive deep stereo
 - https://zpascal.net/cvpr2019/Tonioni_Real-Time_Self-Adaptive_Deep_Stereo_CVPR_2019_paper.pdf
 - https://github.com/CVLAB-Unibo/Real-time-self-adaptive-deep-stereo/blob/master/README.MD
- Pytorch implementation of the several Deep Stereo Matching Network(DSMnet) https://github.com/wyf2017/DSMnet/blob/master/README.md

2emoore4, 2020-08-14:
As a DepthAI user, I want to emphasize the importance of having clean/accurate/precise depth maps - it's clear that deep learning is the key to achieving this.

It's definitely possible to clean up depth maps with more traditional filtering, with something like the Bilateral Solver: https://drive.google.com/file/d/1zFzCaFwkGK1EGmJ_KEqb-ZsRJhfUKN2S/view

However there has been much more work recently to apply deep learning to 3d image generation, and more work is coming all the time.

Stereo Magnification introduced Multi Plane Images, and used differentiable rendering to learn to generate them from stereo images: https://people.eecs.berkeley.edu/~tinghuiz/projects/mpi/

Many have extended on this idea, but much of the latest work uses dozens of input images, instead of just two:

DeepView: https://augmentedperception.github.io/deepview/
Immersive Light Field Video w/ Layered Meshes: https://augmentedperception.github.io/deepviewvideo/
Neural Radiance Fields: https://www.matthewtancik.com/nerf

(Not all of these output MPIs, but all are fairly similar)

There's also plenty of recent work around monocular depth estimation, like MiDaS from Intel: https://github.com/intel-isl/MiDaS

Some take existing 3d photos, and try to inpaint disocclusions, so that inaccuracies are less noticeable: https://shihmengli.github.io/3D-Photo-Inpainting/

Luxonis-Brandon, 2020-08-14:
Thanks @2emoore4 !  Super appreciate it.  Will review all these shortly.  And also sharing with the team!

saching13, 2020-08-25:
I am adding the paper by kydio which carries out end to end learning for stereo. 
https://arxiv.org/pdf/1703.04309.pdf

Luxonis-Brandon, 2020-11-06:
This looks quite interesting:
https://geometry.cs.ucl.ac.uk/projects/2018/depthcut/

Luxonis-Brandon, 2020-11-10:
Check out the datasets referenced near the end of this paper:
https://arxiv.org/pdf/1612.02401.pdf
The approach is also interesting IMO, and could be adapted for deep learning from stereo.
(they are solving a harder problem which is both motion and depth from a pair of images, but you could fix motion since it's known and just focus on the depth part).

Luxonis-Brandon, 2020-12-04:
PatchmatchNet: Learned Multi-View Patchmatch Stereo
Looks like an interesting paper for resource limited devices.
https://github.com/FangjinhuaWang/PatchmatchNet
https://arxiv.org/pdf/2012.01411v1.pdf

Luxonis-Brandon, 2021-03-07:
Some additional resources from [Discord](https://discord.gg/EPsZHkg9Nx):

- [https://www.hindawi.com/journals/cin/2020/8562323/](https://www.hindawi.com/journals/cin/2020/8562323/)

Luxonis-Brandon, 2021-08-13:
https://antabangun.github.io/projects/CoEx/#dem

Luxonis-Brandon, 2021-09-05:
https://github.com/ibaiGorordo/UnrealCV-stereo-depth-generation

tersekmatija, 2021-09-05:
https://arxiv.org/pdf/2007.12140.pdf
https://github.com/ibaiGorordo/HITNET-Stereo-Depth-estimation

dhruvsheth-ai, 2021-09-07:
> https://github.com/ibaiGorordo/HITNET-Stereo-Depth-estimation

This seems to be pretty accurate. Achieved results on TFlite HITNET Stereo Depth Estimation - 
[image]

Compared to original results - 
[image]

Luxonis-Brandon, 2021-09-07:
Looks great - thanks for sharing!

Luxonis-Brandon, 2021-09-08:
https://github.com/cogsys-tuebingen/mobilestereonet - From @PINTO0309 in Discord.

Luxonis-Brandon, 2021-09-12:
The first results are starting to come.  Here's MIT Fast Depth (https://github.com/dwofk/fast-depth) running on OAK-D-(anything):
[image]

nickjrz, 2021-09-13:
Hey @Luxonis-Brandon, this looks like a great starting point for neural network assisted depth estimation. I wonder how precise it can get if we added the depth ground truth in a self-supervised training. Is the inference part running on host and  if this is the case, what would it look like to try to optimize the network run on the OAK-D  onboard?

Luxonis-Brandon, 2021-09-13:
This is running on OAK-D directly, not on the host.  Matija will be making a pull request soon so you'll be able to try it.  (He may have already and I missed it - unsure... he just got it working this weekend.)

nickjrz, 2021-09-14:
I was able to run real-time ish inference on TFLite HITNET Stereo depth estimation (middlebury) using OAK-D and having the inference on the host. Here are my results: 

I will investigate how to run this onboard of the depthAI device and optimize the performance accelerating inference using GPU acceleration

[image]

PINTO0309, 2021-09-14:
Due to a problem with OpenVINO's conversion to Myriad Blob, I submitted an issue to Intel's engineers (OpenVINO). So far, Intel engineers seem to be concerned that the structure of the model is wrong, but we are able to infer it successfully in ONNX runtime.

- [Bug] GatherND shape conversion from ONNX is inaccurate #7379 (HITNET to blob)
https://github.com/openvinotoolkit/openvino/issues/7379

ibaiGorordo, 2021-09-14:
Also, HITNET looks nice, but it is quite slow. Currently, monocular depth estimation models (fastnet, Midas 2.1 small...) seem to be faster than the stereo ones (current ones are too complex with 3D convolutions and the cost volume estimation). But, I still have hope that there is somewhere some fast stereo model :monocle_face:

PINTO0309, 2021-09-21:
It looks like the issue I posted has been triaged and escalated to the development team.

Luxonis-Brandon, 2021-09-21:
Awesome - thanks!

gurbain, 2021-12-10:
> I was able to run real-time inference on TFLite HITNET Stereo depth estimation (middlebury) using OAK-D and having the inference on the host. Here are my results:
> 
> [image]

Hey,

Sorry for the spam but I am trying to reproduce the same example that you showed @nickjrz (stereo depth estimation on the host with an oak-d and hitnet) and I can't get as good results as you show. I actually started from the same project (https://github.com/ibaiGorordo/HITNET-Stereo-Depth-estimation) but it looks like my results are much worse than yours  (maybe the pre-processing?). Could you maybe provide a link to your code, it would be really interesting. Thank you!

Luxonis-Brandon, 2021-12-11:
@tersekmatija may be able to help advise too.

nickjrz, 2021-12-13:
> Hey,
>
> Sorry for the spam but I am trying to reproduce the same example that you showed @nickjrz (stereo depth estimation on the host with an oak-d and hitnet) and I can't get as good results as you [...]

Hey @gurbain,

Due to the nature of the project I am working on I can't really post any code as of now. Some advice is to make sure you have the right parameters for the DepthAI stereo camera you are using such as baseline and focal length. You can also look at your input tensor and make sure it matches the input parameters of the model.  I hope that helps!

PINTO0309, 2021-12-13:
https://github.com/ibaiGorordo/ONNX-HITNET-Stereo-Depth-estimation

PINTO0309, 2021-12-13:
https://github.com/ibaiGorordo/TFLite-HITNET-Stereo-depth-estimation

ibaiGorordo, 2021-12-13:
First, make sure you get the correct disparity map by passing the rectified images to the model. For the disparity yiu should not need any other changes. If the disparity map does not look good, there might be a problem with the rectified images, and you might need to calibrate the board. Does the depthai depth map from the library demo look good?

For the depth, check the depthai documentation on how to get the depth from disparity:
https://docs.luxonis.com/projects/api/en/latest/components/nodes/stereo_depth/#calculate-depth-using-dispairty-map

PINTO0309, 2021-12-13:
A very lightweight stereo depth estimation model. The conversion to OpenVINO was successful, but I am struggling with Myriad Blob because it does not support **`ExtractImagePatches`**. If there is an alternative way to standard operations, it may be possible to convert it. Any suggestions for replacing **`ExtractImagePatches`** with standard operations would be very welcome.

- ONNX, TFLite, OpenVINO (2MB - 11MB)
https://github.com/PINTO0309/PINTO_model_zoo/tree/main/202_stereoDNN

- Original Repo
https://github.com/NVIDIA-AI-IOT/redtail/tree/master/stereoDNN

- Paper
https://arxiv.org/pdf/1803.09719.pdf

- **`ExtractImagePatches`**
https://docs.openvino.ai/latest/openvino_docs_ops_movement_ExtractImagePatches_3.html
https://www.programcreek.com/python/?CodeExample=extract+patches

tersekmatija, 2021-12-14:
Also, a snippet of extract batches equivalent in TF: https://github.com/onnx/tensorflow-onnx/issues/436#issuecomment-993313423.

PINTO0309, 2021-12-14:
@tersekmatija Thank you. The output matched.

PINTO0309, 2021-12-14:
I have successfully replaced **`ExtractImagePatches`** with standard operations, but unfortunately I get an incomprehensible error when converting the subsequent **`Conv3D`** to Myriad Blob. The behavior of **`myriad_compile`** seems to be strange. :cry: 
```
/home/jenkins/agent/workspace/private-ci/ie/build-linux-ubuntu20/b/repos/openvino/inference-engine/src/vpu/graph_transformer/src/frontend/frontend.cpp:439 Failed to compile layer "model/conv3d_8/Conv3D": [ GENERAL_ERROR ] 
/home/jenkins/agent/workspace/private-ci/ie/build-linux-ubuntu20/b/repos/openvino/inference-engine/src/vpu/graph_transformer/src/stages/convolution.cpp:404 number of biases must equal to number of output channels per group, but: channels per group=32, biases=1
```

PINTO0309, 2022-01-06:
I have confirmed that the problem below with errors occurring during HITNet conversion is resolved in OpenVINO 2022.1. In fact, I was able to convert to OpenVINO IR.
https://github.com/luxonis/depthai/issues/173#issuecomment-918991375

However, when compiling to Myriad Blob, I encountered a new error, so I submitted a new issue again.
- [Bug] Const data got different desc and content byte sizes (24 and 96 respectively)" error when converting ConvolutionBackpropData using compile_tools #9517
https://github.com/openvinotoolkit/openvino/issues/9517

Luxonis-Brandon, 2022-04-12:
We wrote an implementation of the paper above and also a training solution for it.  Seems to be initially working and starting to train/converge OK-ish.
[image]
[image]
[image]
[image]
[image]
[image]

Luxonis-Brandon, 2022-04-20:
[image]

ecmnet, 2022-04-21:
> We wrote an implementation of the paper above and also a training solution for it. Seems to be initially working and starting to train/converge OK-ish.

Any code available already?

edgarriba, 2022-04-22:
@Luxonis-Brandon what model are you using in the end? And what's the time performance for the results you show here ? Will be doable to run side to other detection and segmentation networks?

garybradski, 2022-04-22:
Awesome. Of course, it would be nice to know how much of the camera compute resources are used by this in memory and time.

tersekmatija, 2022-04-22:
> Any code available already?

@ecmnet not yet, we are working on getting this out as soon as possible. We are doing a custom implementation of a model based on this paper: https://arxiv.org/pdf/2110.15367.pdf. I think the authors also link to their own implementation here: https://cvlab-unibo.github.io/neural-disparity-refinement-web/, but unfortunately you cannot run this directly on the camera. You can experiment on CPU / GPU. :smiley: 

> @Luxonis-Brandon what model are you using in the end? And what's the time performance for the results you show here ? Will be doable to run side to other detection and segmentation networks?

> Awesome. Of course, it would be nice to know how much of the camera compute resources are used by this in memory and time.

@garybradski @edgarriba The model itself is very heavy, especially the MLP heads at the end. We are working on a lighter version of the model that will be suitable for devices and will likely need a few more iterations before we release it to public. Our first goal is to have the model run on device, with performance that makes it practical. For the first few iterations I'd say it would not be possible to run other NNs in addition to this on Gen2, but we want to achieve this in the future. Not yet sure whether this will be possible or not. Images that @Luxonis-Brandon shared above are from our first implementation of the heavy model, but we are starting to see some results with our lighter version as well. We'll share once we have more! :rocket:

edgarriba, 2022-04-22:
@tersekmatija not sure how much are you planning to tweak the model but replacing mlp by any kind of separable convs might help to reduce the memory consumption. My approach would be, take a simple light unet style network, input 6xHxW (left/right rectified rbg and sgbm stereo) and output disparity which is a better representation. To compute depth you have the camera calibration.

ZlodeiBaal, 2022-05-22:
Just found this in the repo - https://github.com/luxonis/depthai-experiments/tree/master/gen2-crestereo-stereo-matching
Look nice! Pretty slow, but wow. It works even with glass!
@Luxonis-Brandon is this the experiment that you mentioned above?

tersekmatija, 2022-05-23:
Hey @ZlodeiBaal , that's a different model - [CREStereo](https://github.com/megvii-research/CREStereo), which does pretty good on the stereo data (I am appending some images below). 

We are investigating good practices and doing some experiments in the background.
[image]

[image]

gurbain, 2022-05-23:
Hi @ZlodeiBaal ! 

Reall nice, I think that CREStereo is one of the best model that I have tested and it is good to see it has been ported to the OAK-D. By "pretty slow", could you detail a bit more how long it takes approximately per image?

gurbain, 2022-05-23:
Thanks @Luxonis-Brandon! Did not see the FPS in the corner, my bad! :) Seems like a very good FPS given the CREStereo time performances indeed!

PINTO0309, 2022-05-23:
[rtstereonet_maxdisp192_720x1280.zip](https://github.com/luxonis/depthai/files/8755924/rtstereonet_maxdisp192_720x1280.zip)

Luxonis-Brandon, 2022-06-06:
https://twitter.com/nburrus/status/1528750927037046784?s=21&t=-1nO4bfsI7ZhImVwyWTpQw

cyberbeat, 2022-08-10:
> Some other neat ones: https://cvlab-unibo.github.io/neural-disparity-refinement-web/ https://arxiv.org/abs/2110.15367

@Luxonis-Brandon  Did you made some progress with this one? I did not find anything about speed of this? And would it be possible to make it smaller (replacing backbone)?

tersekmatija, 2022-08-10:
We did some extensive experiments on that one @cyberbeat . We replicated the code before it was released actually, but just replacing the backbone isn't enough. The 2 MLP heads at the end are pretty big, so the final model doesn't fit on device. So we've tried experimenting a bit with the two heads, but unfortunately they don't perform well enough when constrained as much as we needed them to be. We are constantly looking at different approaches and how to combine them.

justin-larking-pk, 2022-10-06:
Have you guys looked at RAFT-Stereo (https://github.com/princeton-vl/RAFT-Stereo)? I was looking at implementing a lite variant of this myself (like how the og RAFT optical flow model has RAFT-S). It appears to have good results.

tersekmatija, 2022-10-13:
@justin-larking-pk I don't think we tried exporting it, CREStereo mentioned above seems to get better results. But if you give it a try, do keep us updated! :)

kekeblom, 2022-10-30:
Tried adapting the method from [1] using their code (https://github.com/kevinleestone/mmstereo) to run onboard the device. As such, at 720P resolution, I was not able to fit it in memory and ran into some issues with model export. With a few changes to the model, I was able to export it and run it on the device. Took a bit of tweaking to get the compute to a reasonable level (640x400 resolution, 8x downsampling factor), which runs on the device at ~4 FPS, although the lower resolution and heavy downsampling does come at the cost of accuracy and considerable artifacts are starting to be visible. Trained it on FlyingThings, Middlebury and ETH3D.

Might give it a couple more shots to see if I can find a setting that would work a little bit better.

Here are a couple of results:
[image]
[image]
[image]

Anyone know of any tools or general resources that could help figure out the bottlenecks on the embedded hardware? 

[1] Krishna Shankar, Mark Tjersland, Jeremy Ma, Kevin Stone, and Max Bajracharya. A Learned Stereo Depth System for Robotic Manipulation in Homes. ICRA 2022

tersekmatija, 2022-11-02:
Nice @kekeblom! Do you mind sharing the repository/describing all the changes you've made in the repo? I think this definitely shows some promise and I believe the artifacts could be removed by further tweaking the architecture.

As for figuring out the bottlenecks on the embedded hardware - you can try and install openvino-dev. There should be a benchmark_app which you can use to benchmark the model on the device. Simply add `-d MYRIAD -report_type average_counters` flags when calling it. Note that the version should match the version of OpenVINO used to produce the blob.

You can also inspect the FLOPs and parameters of your architecture using fvcore or something similar beforehand. While more FLOPs in certain module does not always mean there could be a bottleneck, it's a good start and it's usually worth investigating those modules.

kekeblom, 2022-11-02:
Here is the code I used https://github.com/kekeblom/mmstereo. @tersekmatija 

Currently training one with 640x400 resolution, 4x downsampling and 128 disparities computed. This seems to yield quite a bit cleaner output. The settings can be found in the example config `config_sceneflow.yaml`. 

This one hasn't yet fully converged, but using an intermediate checkpoint, I get slightly better output: 
[image]

Admittedly some artifacts are still visible. 

It seems most of the compute is spent in computing and processing the cost volume. Trimming the convolutional network doesn't seem to change throughput, but reducing the size of the cost volume has a considerable impact. Can't think of any easy ways around that besides using lower resolution images. Maybe there exists a more efficient way to match features on this specific hardware. 

I think the sweet spot for this type of network might be running at a lower resolution, but might require some tuning to get rid of the artifacts, as the network is kind of designed for very high resolutions and close range scenes with large dilations.

themarpe, 2022-11-03:
Nice!

@tersekmatija @kekeblom thoughts on perhaps having a network that "validates / augments" depth instead of computing it from scratch? Maybe a L + R + Depth(or Disp) + Confidence Map -> Depth. It'd either recalculate low confidence points, or observe known areas/patterns which should be "filled" (eg flat surfaces, etc...).

Mostly thinking from performance standpoint, running on existing hardware, where the Stereo HW already does a lot of compute and to see if that can be further utilized, without having the network do it. (the whole matching again)

tersekmatija, 2022-11-08:
Nice @kekeblom!

And yeah, cost volumes are expensive. Currently I am not sure if there exist some fast way to compute/process them at arbitrary scales. This is why CREStereo and similar methods usually tackle the problem with RNNs rather than cost volumes.
I think that using smaller dilations could improve the artifacts on lower resolutions. But yeah, it is a challenging problem to do this efficiently with good accuracy.

Re depth/disparity filling - I think approaching this with CNNs might be hard, especially since the holes have dynamic shape and any conditional predictions are hard to do on edge accelerators. I think a more suitable approach would be something along the lines of [NDR](https://arxiv.org/pdf/2110.15367.pdf), but from my experience the MLPs there are the reason for a good performance. Making them lighter massively decreases the performance, but without doing this it's impossible to deploy them.

borongyuan, 2023-06-11:
Here is a TinyHITNet implementation that is much less computationally intensive. But the model conversion looks a bit problematic.
https://github.com/zjjMaiMai/TinyHITNet

themarpe, 2023-06-19:
Nice @borongyuan 

Gave this a quick try - does seem to work fairly well, but has issues recognizing certain objects every now and then.

Looking forward how far you'll be able to bring this!

borongyuan, 2023-06-20:
Hi @themarpe 
In fact, I think the performance is not good enough. Although the MACs of the model appear to be smaller, the U-Net architecture is inefficient on many hardware platforms. We need to look at other options. I also prefer methods of depth refinement/enhancement.

ulkesh11, 2023-11-11:
> Hey @ZlodeiBaal , that's a different model - [CREStereo](https://github.com/megvii-research/CREStereo), which does pretty good on the stereo data (I am appending some images below).
@tersekmatija 
Do you mind sharing the code for this? Would like to try it out

borongyuan, 2024-06-17:
Our new stereo depth solution, running on OAK-D Lite

https://github.com/luxonis/depthai/assets/22918715/681dc3c3-65ff-47c3-9a99-1a604e9cab11

tersekmatija, 2024-06-17:
Nice @borongyuan ! 

Curious how you are approaching this?

[7 more replies]