luxonis/depthai #174: Support 120 FPS

fuzzthink, 2020-08-12:
In the framerate FAQ (last question) in kickstarter, it says 120 FPS in stereo grayscale cameras is possible. This is very important feature for me, enabling this would be great!

Luxonis-Brandon, 2020-08-13:
Thanks @fuzzthink !

Luxonis-Brandon, 2020-08-13:
Oh I do have a question - do you need this for just getting that many images back, or for depth at this framerate?

Luxonis-Brandon, 2020-09-09:
Hi @fuzzthink ,

So 120 FPS on the grayscale cameras is now supported for 640x480 mode of the grayscale cameras (and 60fps for 1280x800 and 1280x720).  

`./depthai-demo.py -s left right -monof 120 -monor 400`

https://github.com/luxonis/depthai/blob/aaa06c460460daa0cd5acdb473d2026c87db4860/depthai_helpers/cli_utils.py#L132

[image]

Thoughts?

P.S.: I think 120FPS at 1280x800 is likely also implementable, but we have not yet investigated it.

Thanks,
Brandon

closed 2020-10-02 by Luxonis-Brandon