luxonis/depthai #181: save camera extrinsics in calibrate.py

2emoore4, 2020-08-19:
Currently, it's required that the developer knows and specifies the distances between the stereo cameras, as well as the color camera. The developer is expected to save these values in their custom board's JSON config.

These values can be inferred during the calibration process. cv.calibrateCamera(...) already returns the poses of the chessboards relative to the cameras (seems they get thrown out and only the homography matrix is saved). With a bit of simple math we can use these poses to find the distance between the cameras (not only the distances actually, but also the relative rotations).

Having a more accurate recording of distance between the cameras should help with stereo depth accuracy, as well as stereo triangulation accuracy.

Luxonis-Brandon, 2020-08-19:
Totally agreed.  We will add this capability for sure.  There are a couple other bits we will probably include as well.  We are making a list now and will consolidate.

Luxonis-Brandon, 2020-08-19:
So more details https://github.com/luxonis/depthai/issues/182 .

closed 2020-10-02 by Luxonis-Brandon