luxonis/depthai #182: save the camera's lens distortion coefficients during calibration

2emoore4, 2020-08-19:
I don't know a lot about stereo depth (so I may be wrong about this stuff), but I don't think that the homography matrix alone handles camera lens un-distortion.

calibrate.py uses cv.calibrateCamera(...) which returns the lens distortion parameters, but they are not saved along with the homography matrix.

Does the Myriad X chip support using these distortion coefficients during stereo matching? I would think that un-distorting the images should help with stereo depth quality/completeness.

Luxonis-Brandon, 2020-08-19:
Thanks.  @saching13 is actually looking into this.  To save these in EEPROM and make them requestable easily through the API for such purposes.

Luxonis-Brandon, 2020-08-19:
So the EEPROM is plenty big enough to store it I’m sure.  So at least:

- left/right rectification
- left intrinsic
- right intrinsic
- RGB intrinsic
- And probably extrinsics as well if possible between the cameras.

And then we should make all of these accessible easily via the API.

Luxonis-Brandon, 2020-09-11:
This is now implemented in https://github.com/luxonis/depthai/pull/190

closed 2020-10-02 by Luxonis-Brandon