luxonis/depthai #183: OpenCV Spatial AI Competition Feature Discovery and Requests

Luxonis-Brandon, 2020-08-19:
## 1. Saving JPEG images  
My understanding is this is a pre-canned example.
So this is actually supposed to be working out of the box.  It seems we broke it w/ some other feature.  Theoretically all you should need to do to get JPEGs is hit the c key when running the python3 test.py script.

## 2. Syncing various streams 
Agreed.  So now that we have all streams have the metadata, @Luxonis-Lukasz is writing examples for this and adding to docs.luxonis.com.
Is this example/how-to is sufficient?
Or should we have a mode and/or API endpoint where you can pull all of them and know they are pre-synched?

That seems like would be great idea.  In that case we could sync them in the C++ library first in that mode.  Or perhaps somehow done on Myriad X?

Thoughts on that?

And for all streams being supported for synch on the host, see here: https://github.com/luxonis/depthai/pull/178

## 3. Obtaining intrinsic params 
Thoughts on this list here?  Please do feel free to add there, or to respond here as well.

## 4. Readily available smooth point clouds 
(without user worrying about which intrinics to use and how to inverse project).

So this is an interesting one.  We actually didn’t think that folks would want to use this for point-cloud generation.  But we are learning now that a ton of folks do.  So yes definitely we will work on this.

Please share any other requests/feeback here as well from other users.

## 5. Out-of-box Stereo neural inference support.
Yes, totally agree.  In fact we planned to have working/clean examples of this prior to even announcing the competition.  So the following example is what we have now, but it needs to be improved:
https://github.com/luxonis/depthai-experiments/tree/master/triangulation-3D-visualizer
So on this one, we could do the matching and projection into 3D points on the Myriad X.  But there is such strong interest in additional additional filtering, rules, flexibility/etc. that we figured it was best for this to be done on the host to afford that flexibility.

Luxonis-Brandon, 2020-08-19:
So Jagadish Mahendran is who helped put the list above together.  And he also took the time to put together a proposal on this functionality below:
[oakd_high_level_architecture.pdf](https://github.com/luxonis/depthai/files/5099157/oakd_high_level_architecture.pdf)

Luxonis-Brandon, 2020-08-24:
As an update on 6 we got initial rectified output support for right and left stream.  Still some work to get these back to the host.

jaggiK, 2020-08-24:
Added `jpegout`but they are out of sync and less fps. This is because `jpegout` requests slows down entire pipeline reducing fps significantly. Hoping there will be a support to obtain seamless `jpegout` streams.

Luxonis-Brandon, 2020-09-04:
Thanks @jaggiK .  We will be adding RAW output for the RGB sensor as well.  Likely next week if all goes well.  This should solve the problem you are seeing.

Luxonis-Brandon, 2020-09-10:
Hi @jaggiK 

The `raw-color` branch in depthai now support `-s raw_color` which opens a window with decoded yuv420p coming from directly after ISP of color camera.  I updated this on the main issue above as well.

themarpe, 2020-09-10:
@Luxonis-Brandon @jaggiK 
Note: The raw-color branch was updated to rename the stream from `raw_color` to `color` as it is more inline with the operation of streams `left` and `right` (image frames after ISP processing)

Luxonis-Brandon, 2020-11-11:
subpixel, LR-check, and Extended disparity are now initially implemented:
https://github.com/luxonis/depthai/issues/163

Luxonis-Brandon, 2021-07-06:
All features in this grouping of requests has been implemented with the exception of generating the point cloud on DepthAI directly, which is now tracked separately in https://github.com/luxonis/depthai/issues/180.

closed 2021-07-06 by Luxonis-Brandon