luxonis/depthai #184: Hard to estimate surface normal

kota-rr, 2020-08-20:
Start with the Why:
Surface normals give us a lot of information on object shapes. However, from pointclouds obtained from current depth map, it is hard to estimate the surface normal. It is because depth resolution is not small enough so that the depth map expresses the detailed depth changes of the surface. 

This is because current OAK-D models have only 96 depth levels, hence the object surfaces of nearby pixels tend to have exactly the same depth, and the surface normals computed using the 3D positions of neighboring pixels tend to be almost near parallel to z axis unless smooth depth maps using filters with large karnels. 

[image]
[image]

The How:
Subpixel estimation of the depth increase the disparity resolution. That helps estimating surface normals even with small kernel filters.

The What:
This feature will not only help estimating surface normals, it might also help getting more precise object distances/positions.

Luxonis-Brandon, 2020-08-20:
Thanks @kota-rr .  Makes a ton of sense to us.  We will be working on sub-pixel soon once we figure out the bug we are having with LR-Check.

Much appreciate the feature request and appreciate following our feature request format!

Luxonis-Brandon, 2020-11-11:
So @kota-rr we now have initial subpixel support.  Notice there are way more points along the plane.

[image]
[image]

Here is the demo, adding the point cloud projector: https://github.com/luxonis/depthai-experiments/commit/e904ac58
Go to the repo depthai-experiments , check out the branch gen2_stereo and run:
```
cd gen2-camera-demo
python3 -m pip install -r requirements.txt
python3 main.py 
```
It is configured with LRcheck + Subpixel by default.
I will also add command line options, but for now please edit the script to change the config:
```
lrcheck  = True 
extended = False
subpixel = True 
```
Note that as of now there is no median filtering when any of the lrcheck, subpixel, or extended disparity are enabled.  So that is why it looks a bit grainy.
And there’s some error when doing the point cloud projection where a bunch of valid depth points (maybe 1/2 of them?) don’t get projected.

Luxonis-Brandon, 2020-11-11:
I'm going to try to project like the original image from @kota-rr .  Actually, @kota-rr - would you mind sharing the code you used to make those 3D views in Open3D?  That way we can use exactly the same settings for comparison.

[image]

Luxonis-Brandon, 2020-11-12:
Hi @kota-rr we got a similar view running.  The colors are currently reversed of the colorized depth, but it gives an idea on how much easier it is to do plane detection now.  CC:  @themarpe .

[image]
[image]

Luxonis-Brandon, 2020-11-12:
And note that we have some error where the point cloud projection is throwing away a bunch of valid depth points.  We're going to look into it.  But you can see already that the plane estimation should be WAY easier now.

closed 2021-01-11 by Luxonis-Brandon

ricardosantana-sensei, 2021-09-23:
Dear Brandon,

May I ask you why isn't it possible to set a stereo confidence threshold when subpixel is active?

I've tried changing the values inside stereo.initialConfig.setConfidenceThreshold() and nothing changes if subpixel is True. Otherwise, this property works properly.

szabi-luxonis, 2021-09-23:
> Dear @Luxonis-Brandon,
> 
> May I ask you why isn't it possible to set a stereo confidence threshold when subpixel is active?
> 
> I've tried changing the values inside stereo.initialConfig.setConfidenceThreshold() and nothing changes if subpixel is True. Otherwise, this property works properly.

Hello.
It's possible to change in the current `develop` branch, we are going to make a new release in the upcoming days.
If you are interested in more stereo configuration options you can check out [this](https://github.com/luxonis/depthai-python/blob/develop/examples/stereo_depth_from_host.py)  example.

ricardosantana-sensei, 2021-09-29:
I want to use the develop branch for the purpose of setting a confidence threshold when using subpixel. However, when trying to build from source following your instructions, I get the following error:

FileNotFoundError: [Errno 2] No such file or directory: '/home/rsantana/depthai-python/depthai-core/CMakeLists.txt'

Could you please help me? When do you expect to release this feature on the main repo?

Thanks in advance.

ricardosantana-sensei, 2021-09-29:
I want to use the develop branch for the purpose of setting a confidence threshold when using subpixel. However, when trying to build from source following your instructions, I get the following error:

FileNotFoundError: [Errno 2] No such file or directory: '/home/user/depthai-python/depthai-core/CMakeLists.txt'

Could you please help me? When do you expect to release this feature on the main repo?

Thanks in advance. @Luxonis-Brandon @szabi-luxonis

szabi-luxonis, 2021-09-29:
> I want to use the develop branch for the purpose of setting a confidence threshold when using subpixel. However, when trying to build from source following your instructions, I get the following [...]

Hello.
You can install current latest develop by running:
```
python3 -m pip install --user --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local "depthai==2.10.0.0.dev+388dc8f7178cbfe1e24fc03d5f190ea9671ce0aa"
```

At the moment we have some broken things in `develop` that we need to fix/revert before making a release. 
Hopefully next week we will release 2.11.0.0.

rsantana-sensei, 2021-10-08:
May I also ask you if you know why all my Point Clouds appear like a pyramid whose apex is the optical center of the camera. The points are distributed all over this volume and only when I zoom in I can recognize some of the surfaces in the image. However there is quite a lot of noise. Is the a fix to this issue? @Luxonis-Brandon @szabi-luxonis

Luxonis-Brandon, 2021-10-08:
Not sure.  Can you share what you are seeing here?  We have now heard similar from several customers (whereas the above test I did looks clean).  So perhaps there is some regression we didn't catch.

rsantana-sensei, 2021-11-02:
Perfect! Just one more question. How do I subscribe the camera output thread from a different script? I've been trying to do it but couldn't find a way to make it work. Thanks. @Luxonis-Brandon @szabi-luxonis

szabi-luxonis, 2021-11-02:
> Perfect! Just one more question. How do I subscribe the camera output thread from a different script? I've been trying to do it but couldn't find a way to make it work. Thanks. @Luxonis-Brandon @szabi-luxonis

For consuming camera frames you can check any of the examples, like this [one](https://github.com/luxonis/depthai-python/blob/main/examples/StereoDepth/depth_preview.py).

rsantana-sensei, 2021-11-02:
Yes, but I wanted to have the camera running and subscribe to their frames in another python file. I'd like to have a method that would give me the last captured frame without having to have the camera settings and configurations in the same file (just as when I subscribe to the webcam of my laptop). Is this possible? @szabi-luxonis

szabi-luxonis, 2021-11-02:
> Yes, but I wanted to have the camera running and subscribe to their frames in another python file. I'd like to have a method that would give me the last captured frame without having to have the [...]

If you are interested in communication between processes I suggest looking into inter-process communication concept, there are multiple methods, not specific to depthai library.

rsantana-sensei, 2021-11-11:
Hi, I'm working with OAK-D PoE and I want to connect my device. I'm able to ping it, however, when I run this code:

import cv2
import depthai as dai

pipeline = dai.Pipeline()

camRgb = pipeline.createColorCamera()

xoutRgb = pipeline.createXLinkOut()
xoutRgb.setStreamName("rgb")
camRgb.preview.link(xoutRgb.input)

device_info = dai.DeviceInfo()
device_info.state = dai.XLinkDeviceState.X_LINK_BOOTLOADER
device_info.desc.protocol = dai.XLinkProtocol.X_LINK_TCP_IP
device_info.desc.name = "169.254.1.222"

with dai.Device(pipeline, device_info) as device:
    qRgb = device.getOutputQueue(name="rgb", maxSize=4, blocking=False)
    while True:
        cv2.imshow("rgb", qRgb.get().getCvFrame())
        if cv2.waitKey(1) == ord('q'):
            break
            
specifying the correct address I got the following error message: X_LINK_DEVICE_NOT_FOUND.

Do you know what it might be?

Thanks! @szabi-luxonis @Luxonis-Brandon

themarpe, 2021-11-16:
@rsantana-sensei 

I'd suggest using default constructor to connect to the device in this case.
```
with dai.Device(pipeline) as device: 
   ...
```

We'll rework means of connecting to specific devices using IP soon