luxonis/depthai #187: Two homography to attain stereo normal case

saching13, 2020-09-02:
## Start with the `why`:
Current implementation uses single homography to align the right camera to the left camera to reach the stereo normal case by making the relative rotation between stereo camera's  near to zero.

However this method highly depends on the accuracy of the relative rotation that we have estimated during calibration and error-prone when the stereo cameras are far from stereo normal case. To overcome this issue we can use two homography which rotates both the camera's images to attain stereo normal case.
Apart from the two homography we also try to undistort the image before applying two thomography.

## The `How`:
To achieve this we use Bouguet's `cv2::stereoRectify()` or Hartley's algorithm `cv2:: stereoRectifyUncalibrated()` which uses 3D geometry and least square approach in estimating the required parameters to build two homographies to achieve the stereo normal case.

## The `what`
Achieving stereo normal case makes the epipoles to move to infinity and helps in simplifying matching for triangulation and intrun helping us in obtaining better depth map.

saching13, 2020-09-03:
@Luxonis-Brandon just adding you here

Luxonis-Brandon, 2020-09-03:
Thanks.  So actually I get auto-added to all new issues. Appreciate you looking out though.  :-)

Luxonis-Brandon, 2020-09-11:
This is implemented in PR https://github.com/luxonis/depthai/pull/190

Luxonis-Brandon, 2020-09-21:
Just as an update this has been merged to `develop` and is soaking there for eventual merge to `main`.

closed 2020-10-02 by Luxonis-Brandon