luxonis/depthai #195: pip install of requirements fails on RPI (openvino_tensor_representation_develop_merge)

rychardeh, 2020-09-18:
open3d is not supported on RPI (in fact no ARM builds on pypi) so I commented out the dependency in requirements.txt so that the remainder of the requirements could be installed by pip.

I also noticed the following:

```
$ pip install -r requirements.txt
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring opencv-python: markers 'platform_machine != "armv7l"' don't match your environment
...
```

I think it might be caused by missing dependencies [https://stackoverflow.com/questions/60096360/is-there-an-easy-way-to-install-opencv-on-raspberry-pi](url).  I have a specific build of opencv on my system already because I've previously installed openvino so that is probably why I could run without it.  I'm not sure what would have happened if the attempt to install opencv-python had succeeded - I wonder if it ought to be optional?

szabi-luxonis, 2020-09-19:
What you have seen @rychardeh is conditional installing of opencv for armv7l platforms.
See here: https://github.com/luxonis/depthai/blob/openvino_tensor_representation_develop_merge/requirements.txt#L2
```
opencv-python==4.2.0.34; platform_machine != "armv7l"
opencv-python==4.1.0.25; platform_machine == "armv7l"
```
Basically it means:
install `opencv-python==4.1.0.25` if platform is armv7l, else install opencv-python==4.2.0.34.
There are problems for newer opencv versions on RPI.

The open3d issue has been fixed in PR #196 by @saching13

rychardeh, 2020-09-19:
Thanks for the explanation and fix.  I wasn't aware of the version issues.  I'm actually using an openvino tweaked build (4.3.0-openvino-2020.3.0) within a virtualenv but not installed via pip.

Luxonis-Brandon, 2020-10-02:
And as an update we are implementing support for the latest OpenVINO now.

closed 2020-10-02 by Luxonis-Brandon