luxonis/depthai #210: Segmentation fault when requesting camera metadata for rectified_left/right

rychardeh, 2020-10-02:
On branch `openvino_tensor_representation_develop_merge`, calling `data_packet.getMetadata().getCameraName()` results in a Segmentation fault when used on streams produced by `rectified_left` or `rectified_right`.  Works fine on `left`, `right`, etc. as usual.

Luxonis-Brandon, 2020-10-02:
Thanks for the report @rychardeh - we'll get this fixed.

Luxonis-Brandon, 2020-10-02:
(As an update, we have been discussing internally on what is best here for handling this case and more that would come up.)

alex-luxonis, 2020-10-07:
Should be fixed on: https://github.com/luxonis/depthai/pull/221, https://github.com/luxonis/depthai-core/pull/12
Before, the instance number (camera ID) was not populated from device side, and was random.

The same issue was also present for `disparity` / `depth` streams. Now the field is set to match with the "dominant eye" camera (the one that the disparity frame pixels are matching/aligned with). That is:
- with `-mirror_rectified true` (default), `getCameraName` returns `right`.
- with `-mirror_rectified false`, `getCameraName` returns `left`.

closed 2020-10-15 by szabi-luxonis