luxonis/depthai #222: demo not working

abhatikar, 2020-10-08:
(cv_depthai) iotuser@iotuser-Blade:~/depthai$ git branch 
* develop
  main
(cv_depthai) iotuser@iotuser-Blade:~/depthai$ python3 depthai_demo.py  
Using depthai module from:  /home/iotuser/cv_depthai/lib/python3.6/site-packages/depthai.cpython-36m-x86_64-linux-gnu.so
Using Custom Calibration File: depthai.calib
WARNING: Depth calculation with raw output format is not supported! It's only supported for YOLO/mobilenet based NNs, disabling calc_dist_to_bb
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999987,   -0.001160,    0.004909,
    0.001150,    0.999997,    0.002119,
   -0.004912,   -0.002113,    0.999986,
  Rectification Rotation R2 (right):
    0.999988,   -0.003005,    0.003789,
    0.003013,    0.999993,   -0.002110,
   -0.003782,    0.002122,    0.999991,
  Calibration intrinsic matrix M1 (left):
  869.771057,    0.000000,  647.535095,
    0.000000,  871.483765,  418.473907,
    0.000000,    0.000000,    1.000000,
  Calibration intrinsic matrix M2 (right):
  869.137817,    0.000000,  644.909241,
    0.000000,  870.417175,  407.580444,
    0.000000,    0.000000,    1.000000,
  Calibration rotation matrix R:
    0.999998,    0.001861,    0.001134,
   -0.001866,    0.999989,    0.004226,
   -0.001126,   -0.004228,    0.999990,
  Calibration translation matrix T:
   -7.487395,
    0.022499,
   -0.028367,
  Calibration Distortion Coeff d1 (Left):
    0.106500,   43.963959,    0.000695,    0.002498,   23.748280,    0.125551,   42.827187,
   26.160654,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
  Calibration Distortion Coeff d2 (Right):
    2.987483,   69.966843,    0.000401,    0.001771,   14.714975,    3.052096,   68.236557,
   17.965635,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
['metaout', 'previewout']
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker', 'rectified_left', 'rectified_right', 'color']
CNN configurations read: /home/iotuser/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json
444
445
config_h2d json:
{"_board":{"calib_data":[0.9999873042106628,-0.001159933628514409,0.004909291863441467,0.0011495426297187805,0.999997079372406,0.0021188806276768446,-0.004911735188215971,-0.0021132102701812983,0.9999856948852539,0.999988317489624,-0.0030048189219087362,0.0037885296624153852,0.0030128289945423603,0.9999932050704956,-0.0021103511098772287,-0.00378216290846467,0.0021217407193034887,0.9999905824661255,869.7710571289063,0.0,647.5350952148438,0.0,871.4837646484375,418.4739074707031,0.0,0.0,1.0,869.1378173828125,0.0,644.9092407226563,0.0,870.4171752929688,407.5804443359375,0.0,0.0,1.0,0.999997615814209,0.0018608927493914962,0.001133509329520166,-0.001865667407400906,0.9999893307685852,0.00422582495957613,-0.0011256333673372865,-0.004227929748594761,0.9999904036521912,-7.4873948097229,0.02249852940440178,-0.02836654894053936,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.10649976134300232,43.963958740234375,0.0006948882946744561,0.002498225774616003,23.748279571533203,0.1255507916212082,42.827186584472656,26.160654067993164,0.0,0.0,0.0,0.0,0.0,0.0,2.987483263015747,69.96684265136719,0.00040063654887489974,0.0017713055713102221,14.714975357055664,3.052095651626587,68.23655700683594,17.965635299682617,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"mesh_left":[0.0],"mesh_right":[0.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"}]},"ai":{"NCEs":1,"blob0_size":13441536,"blob1_size":0,"calc_dist_to_bb":false,"camera_input":"rgb","cmx_slices":7,"keep_aspect_ratio":true,"num_stages":1,"shaves":7},"app":{"sync_video_meta_streams":false},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":true,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"confidence_threshold":0.5,"depth_limit_mm":10000,"lr_check":false,"median_kernel_size":7,"padding_factor":0.30000001192092896,"warp_rectify":{"edge_fill_color":0,"mirror_frame":true,"use_mesh":false}},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
size of input string json_config_obj to config_h2d is ->2376
size of json_config_obj that is expected to be sent to config_h2d is ->1048576
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1048576 bytes to config_h2d
!!! XLink write successful: config_h2d (1048576)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
Successfully opened stream host_capture with ID #0!
Creating observer stream host_capture: DONE.
Read: 13441536
Attempting to open stream inBlob
Successfully opened stream inBlob with ID #1!
Writing 13441536 bytes to inBlob
!!! XLink write successful: inBlob (13441536)
Closing stream inBlob: ...
Closing stream inBlob: DONE.
depthai: done sending Blob file /home/iotuser/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.blob
Attempting to open stream outBlob
Successfully opened stream outBlob with ID #2!
Closing stream outBlob: ...
Closing stream outBlob: DONE.
CNN input width: 300
CNN input height: 300
CNN input num channels: 3
CNN to depth bounding-box mapping: start(0, 0), max_size(0, 0)
Host stream start:metaout
Opening stream for read: metaout
Attempting to open stream metaout
Successfully opened stream metaout with ID #1!
Starting thread for stream: metaout
Started thread for stream: metaout
Host stream start:previewout
Opening stream for read: previewout
Attempting to open stream previewout
Successfully opened stream previewout with ID #2!
Starting thread for stream: previewout
depthai: INIT OK!
Started thread for stream: previewout
Traceback (most recent call last):
  File "depthai_demo.py", line 375, in <module>
    dai.startLoop()
  File "depthai_demo.py", line 362, in startLoop
    keypress_handler(self, key, stream_names)
  File "depthai_demo.py", line 148, in keypress_handler
    cam_l = depthai.CameraControl.CamId.LEFT
AttributeError: module 'depthai' has no attribute 'CameraControl'
=== New data in observer stream host_capture, size: 4
Writing 4 bytes to host_capture
!!! XLink write successful: host_capture (4)
Stopping threads: ...
E: [global] [    869285] [python3] addEvent:262	Condition failed: event->header.flags.bitField.ack != 1
E: [global] [    869285] [python3] addEventWithPerf:276	E: [global] [    869285] [python3] addEvent:262	Condition failed: event->header.flags.bitField.ack != 1
E: [global] [    869285] [python3] addEventWithPerf:276	 addEvent(event) method call failed with an error: 3
E: [global] [    869285] [python3] XLinkReadData:156	Condition failed: (addEventWithPerf(&event, &opTime))
Device get data failed: 7
Closing stream metaout: ...
E: [global] [    869285] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    869285] [Scheduler00Thr] sendEvents:1027	Event sending failed
Closing stream metaout: DONE.
Thread for metaout finished.
 addEvent(event) method call failed with an error: 3
E: [global] [    869285] [python3] XLinkReadData:156	Condition failed: (addEventWithPerf(&event, &opTime))
Device get data failed: 7
Closing stream previewout: ...
Closing stream previewout: DONE.
Thread for previewout finished.
E: [global] [    869285] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    869285] [Scheduler00Thr] sendEvents:1027	Event sending failed
Stopping threads: DONE 0.756s.
Closing all observer streams: ...
Closing stream host_capture: ...
Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    869285] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    869285] [Scheduler00Thr] sendEvents:1027	Event sending failed
Reseting: DONE.

szabi-luxonis, 2020-10-08:
You forgot to update Depthai python lib
Run:
 `./install_dependencies.sh`

szabi-luxonis, 2020-10-08:
We encountered some issues with our CI, the depthai libraries are not uploaded to our server. We are working on the fix.

szabi-luxonis, 2020-10-08:
Fixed. 
Please run:
`./install_dependencies.sh`

closed 2020-10-08 by abhatikar

pablogs9, 2020-10-15:
Hello, I'm facing this same issue. My workflow is:

```bash
# from https://docs.luxonis.com/api/#install
curl -fL http://docs.luxonis.com/install_dependencies.sh | bash

echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger

python3 -m pip install depthai

git clone https://github.com/luxonis/depthai.git
cd depthai
git checkout develop

python3 -m pip install -r requirements.txt
. install_requirements.sh
```

Then I run the demo app:

```bash
user@user:/workspaces/develop_luxonis/depthai$ python3 depthai_demo.py 
Using depthai module from:  /home/vscode/.local/lib/python3.8/site-packages/depthai.cpython-38-x86_64-linux-gnu.so
Depthai version installed:  0.2.0.1+f91a36fac86a0f4383e863e04cb651869a1603ac
Depthai version required:   0.2.0.1+f91a36fac86a0f4383e863e04cb651869a1603ac
No calibration file. Using Calibration Defaults.
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v3)
  Board name     : BW1092
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 4 cm
  L-RGB distance : 1 cm
  L/R swapped    : yes
  L/R crop region: center
  Calibration homography right to left (legacy, please consider recalibrating):
    1.004923,    0.000706,   -5.708157,
   -0.003378,    0.998806,   12.638306,
    0.000005,   -0.000007,    1.000000,
['metaout', 'previewout']
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker', 'rectified_left', 'rectified_right', 'color']
/workspaces/develop_luxonis/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json
depthai: Calibration file is not specified, will use default setting;
config_h2d json:
{"_board":{"calib_data":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"mesh_left":[0.0],"mesh_right":[0.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"}]},"ai":{"NCEs":1,"NN_config":{"NN_family":"mobilenet","confidence_threshold":0.5,"output_format":"detection"},"blob0_size":14489664,"blob1_size":0,"calc_dist_to_bb":true,"camera_input":"rgb","cmx_slices":7,"keep_aspect_ratio":true,"num_stages":1,"shaves":7},"app":{"sync_video_meta_streams":false},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":false,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"depth_limit_mm":10000,"lr_check":false,"median_kernel_size":7,"padding_factor":0.30000001192092896,"warp_rectify":{"edge_fill_color":0,"mirror_frame":true,"use_mesh":false}},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
size of input string json_config_obj to config_h2d is ->1580
size of json_config_obj that is expected to be sent to config_h2d is ->1048576
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1048576 bytes to config_h2d
!!! XLink write successful: config_h2d (1048576)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
Successfully opened stream host_capture with ID #0!
Creating observer stream host_capture: DONE.
Read: 14489664
Attempting to open stream inBlob
Successfully opened stream inBlob with ID #1!
[... 56 lines cut ...]
Closing stream metaout: ...
E: [global] [    665019] [Scheduler00Thr] dispatcherEventSend:53        Write failed (header) (err -4) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    665019] [Scheduler00Thr] sendEvents:1027        Event sending failed
Closing stream metaout: DONE.
Thread for metaout finished.
watchdog triggered 
=== New data in observer stream host_capture, size: 4
Writing 4 bytes to host_capture
!!! XLink write successful: host_capture (4)
E: [global] [    671127] [Scheduler00Thr] dispatcherEventSend:53        Write failed (header) (err -4) | event XLINK_WRITE_REQ

Stopping threads: ...
E: [xLink] [    671127] [Scheduler00Thr] sendEvents:1027        Stopping threads: DONE 0.000s.
Closing all observer streams: ...
Event sending failedClosing stream host_capture: ...

Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    671128] [Scheduler00Thr] dispatcherEventSend:53        Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    671128] [Scheduler00Thr] sendEvents:1027        Event sending failed
Reseting: DONE.
```

alex-luxonis, 2020-10-15:
Hello @pablogs9 ,

There is an errata for BW1092, and it may be possible that the connection is limited to USB2 only. Could you check if that's the case by running this command in a separate terminal `dmesg -w | grep -i speed -A 1`

If the connection is USB3, it should print like (notice the `SuperSpeed` for `idProduct=f63b`):

```
[305272.482342] usb 1-1: new high-speed USB device number 80 using xhci_hcd
[305272.631003] usb 1-1: New USB device found, idVendor=03e7, idProduct=2485
--
[305273.526431] usb 2-1: new SuperSpeed USB device number 57 using xhci_hcd
[305273.550729] usb 2-1: New USB device found, idVendor=03e7, idProduct=f63b
```

But if it is USB2, it should look like:
```
305510.284405] usb 1-1: new high-speed USB device number 86 using xhci_hcd
[305510.433194] usb 1-1: New USB device found, idVendor=03e7, idProduct=2485
--
[305511.284307] usb 1-1: new high-speed USB device number 87 using xhci_hcd
[305511.433280] usb 1-1: New USB device found, idVendor=03e7, idProduct=f63b
```

Also do you use by chance a type-C USB port on the host machine? If it's the case, could you try switching to a type-A port?

And one more option to try: `python3 depthai_demo.py -fusb2`
`-fusb2`/`--force_usb2` limits the connection to USB2 only from firmware, and it may help in some cases, even if the cable/board connection is already USB2.

pablogs9, 2020-10-15:
Solved! Thank you so much.

With the `dmesg` I found that I'm using USB2. So with a type-A cable and `-fusb2` it works as expected.

alex-luxonis, 2020-10-15:
Great!

It may also be possible to still get an USB3 connection with the current revisions of BW1092, if the type-C cable connector is being slowly plugged in into the board, so the power pins will make contact first, and data pins will make contact after a short time (best to have a delay between power and data contact greater than 0.5 seconds). But it may work only with a given orientation/side of the type-C cable plug, so both sides should be tried, and could only work with a type-A port on the host (not type-C).

First when the USB is detected, it will always appear as USB2 in dmesg (that is the BootROM USB device `03e7:2485`), but when depthai_demo.py is started, it can be checked if the device was enumerated in USB2 or USB3 mode (`03e7:f63b`). If USB3 was detected, it should persist for multiple restarts of the app, until a board power cycle.

If trying this, make sure to remove the `-fusb2` option.

pablogs9, 2020-10-15:
Love that super hacky solution!! I'll try it. Thanks a lot.

pablogs9, 2020-10-15:
One completely off-topic question here: does the BW1092 have relays? I hear some switching sound when the board is loading.

Are your schematics open sourced for both the BW1092 and the SoM?

Luxonis-Brandon, 2020-10-15:
The BW1092 hardware design is open-sourced, yes:
[https://github.com/luxonis/depthai-hardware/tree/master/BW1092_ESP32_Embedded_WIFI_BT](https://github.com/luxonis/depthai-hardware/tree/master/BW1092_ESP32_Embedded_WIFI_BT)

And the clicking you hear is actually the auto-focus of the color/RGB camera.  When it initializes and some times when it focuses, it will test the focal extents - so this is the lens hitting the camera body.

The SoM is not open-sourced as we are not allowed to open source any design/code/etc. files that directly touch the Myriad X itself (as it is considered proprietary).  This constraint is one of the reasons we made the SoM - so that we could open source all the rest.

Luxonis-Brian, 2020-10-15:
Pablo,

What you hear is the VCM of the RGB camera. When it's powered, the
auto-focus VCM activates and "clicks". It's totally normal and we use that
sound internally as a sign of proper operation!

Best of luck!

*Brian Weinstein*
Luxonis Holding Corporation
1-970-593-8043
brian@luxonis.com

pablogs9, 2020-10-15:
Thanks @Luxonis-Brandon!