luxonis/depthai #223: depth_raw error with demo app

abhatikar, 2020-10-08:
```
(cv_depthai) iotuser@iotuser-Blade:~/depthai$ python3 depthai_demo.py -s metaout previewout depth_raw -ff -bb
Using depthai module from:  /home/iotuser/cv_depthai/lib/python3.6/site-packages/depthai.cpython-36m-x86_64-linux-gnu.so
Using Custom Calibration File: depthai.calib
Using Arguments= {'config_overwrite': None, 'board': None, 'shaves': None, 'cmx_slices': None, 'NN_engines': None, 'rgb_resolution': 1080, 'rgb_fps': 30.0, 'mono_resolution': 720, 'mono_fps': 30.0, 'disparity_confidence_threshold': 200, 'field_of_view': 71.86, 'rgb_field_of_view': 68.7938, 'baseline': 9.0, 'rgb_baseline': 2.0, 'swap_lr': True, 'store_eeprom': False, 'clear_eeprom': False, 'override_eeprom': False, 'device_id': '', 'dev_debug': None, 'force_usb2': None, 'cnn_model': 'mobilenet-ssd', 'cnn_model2': '', 'cnn_camera': 'rgb', 'disable_depth': False, 'draw_bb_depth': True, 'full_fov_nn': True, 'sync_video_meta': False, 'streams': [{'name': 'metaout'}, {'name': 'previewout'}, {'name': 'depth_raw'}], 'video': None}
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999987,   -0.001160,    0.004909,
    0.001150,    0.999997,    0.002119,
   -0.004912,   -0.002113,    0.999986,
  Rectification Rotation R2 (right):
    0.999988,   -0.003005,    0.003789,
    0.003013,    0.999993,   -0.002110,
   -0.003782,    0.002122,    0.999991,
  Calibration intrinsic matrix M1 (left):
  869.771057,    0.000000,  647.535095,
    0.000000,  871.483765,  418.473907,
    0.000000,    0.000000,    1.000000,
  Calibration intrinsic matrix M2 (right):
  869.137817,    0.000000,  644.909241,
    0.000000,  870.417175,  407.580444,
    0.000000,    0.000000,    1.000000,
  Calibration rotation matrix R:
    0.999998,    0.001861,    0.001134,
   -0.001866,    0.999989,    0.004226,
   -0.001126,   -0.004228,    0.999990,
  Calibration translation matrix T:
   -7.487395,
    0.022499,
   -0.028367,
  Calibration Distortion Coeff d1 (Left):
    0.106500,   43.963959,    0.000695,    0.002498,   23.748280,    0.125551,   42.827187,
   26.160654,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
  Calibration Distortion Coeff d2 (Right):
    2.987483,   69.966843,    0.000401,    0.001771,   14.714975,    3.052096,   68.236557,
   17.965635,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker', 'rectified_left', 'rectified_right', 'color']
CNN configurations read: /home/iotuser/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd_depth.json
444
445
config_h2d json:
{"_board":{"calib_data":[0.9999873042106628,-0.001159933628514409,0.004909291863441467,0.0011495426297187805,0.999997079372406,0.0021188806276768446,-0.004911735188215971,-0.0021132102701812983,0.9999856948852539,0.999988317489624,-0.0030048189219087362,0.0037885296624153852,0.0030128289945423603,0.9999932050704956,-0.0021103511098772287,-0.00378216290846467,0.0021217407193034887,0.9999905824661255,869.7710571289063,0.0,647.5350952148438,0.0,871.4837646484375,418.4739074707031,0.0,0.0,1.0,869.1378173828125,0.0,644.9092407226563,0.0,870.4171752929688,407.5804443359375,0.0,0.0,1.0,0.999997615814209,0.0018608927493914962,0.001133509329520166,-0.001865667407400906,0.9999893307685852,0.00422582495957613,-0.0011256333673372865,-0.004227929748594761,0.9999904036521912,-7.4873948097229,0.02249852940440178,-0.02836654894053936,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.10649976134300232,43.963958740234375,0.0006948882946744561,0.002498225774616003,23.748279571533203,0.1255507916212082,42.827186584472656,26.160654067993164,0.0,0.0,0.0,0.0,0.0,0.0,2.987483263015747,69.96684265136719,0.00040063654887489974,0.0017713055713102221,14.714975357055664,3.052095651626587,68.23655700683594,17.965635299682617,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"mesh_left":[0.0],"mesh_right":[0.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"},{"name":"depth_raw"}]},"ai":{"NCEs":1,"blob0_size":13441536,"blob1_size":0,"calc_dist_to_bb":true,"camera_input":"rgb","cmx_slices":7,"keep_aspect_ratio":false,"num_stages":1,"shaves":7},"app":{"sync_video_meta_streams":false},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":true,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"confidence_threshold":0.5,"depth_limit_mm":10000,"lr_check":false,"median_kernel_size":7,"padding_factor":0.30000001192092896,"warp_rectify":{"edge_fill_color":-1,"mirror_frame":true,"use_mesh":false}},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
size of input string json_config_obj to config_h2d is ->2398
size of json_config_obj that is expected to be sent to config_h2d is ->1048576
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1048576 bytes to config_h2d
!!! XLink write successful: config_h2d (1048576)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
Successfully opened stream host_capture with ID #0!
[... 29 lines cut ...]
Failed to open stream inBlob ! Retrying ...
Stream not opened: inBlob
depthai: pipelineConfig write error: Blob size too big: 13441536
Error: device does not provide stream: depth_raw
Pipeline is not created.
=== New data in observer stream host_capture, size: 4
Writing 4 bytes to host_capture
!!! XLink write successful: host_capture (4)
E: [global] [    681532] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_WRITE_REQ

Stopping threads: ...
Stopping threads: DONE 0.000s.
Closing all observer streams: ...
Closing stream host_capture: ...
E: [xLink] [    681532] [Scheduler00Thr] sendEvents:1027	Event sending failed
Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    681532] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    681532] [Scheduler00Thr] sendEvents:1027	Event sending failed
Reseting: DONE.
(cv_depthai) iotuser@iotuser-Blade:~/depthai$ 

```

Luxonis-Brandon, 2020-10-08:
Hi @abhatikar ,

Sorry about the trouble.

So we relabeled the CLI for `depth_raw` to `depth` to match the naming conventions of other streams.  So please use this command instead:

`python3 depthai_demo.py -s metaout previewout depth -ff -bb`

And if you don't mind, where did you see the reference to depth_raw?  If that is still in our documentation somewhere I'd like to get it fixed so others don't run into this.

Thanks,
Brandon

abhatikar, 2020-10-08:
Thanks Brandon. https://docs.luxonis.com/faq/

abhatikar, 2020-10-08:
```
(cv_depthai) iotuser@iotuser-Blade:~/depthai$ python3 depthai_demo.py -cnn mobilenet-ssd -s metaout previewout depth -ff -bb
Using depthai module from:  /home/iotuser/cv_depthai/lib/python3.6/site-packages/depthai.cpython-36m-x86_64-linux-gnu.so
Using Custom Calibration File: depthai.calib
WARNING: Depth calculation with raw output format is not supported! It's only supported for YOLO/mobilenet based NNs, disabling calc_dist_to_bb
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999987,   -0.001160,    0.004909,
    0.001150,    0.999997,    0.002119,
   -0.004912,   -0.002113,    0.999986,
  Rectification Rotation R2 (right):
    0.999988,   -0.003005,    0.003789,
    0.003013,    0.999993,   -0.002110,
   -0.003782,    0.002122,    0.999991,
  Calibration intrinsic matrix M1 (left):
  869.771057,    0.000000,  647.535095,
    0.000000,  871.483765,  418.473907,
    0.000000,    0.000000,    1.000000,
  Calibration intrinsic matrix M2 (right):
  869.137817,    0.000000,  644.909241,
    0.000000,  870.417175,  407.580444,
    0.000000,    0.000000,    1.000000,
  Calibration rotation matrix R:
    0.999998,    0.001861,    0.001134,
   -0.001866,    0.999989,    0.004226,
   -0.001126,   -0.004228,    0.999990,
  Calibration translation matrix T:
   -7.487395,
    0.022499,
   -0.028367,
  Calibration Distortion Coeff d1 (Left):
    0.106500,   43.963959,    0.000695,    0.002498,   23.748280,    0.125551,   42.827187,
   26.160654,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
  Calibration Distortion Coeff d2 (Right):
    2.987483,   69.966843,    0.000401,    0.001771,   14.714975,    3.052096,   68.236557,
   17.965635,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
['metaout', 'previewout', 'depth']
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker', 'rectified_left', 'rectified_right', 'color']
CNN configurations read: /home/iotuser/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json
444
445
config_h2d json:
{"_board":{"calib_data":[0.9999873042106628,-0.001159933628514409,0.004909291863441467,0.0011495426297187805,0.999997079372406,0.0021188806276768446,-0.004911735188215971,-0.0021132102701812983,0.9999856948852539,0.999988317489624,-0.0030048189219087362,0.0037885296624153852,0.0030128289945423603,0.9999932050704956,-0.0021103511098772287,-0.00378216290846467,0.0021217407193034887,0.9999905824661255,869.7710571289063,0.0,647.5350952148438,0.0,871.4837646484375,418.4739074707031,0.0,0.0,1.0,869.1378173828125,0.0,644.9092407226563,0.0,870.4171752929688,407.5804443359375,0.0,0.0,1.0,0.999997615814209,0.0018608927493914962,0.001133509329520166,-0.001865667407400906,0.9999893307685852,0.00422582495957613,-0.0011256333673372865,-0.004227929748594761,0.9999904036521912,-7.4873948097229,0.02249852940440178,-0.02836654894053936,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.10649976134300232,43.963958740234375,0.0006948882946744561,0.002498225774616003,23.748279571533203,0.1255507916212082,42.827186584472656,26.160654067993164,0.0,0.0,0.0,0.0,0.0,0.0,2.987483263015747,69.96684265136719,0.00040063654887489974,0.0017713055713102221,14.714975357055664,3.052095651626587,68.23655700683594,17.965635299682617,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"mesh_left":[0.0],"mesh_right":[0.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"},{"data_type":"uint16","name":"depth"}]},"ai":{"NCEs":1,"blob0_size":13441536,"blob1_size":0,"calc_dist_to_bb":false,"camera_input":"rgb","cmx_slices":7,"keep_aspect_ratio":false,"num_stages":1,"shaves":7},"app":{"sync_video_meta_streams":false},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":true,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"confidence_threshold":0.5,"depth_limit_mm":10000,"lr_check":false,"median_kernel_size":7,"padding_factor":0.30000001192092896,"warp_rectify":{"edge_fill_color":0,"mirror_frame":true,"use_mesh":false}},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
size of input string json_config_obj to config_h2d is ->2415
size of json_config_obj that is expected to be sent to config_h2d is ->1048576
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1048576 bytes to config_h2d
!!! XLink write successful: config_h2d (1048576)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
[... cut ...]
Writing 1 bytes to host_capture
!!! XLink write successful: host_capture (1)
Traceback (most recent call last):
  File "depthai_demo.py", line 375, in <module>
    dai.startLoop()
  File "depthai_demo.py", line 362, in startLoop
    keypress_handler(self, key, stream_names)
  File "depthai_demo.py", line 148, in keypress_handler
    cam_l = depthai.CameraControl.CamId.LEFT
AttributeError: module 'depthai' has no attribute 'CameraControl'
malloc_consolidate(): invalid chunk size
Aborted (core dumped)

```

abhatikar, 2020-10-08:
Brandon,

this is develop branch

```
(cv_depthai) iotuser@iotuser-Blade:~/depthai$ python3 depthai_demo.py -s metaout previewout depth -ff -bb
Using depthai module from:  /home/iotuser/cv_depthai/lib/python3.6/site-packages/depthai.cpython-36m-x86_64-linux-gnu.so
Using Custom Calibration File: depthai.calib
WARNING: Depth calculation with raw output format is not supported! It's only supported for YOLO/mobilenet based NNs, disabling calc_dist_to_bb
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Loading config file
watchdog started Attempting to open stream config_d2h

Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999987,   -0.001160,    0.004909,
    0.001150,    0.999997,    0.002119,
   -0.004912,   -0.002113,    0.999986,
  Rectification Rotation R2 (right):
    0.999988,   -0.003005,    0.003789,
    0.003013,    0.999993,   -0.002110,
   -0.003782,    0.002122,    0.999991,
  Calibration intrinsic matrix M1 (left):
  869.771057,    0.000000,  647.535095,
    0.000000,  871.483765,  418.473907,
    0.000000,    0.000000,    1.000000,
  Calibration intrinsic matrix M2 (right):
  869.137817,    0.000000,  644.909241,
    0.000000,  870.417175,  407.580444,
    0.000000,    0.000000,    1.000000,
  Calibration rotation matrix R:
    0.999998,    0.001861,    0.001134,
   -0.001866,    0.999989,    0.004226,
   -0.001126,   -0.004228,    0.999990,
  Calibration translation matrix T:
   -7.487395,
    0.022499,
   -0.028367,
  Calibration Distortion Coeff d1 (Left):
    0.106500,   43.963959,    0.000695,    0.002498,   23.748280,    0.125551,   42.827187,
   26.160654,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
  Calibration Distortion Coeff d2 (Right):
    2.987483,   69.966843,    0.000401,    0.001771,   14.714975,    3.052096,   68.236557,
   17.965635,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
['metaout', 'previewout', 'depth']
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker', 'rectified_left', 'rectified_right', 'color']
CNN configurations read: /home/iotuser/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json
444
445
config_h2d json:
{"_board":{"calib_data":[0.9999873042106628,-0.001159933628514409,0.004909291863441467,0.0011495426297187805,0.999997079372406,0.0021188806276768446,-0.004911735188215971,-0.0021132102701812983,0.9999856948852539,0.999988317489624,-0.0030048189219087362,0.0037885296624153852,0.0030128289945423603,0.9999932050704956,-0.0021103511098772287,-0.00378216290846467,0.0021217407193034887,0.9999905824661255,869.7710571289063,0.0,647.5350952148438,0.0,871.4837646484375,418.4739074707031,0.0,0.0,1.0,869.1378173828125,0.0,644.9092407226563,0.0,870.4171752929688,407.5804443359375,0.0,0.0,1.0,0.999997615814209,0.0018608927493914962,0.001133509329520166,-0.001865667407400906,0.9999893307685852,0.00422582495957613,-0.0011256333673372865,-0.004227929748594761,0.9999904036521912,-7.4873948097229,0.02249852940440178,-0.02836654894053936,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.10649976134300232,43.963958740234375,0.0006948882946744561,0.002498225774616003,23.748279571533203,0.1255507916212082,42.827186584472656,26.160654067993164,0.0,0.0,0.0,0.0,0.0,0.0,2.987483263015747,69.96684265136719,0.00040063654887489974,0.0017713055713102221,14.714975357055664,3.052095651626587,68.23655700683594,17.965635299682617,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"mesh_left":[0.0],"mesh_right":[0.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"},{"data_type":"uint16","name":"depth"}]},"ai":{"NCEs":1,"blob0_size":13441536,"blob1_size":0,"calc_dist_to_bb":false,"camera_input":"rgb","cmx_slices":7,"keep_aspect_ratio":false,"num_stages":1,"shaves":7},"app":{"sync_video_meta_streams":false},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":true,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"confidence_threshold":0.5,"depth_limit_mm":10000,"lr_check":false,"median_kernel_size":7,"padding_factor":0.30000001192092896,"warp_rectify":{"edge_fill_color":0,"mirror_frame":true,"use_mesh":false}},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
size of input string json_config_obj to config_h2d is ->2415
size of json_config_obj that is expected to be sent to config_h2d is ->1048576
Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1048576 bytes to config_h2d
!!! XLink write successful: config_h2d (1048576)
[... 29 lines cut ...]
Opening stream for read: metaout
Attempting to open stream metaout
Successfully opened stream metaout with ID #2!
[... cut ...]
=== New data in observer stream host_capture, size: 1
Writing 1 bytes to host_capture
!!! XLink write successful: host_capture (1)
Traceback (most recent call last):
  File "depthai_demo.py", line 375, in <module>
    dai.startLoop()
  File "depthai_demo.py", line 362, in startLoop
    keypress_handler(self, key, stream_names)
  File "depthai_demo.py", line 148, in keypress_handler
    cam_l = depthai.CameraControl.CamId.LEFT
AttributeError: module 'depthai' has no attribute 'CameraControl'

```

Luxonis-Brandon, 2020-10-08:
So @abhatikar 

It looks like you are partially using the develop branch, and partially using main.  And the code are not cross-compatible.

Can you do the following to update?

[https://docs.luxonis.com/api/#usingtesting-a-specific-branchpr](https://docs.luxonis.com/api/#usingtesting-a-specific-branchpr)

Luxonis-Brandon, 2020-10-08:
Probably worth doing a `git pull` as well before running:
```
python3 -m pip install -U pip
python3 -m pip install -r requirements.txt
```

This should take care of the incompatibility issue you are seeing.  I'll test develop real quick just to double-check.

Luxonis-Brandon, 2020-10-08:
Yes, develop works for me:
[image]

I also just noticed that a difference between we are running the script is:
 - I am running from the local repository with `./depthai_demo.py -s metaout previewout depth -ff -bb` and you are running from your site packages with `depthai_demo.py -s metaout previewout depth -ff -bb`.

So likely the site package needs to be upgraded to develop as well in that case.  Looking how to do that.

Luxonis-Brandon, 2020-10-08:
Oh, sorry, disregard - you had simply added depthai_demp.py to your path I think.  Anyway, the [above](https://github.com/luxonis/depthai/issues/223#issuecomment-705631320) should fix it.  If not please do ping us.  :-)

closed 2020-10-08 by abhatikar