luxonis/depthai #226: Use of full FOV with rectified images causes sporadic scaling issues with image/NN output

rychardeh, 2020-10-09:
On develop branch, to reproduce:  
```
python3 depthai_demo.py -cnn openpose2 -sh 14 -cmx 14 -nce 2 -cam rectified_left_right --mono_resolution 400 --mono_fps 2 -ff
```
Note that both openpose and openpose2 are missing their respective json config files.  However, I have reproduced this behaviour with my own code where I do provide the json config so I do not think this is the issue.

It appears that the image being used for neural inference is being transformed sporadically resulting in shifting NN output.

If you use `openpose` instead of `openpose2` the transform switches to be vertical rather than horizontal!

Luxonis-Brandon, 2020-10-09:
Thanks for the report.  We'll take a look.

Luxonis-Brandon, 2020-10-12:
Was just discussing with @alex-luxonis offline and we are struggling to reproduce this reliably.  We are able to sometimes, but then it will run forever without happening.  So we haven't been able to root-cause yet.

Any times @rychardeh on how to get it to occur quickly?

alex-luxonis, 2020-10-12:
I continued to debug it, and actually found the cause, an uninitialized internal metadata field for the rectified frames. Will push a patch for this issue.

rychardeh, 2020-10-12:
Problem solved.  Thanks!

Luxonis-Brandon, 2020-10-12:
Awesome thanks for finding and reporting the bug!  Will close this issue now as this was just merged to `develop` and will be going to `main` soon.

closed 2020-10-12 by Luxonis-Brandon