luxonis/depthai #247: Edge Detection

Luxonis-Brandon, 2020-10-20:
### Start with the `why`:

Edge detection is a useful building block in many CV/AI pipelines and the DepthAI hardware has hardware acceleration for this feature.

### Move to the `how`:

Implement the edge detection hardware acceleration as a node in the Gen2 pipeline builder (https://github.com/luxonis/depthai/issues/136).

### Move to the `what`:

Support hardware-accelerated edge detection as a pipeline node.

matanster, 2021-04-08:
I have seen this feature mentioned now on [the store](https://store.opencv.ai/products/oak-d). So is it really in yet?

It says there:

[image]

Luxonis-Brandon, 2021-04-08:
Hi @matanster ,

Sorry about these not being implemented yet.  We planned to have these out in December, but have been behind.  

This is among 10 features we planned to release in December in conjunction with our bulk shipment of devices, but have not been finished yet.  

For these 10 not-yet-started features, and also then others that are in-progress and finished, see here:
https://github.com/orgs/luxonis/projects/2

Anyway, we're more than happy to issue a full refund on the device here if you would like.  I'm not immediately sure on the timing to get these out.  Right now we're heads-down on fixing a subpixel crash (https://github.com/luxonis/depthai-python/pull/147) that can occur in some cases and getting full-rate IMU support out (in addition to Gen2 calibration improvements, and scripting capability https://github.com/luxonis/depthai/issues/207#issuecomment-813478224).

Thoughts?

Sorry about this not being out yet.  

-Brandon

Luxonis-Brandon, 2021-04-08:
Oh and if you'd like a re, please shoot

Luxonis-Brandon, 2021-04-08:
(Accidentally hit the close button, just reopened.)

matanster, 2021-04-09:
All cool.

Luxonis-Brandon, 2021-04-09:
Hi @matanster ,

So these are always famous last words, but I think we should be able to have both of these at least initially released this month, if all goes well:

 - Edge Detector (Canny Filter, specifically).
 - Harris Filter (Corner detector).

Thoughts?

Thanks,
Brandon

matanster, 2021-04-10:
I would by no means re-prioritize the workflow just due to one Github question ;-)

dhruvsheth-ai, 2021-04-10:
Hi @matanster @Luxonis-Brandon I published an unofficial example of Edge Detection using Harris filtering today, so you can try it out - 
https://github.com/dhruvsheth-ai/gen2-api-depth-experiments/tree/main/edge-harris

I'll be adding a demo for canny edge detection too, this week.

Also, in case you face any issue, you can open an issue on the repo or here. 

Thanks!

Luxonis-Brandon, 2021-04-12:
> I would by no means re-prioritize the workflow just due to one Github question ;-)

Sounds good.  Actually we ran into an issue with the latest Raspberry Pi OS that we need to debug (https://github.com/luxonis/depthai/issues/354).  So will jump on that first.

And thanks @dhruvsheth-ai !

Luxonis-Brandon, 2021-06-01:
We're still working on this as a firmware module, but in the meantime Kunal implemented this as a node that runs on DepthAI [here](https://github.com/kunaltyagi/pytorch_harris), using the technique from [here](https://rahulrav.com/blog/depthai_camera.html).

[image]

Luxonis-Brandon, 2021-06-23:
First version of the hardware block in DepthAI being used is below:

[image]

eric-schleicher, 2021-06-25:
melike!

eric-schleicher, 2021-06-25:
note: deadlink.    is this example published?

> Hi @matanster @Luxonis-Brandon I published an unofficial example of Edge Detection using Harris filtering today, so you can try it out - [...]

Luxonis-Brandon, 2021-06-25:
Those ones were host-side versions of those; not running on DepthAI.

The example Szabi made is using a hardware bock in DepthAI.  And [this](https://github.com/kunaltyagi/pytorch_harris) one is using the OpenVINO technique to run on DepthAI, using the SHAVES probably.

dhruvsheth-ai, 2021-06-26:
> note: deadlink. is this example published?

Sorry for the trouble @eric-schleicher . Thanks for the PR. Since I migrated the example to a different folder, the link might have gone broken.

dhruvsheth-ai, 2021-06-26:
The link - https://github.com/dhruvsheth-ai/gen2-depthai-experiments/tree/main/gen2-canny-edge-detection