luxonis/depthai #248: Harris Filtering Support

Luxonis-Brandon, 2020-10-20:
### Start with the `why`:

Corner detection ([Harris corner filtering](https://en.wikipedia.org/wiki/Harris_Corner_Detector)) is often useful in a variety of CV/AI pipelining usecases and DepthAI has hardware which allows hardware acceleration of this CV function.

### Move to the `how`:

Implement Harris Filtering as a node in the Gen2 pipeline builder (https://github.com/luxonis/depthai/issues/136).

### Move to the `what`:

Support Harris Filtering as a Gen2 node.

eric-schleicher, 2021-05-24:
+1 for this getting prioritized sooner than later.

szabi-luxonis, 2021-07-22:
This is now possible on https://github.com/luxonis/depthai-python/pull/317
Either `SHI_THOMASI` or `HARRIS` can be selected for feature detection.

```
featureTracker->initialConfig.setAlgorithmType(dai::FeatureTrackerConfigData::AlgorithmType::CORNER_DETECTION);
featureTracker->initialConfig.setCornerDetector(dai::FeatureTrackerConfigData::CornerDetector::HARRIS);
```

asa, 2021-07-23:
Testing this with dai::FeatureTrackerConfigData::CornerDetector::SHI_THOMASI doesn't succeed. The first few frames of tracks show up when printing, but then no new tracks appear. Sometimes it starts at the beginning for a few frames, then ends for a while, then starts back up after 10 seconds or so. Inconsistent.

asa, 2021-07-23:
After working with it more it appears there is a temporal filter on the quantity that are on screen at any one time and they are slowly added, one at a time till tracking a few hundred on an 800_P frame in my environment. They do look cleaner than the harris corners. Nice!

asa, 2021-07-23:
Well, that might have just been because I was running the Left tracker with SHI_THOMASI and the right one with HARRIS for comparison. Setting them both to SHI_THOMASI seems to solve the issue. Sorry for the noise!

szabi-luxonis, 2021-08-10:
Implemented in: https://github.com/luxonis/depthai-python/pull/317

closed 2021-08-10 by szabi-luxonis