luxonis/depthai #259: Crash on "del device"

geaxgx, 2020-10-31:
In my program, I may need to call the following code several times in succession:

```
device = depthai.Device('', False)
pipeline = device.create_pipeline(...)
... # Some stuff
del device
```

On the 2nd call to 'del device', the program crashes:
terminate called after throwing an instance of 'std::system_error'
  what():  Argument invalide
Abandon (core dumped)

I am on ubuntu 18.04 and I can reproduce the problem easily by simply duplicating the hello_world.py code https://github.com/luxonis/depthai-tutorials/blob/master/1-hello-world/hello_world.py

alex-luxonis, 2020-10-31:
Hi @geaxgx , I can reproduce the crash, it's related to this code on C++ side, the variable `once` is already cleared at the second run, the watchdog thread is not recreated, and it will fail in the `join()` call.
https://github.com/luxonis/depthai-core/blob/678e612/src/device.cpp#L151...L167

I'll create a quick fix.

alex-luxonis, 2020-10-31:
@geaxgx Please check if the issue is resolved by reinstalling `depthai` from the branch `fix_recreate_device_loop`:
`python3 -m pip install -r requirements.txt`

And thanks for the issue report!

geaxgx, 2020-10-31:
Problem solved ! 
Thank you @alex-luxonis !

closed 2020-10-31 by geaxgx