luxonis/depthai #263: Model Downloader fails on Windows 10 and Python 3.9

VanDavv, 2020-11-01:
I tried to run `depthai_demo.py` on Windows 10 and Python 3.9.

This is the error I've got

```
$ C:\\Users\\VanDa\\AppData\\Local\\Programs\\Python\\Python39\\python.exe depthai_demo.py
No calibration file. Using Calibration Defaults.
Using depthai module from:  C:\Users\VanDa\AppData\Local\Programs\Python\Python39\lib\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  0.3.0.0+dev
Depthai version required:   0.3.0.0
Depthai development version found, skipping check.
Compiling model for 14 shaves, 14 cmx_slices and 1 NN_engines
model_compilation_target: cloud
Traceback (most recent call last):
  File "C:\Users\VanDa\dev\depthai\depthai_demo.py", line 370, in <module>
    dai.startLoop()
  File "C:\Users\VanDa\dev\depthai\depthai_demo.py", line 45, in startLoop
    configMan = DepthConfigManager(args)
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 26, in __init__
    self.generateJsonConfig()
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 192, in generateJsonConfig
    blobMan = BlobManager(self.args, self.calc_dist_to_bb, shave_nr, cmx_slices, NCE_nr)
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 377, in __init__
    self.blob_file = self.compileBlob(self.args['cnn_model'], self.args['model_compilation_target'])
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 431, in compileBlob
    ret = download_and_compile_NN_model(nn_model, model_zoo_folder, shave_nr_opt, cmx_slices_opt, NCE_nr, outblob_file, model_compilation_target)
  File "C:\Users\VanDa\dev\depthai\model_compiler\model_compiler.py", line 133, in download_and_compile_NN_model
    download_location = download_model(model, model_zoo_folder)
  File "C:\Users\VanDa\dev\depthai\model_compiler\model_compiler.py", line 28, in download_model
    result = subprocess.run(downloader_cmd)
  File "C:\Users\VanDa\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 501, in run
    with Popen(*popenargs, **kwargs) as process:
  File "C:\Users\VanDa\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 947, in __init__
    self._execute_child(args, executable, preexec_fn, close_fds,
  File "C:\Users\VanDa\AppData\Local\Programs\Python\Python39\lib\subprocess.py", line 1416, in _execute_child
    hp, ht, pid, tid = _winapi.CreateProcess(executable, args,
FileNotFoundError: [WinError 2] Nie mo▒na odnale▒▒ okre▒lonego pliku
```
Where `Nie mo▒na odnale▒▒ okre▒lonego pliku` means `Cannot find a specified file`

Tested on latest `develop`

szabi-luxonis, 2020-11-01:
Seems to be the same issue as one of our customers recently encountered:
https://luxonis-community.slack.com/archives/CT79Z6GP5/p1603936232233500?thread_ts=1603936232.233500&cid=CT79Z6GP5

VanDavv, 2020-11-01:
I've followed the instructions, installed Python 3.9 from MS Store and this error does not occur

(Note that for Python 3.9, you need to install Visual Studio 2017 with the following components to be able to install our requirements)
[image]

Now, I have the following error

```
PS C:\Users\VanDa\dev\depthai> python3.9.exe .\depthai_demo.py
Using depthai module from:  C:\Users\VanDa\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  0.3.0.0+dev
Depthai version required:   0.3.0.0
Depthai development version found, skipping check.
No calibration file. Using Calibration Defaults.
[91mCompiling model for 14 shaves, 14 cmx_slices and 1 NN_engines [0m
model_compilation_target: cloud
Traceback (most recent call last):
  File "C:\Users\VanDa\dev\depthai\depthai_demo.py", line 382, in <module>
    dai.startLoop()
  File "C:\Users\VanDa\dev\depthai\depthai_demo.py", line 58, in startLoop
    configMan = DepthConfigManager(args)
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 26, in __init__
    self.generateJsonConfig()
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 192, in generateJsonConfig
    blobMan = BlobManager(self.args, self.calc_dist_to_bb, shave_nr, cmx_slices, NCE_nr)
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 377, in __init__
    self.blob_file = self.compileBlob(self.args['cnn_model'], self.args['model_compilation_target'])
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 431, in compileBlob
    ret = download_and_compile_NN_model(nn_model, model_zoo_folder, shave_nr_opt, cmx_slices_opt, NCE_nr, outblob_file, model_compilation_target)
  File "C:\Users\VanDa\dev\depthai\model_compiler\model_compiler.py", line 133, in download_and_compile_NN_model
    download_location = download_model(model, model_zoo_folder)
  File "C:\Users\VanDa\dev\depthai\model_compiler\model_compiler.py", line 30, in download_model
    raise RuntimeError("Model downloader failed! Not connected to the internet?")
RuntimeError: Model downloader failed! Not connected to the internet?
```

I think we should improve the error message on this one, no info about the cause of the error

szabi-luxonis, 2020-11-01:
Should be fixed in #264

VanDavv, 2020-11-01:
Fixed, thanks @szabi-luxonis

VanDavv, 2020-11-01:
I need to reopen - the downloader successfully downloaded mobilenet-ssd and now is refusing to download any other model.

That said, running

```
python3.9.exe .\depthai_demo.py
```

works fine, whereas 

```
python3.9.exe .\depthai_demo.py -cnn face-detection-retail-0004
```

fails with the following error

```
No calibration file. Using Calibration Defaults.
Using depthai module from:  C:\Users\VanDa\AppData\Local\Packages\PythonSoftwareFoundation.Python.3.9_qbz5n2kfra8p0\LocalCache\local-packages\Python39\site-packages\depthai.cp39-win_amd64.pyd
Depthai version installed:  0.3.0.0+dev
Depthai version required:   0.3.0.0
Depthai development version found, skipping check.
[91mCompiling model for 14 shaves, 14 cmx_slices and 1 NN_engines [0m
model_compilation_target: cloud
Traceback (most recent call last):
  File "C:\Users\VanDa\dev\depthai\depthai_demo.py", line 370, in <module>
    dai.startLoop()
  File "C:\Users\VanDa\dev\depthai\depthai_demo.py", line 45, in startLoop
    configMan = DepthConfigManager(args)
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 26, in __init__
    self.generateJsonConfig()
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 192, in generateJsonConfig
    blobMan = BlobManager(self.args, self.calc_dist_to_bb, shave_nr, cmx_slices, NCE_nr)
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 377, in __init__
    self.blob_file = self.compileBlob(self.args['cnn_model'], self.args['model_compilation_target'])
  File "C:\Users\VanDa\dev\depthai\depthai_helpers\config_manager.py", line 431, in compileBlob
    ret = download_and_compile_NN_model(nn_model, model_zoo_folder, shave_nr_opt, cmx_slices_opt, NCE_nr, outblob_file, model_compilation_target)
  File "C:\Users\VanDa\dev\depthai\model_compiler\model_compiler.py", line 133, in download_and_compile_NN_model
    download_location = download_model(model, model_zoo_folder)
  File "C:\Users\VanDa\dev\depthai\model_compiler\model_compiler.py", line 30, in download_model
    raise RuntimeError("Model downloader failed! Not connected to the internet?")
RuntimeError: Model downloader failed! Not connected to the internet?
```

szabi-luxonis, 2020-11-01:
It works for me just fine, tried on clean repo.

szabi-luxonis, 2020-11-01:
It works fine for me, just tried on clean repo.

szabi-luxonis, 2020-11-02:
I assume you didn't merge the changes from #264 into your branch that you used for testing, and you already had the `mobilenet-ssd` downloaded.

VanDavv, 2020-11-02:
That was the case, sorry @szabi-luxonis, works great

closed 2020-11-02 by VanDavv