luxonis/depthai #267: opevino 2020.4/2021.1 model download process fails

rychardeh, 2020-11-03:
In both the openvino_2020.4 and openvino_2021.1 branches, the model compilation process fails to complete because it appears that the post processor is looking for the model files in a different folder to the one they were downloaded to - `FP16` not used when downloading.  E.g. using 2020.04 (on RPI):

```
python3 depthai_demo.py
Using depthai module from:  /home/rych/.virtualenvs/openvino/lib/python3.7/site-packages/depthai.cpython-37m-arm-linux-gnueabihf.so
Depthai version installed:  0.2.0.1+09142c24eb487df4e5f63413bc7ee895113bdd4b
Depthai version required:   0.2.0.1+09142c24eb487df4e5f63413bc7ee895113bdd4b
No calibration file. Using Calibration Defaults.
Compiling model for 14 shaves, 14 cmx_slices and 1 NN_engines
Openvino installation detected /home/rych/openvino
Unsupported openvino version installed at /home/rych/openvino, supported version is: 2020.4
model_compilation_target: cloud
################|| Downloading models ||################

========== Retrieving /home/rych/depthai/model_compiler/downloads/mobilenet-ssd/mobilenet-ssd.prototxt from the cache

========== Retrieving /home/rych/depthai/model_compiler/downloads/mobilenet-ssd/mobilenet-ssd.caffemodel from the cache

################|| Post-processing ||################

Traceback (most recent call last):
  File "depthai_demo.py", line 382, in <module>
    dai.startLoop()
  File "depthai_demo.py", line 58, in startLoop
    configMan = DepthConfigManager(args)
  File "/home/rych/depthai/depthai_helpers/config_manager.py", line 26, in __init__
    self.generateJsonConfig()
  File "/home/rych/depthai/depthai_helpers/config_manager.py", line 192, in generateJsonConfig
    blobMan = BlobManager(self.args, self.calc_dist_to_bb, shave_nr, cmx_slices, NCE_nr)
  File "/home/rych/depthai/depthai_helpers/config_manager.py", line 377, in __init__
    self.blob_file = self.compileBlob(self.args['cnn_model'], self.args['model_compilation_target'])
  File "/home/rych/depthai/depthai_helpers/config_manager.py", line 431, in compileBlob
    ret = download_and_compile_NN_model(nn_model, model_zoo_folder, shave_nr_opt, cmx_slices_opt, NCE_nr, outblob_file, model_compilation_target)
  File "/home/rych/depthai/model_compiler/model_compiler.py", line 170, in download_and_compile_NN_model
    raise RuntimeError(f"{ir_model_location} doesn't exist for downloaded model!")
RuntimeError: /home/rych/depthai/model_compiler/downloads/mobilenet-ssd/FP16 doesn't exist for downloaded model!
```

Ignore the unsupported openvino version.  It is correct but the method for detecting it doesn't work unless `INTEL_OPENVINO_DIR` contains the version number in the path and mine doesn't....

szabi-luxonis, 2020-11-03:
As mentioned [here](https://luxonis-community.slack.com/archives/CT79Z6GP5/p1603471447175500) openvino 2020.4 and 2021.1 works currently only if openvino is installed locally, that's why compilation is not working : `model_compilation_target: cloud`, our cloud compiler is not ready to do caffemodel->IR conversion automatically based on the `yml` config.
At the moment openvino doesn't provide any environment variables set to determine the installed openvino version, I will find another way by parsing the `version.txt` in deployment_tools/inference_engine

For now I removed the check for openvino version based on folder, so it works even with non-standard openvino installation.

rychardeh, 2020-11-03:
Ah, sorry, I missed that.   I see that it only happened because I was running on a RPI and there is no OpenVINO distribution for it that includes the model optimiser so it has to depend on cloud.

szabi-luxonis, 2020-11-03:
> Ah, sorry, I missed that. I see that it only happened because I was running on a RPI and there is no OpenVINO distribution for it that includes the model optimiser so it has to depend on cloud.

We are going to update our blobconverter, so everything will be in cloud based on the `yml` config for all openvino versions.

closed 2021-08-13 by szabi-luxonis