luxonis/depthai #283: UVC Output as a Pipeline Node

Luxonis-Brandon, 2020-12-09:
### Start with the `why`:

In some cases it may be desirable to feed UVC (USB Video Class, e.g. a webcam) output to have compatibility with existing software stacks/etc.  To provide `instant integration` sort of compatibility, particularly if working with closed-source tools (or hard to modify tools) which already work with UVC-compatible inputs.

### Move to the `how`:

Luxonis-Brandon, 2021-03-04:
We have this initially working now.  Need to see if the code is externally usable now.

dhruvsheth-ai, 2021-03-05:
Eager to test it out

Luxonis-Brandon, 2021-03-05:
This should work on Linux, but may have issues on other OSes:
https://github.com/luxonis/depthai-python/blob/gen2_uvc/examples/19_uvc_video.py
`python3 -m pip install depthai==0.0.2.1+e70becfe1c1908c6148f842006b9986ebab39cab --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local`

dhruvsheth-ai, 2021-03-05:
Thanks! Will report how it goes

dhruvsheth-ai, 2021-03-06:
So, I tried it on armv7l, it connects as UVC output, however fails to transmit image data to any source, for eg- connects as Intel device on a webcam test, fails to transmit image data though

ORippler, 2021-03-06:
Works for me under Ubuntu 18.04 and the above linked snapshot + python file. Tested it out with Zoom.

dhruvsheth-ai, 2021-03-06:
Interesting...I'll see what's causing error working on armv7l

cafemoloko, 2021-03-06:
@Luxonis-Brandon can we close this issue as it seems resolved?

Luxonis-Brandon, 2021-03-07:
So let's leave it open for now as it's not formally integrated with DepthAI so far (I think).  Actually, let's see what @alex-luxonis has to say on this.  He knows better than me.

mx1up, 2021-04-25:
I can confirm it works as expected with OBS studio. I will try it out for streaming. I commented out https://github.com/luxonis/depthai-python/blob/e70becfe1c1908c6148f842006b9986ebab39cab/examples/19_uvc_video.py#L15 in order to have autofocus.
this sample or code does not seem to be integrated yet in the main distribution

mx1up, 2021-04-25:
@Luxonis-Brandon which would be the most efficient video format: YUV420, BGR3 or YV12?

Luxonis-Brandon, 2021-04-25:
Thanks for the test and information here @mx1up !  

On which is the most efficient, I'm not sure actually.  I'm curious to know as well.  And I don't know WRT if it would be host/video-card dependent either (but great question/point).  @alex-luxonis is probably the most likely among us to know on both.

alex-luxonis, 2021-04-25:
At the moment only the YUV420 (YU12) format is exposed. Some programs like `guvcview` show those additional formats, not sure if that's an issue with those programs, or somehow caused by the USB descriptor. It's listed properly by this command:
`v4l2-ctl -d /dev/video0 --list-formats`
```
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YU12' (Planar YUV 4:2:0)
```

It is not merged yet to mainline, as we don't have a clean way to expose the UVC interface only when the UVC node is added to the pipeline. With the `depthai` version here, the UVC device always shows up on host, but it operational only with the UVC node linked in the pipeline. (It's not harmful otherwise to the normal operation of XLink VSC interface, for pipelines not using UVC, but might be confusing.)

For a clean integration, we need these (WIP, I think @themarpe is looking at):
- pre-boot config, to send the Gen2 pipeline (or a part thereof) to device prior to it booting up the DepthAI FW;
- dynamically constructing the USB descriptor, based on the interfaces required by the nodes/config in the Gen2 pipeline.

mx1up, 2021-04-26:
thanks for the explanation.
FYI: OBS studio does mention "(emulated)" after BGR3 and YV12. I did some simplistic performance measurements (both windowed and fullscreen on an intel chipset) using vmstat and it seems YUV420 and YV12 perform more or less the same while BGR3 is definitely slower.

for example:
BGR3 (emulated)
```
 us  sy  id  wa  st
  5   3  92   0   0
  5   3  92   0   0
  6   4  90   0   0
  6   2  92   0   0
  6   4  90   0   0
  6   3  91   0   0
  6   4  90   0   0

```
YUV420
```
 us  sy  id  wa  st
  2   3  94   0   0
  3   4  92   0   0
  2   5  93   0   0
  2   4  94   0   0
  3   4  93   0   0
  3   5  92   0   0
  3   4  93   0   0
```

YV12 (emulated)
```
 us  sy  id  wa  st
  2   3  94   0   0
  3   4  92   0   0
  2   5  93   0   0
  2   4  94   0   0
  3   4  93   0   0
  3   5  92   0   0
  3   4  93   0   0
```

linuxbunter213, 2021-05-27:
Nice

TannerGilbert, 2021-06-05:
@Luxonis-Brandon is `https://github.com/luxonis/depthai-python/blob/gen2_uvc/examples/19_uvc_video.py` still the newest version available and is there already an implementation for the depth output?

alex-luxonis, 2021-06-05:
Yes, that is still the latest for now. But soon we'll start to work again on it and integrate it in mainline.

About depth, it may be possible for now to link other streams to `uvc.input`, but the UVC descriptor is hardcoded to 1920x1080 NV12 (12 bits per pixel), so the host could reject the frames (`uvcvideo` in newer Linux kernels may do that, and configuring it with `nodrop=1` as [here](https://help.ubuntu.com/community/UVC) could be a workaround). We'll also properly configure the UVC descriptor to match with the frame format.

TannerGilbert, 2021-06-05:
@alex-luxonis Thanks for the quick reply. I'm looking forward to the integration.

bmorawska, 2021-06-16:
Hi, Does anyone has faced the problem of noisy image from uvc camera (oak-d, ubuntu 20.04, AMD FX(tm)-6350 Six-Core Processor, usb 3.0, 19_uvc_video.py). I see some kind of green dots/grid/perpendicular lines and defective image underneath (no colors, the wall on left is actually yellow). I use guvcview.

[image]

AChangXD, 2021-08-03:
@alex-luxonis @Luxonis-Brandon What is the timeline for this working on Windows?

Luxonis-Brandon, 2021-08-03:
So we haven't done anything more on it directly.  But we've done a ton of bootloader work which is the crux of getting it to work properly.
So I don't know when the Windows setup will work better.
That said, if you wanted you could get a OAK-SoM-IoT and install it on an OAK-D and it can be pre-flashed with UVC pipeline so that then it will work with Windows now @AChangXD  .
https://shop.luxonis.com/collections/iot

Or simpler you could get one of these and flash them with a UVC pipeline, and they'll boot up right away as UVC so will work fine with Windows as well.

Once the underpinnings are there (the total rework of the bootloader), I'm just not sure yet how long doing the refactoring on the UVC node will take to get Windows to be OK with it.  The flashed-device route will definitely take care of it though, and will work now.

Thoughts?

Thanks,
Brandon

afakhry01, 2021-08-12:
Is there any chance the UVC branches are updated to work on POE OAKs?

themarpe, 2021-08-12:
@afakhry01 PoE OAKs aren't priority for UVC integration, but could still be done. Right now we are in progress of making this work seamlessly for USB models. Afterwards this could be looked into as well. 

That said - the use case with current models not having exposed USB isn't there as is for other models. If you do access the internals of the device to get to USB connector, you can use it as USB device by modifying the boot switches to 0x16. But this is a bit out of scope of regular usage and I only advise doing that with caution. Also flashing the UVC pipeline in this case won't work as expected.

ES-Alexander, 2021-09-19:
Will/does the UVC node/pipeline support common camera settings through v4l2-ctl?
And even better, any chance it'll support changing model parameters once it's in its final form (e.g. setting disparity confidence for a depth node)? This would make configurable deployment to users much simpler, since it could present as a normal camera with the processing pipeline pre-loaded and relevant settings tuneable, with no required programming/running code on the users' part.

ES-Alexander, 2021-09-21:
> what is the usual case of disparity/depth map consumption in other software that take UVC?

I'm not sure of anything that meaningfully would take depth through UVC - I was more thinking that a plug-and-play product with a depth-aware filtering pipeline could be useful, but as with all filters it would be nice to be able to tweak some parameters while using it (in the same way that it's often useful to tweak normal camera brightness + contrast and other settings, to set up profiles for a set of operating conditions). It's of course possible to create the plug-and-play nature with a custom setup that allows setting both standard UVC controls and separate model controls, and handles the details under the hood, but that requires custom software running on the user's computer, whereas there are already existing tools for controlling UVC parameters that would be able to auto-detect new ones if we were able to add them.

The main benefit is easier integration for products that don't need much modification or a high-speed response from the host device, but which would benefit from the user being able to tweak some settings. 

It may also be possible to do some kind of hybrid setup, where a program sets up the pipeline (and optionally provides any required high-speed responses to the camera), but user settings are exposed as UVC settings, in which case custom code is being run but at least an interface doesn't need to be developed. That also solves the 'no flash memory on OAK-D-Lite' issue.

Another alternative is making a standardised depthai settings API, and some software to interface with it, which is probably a good idea in itself (completely unrelated to UVC).

ES-Alexander, 2021-09-21:
> products that don't need much modification or a high-speed response from the host device, but which would benefit from the user being able to tweak some settings.

An easy example of this would be for PTZ amounts (see #135). Ideally that would be supported out of the box if doing a UVC boot, since it's functionality that anybody using a depthai camera as just a camera would benefit from.

themarpe, 2021-09-21:
@ES-Alexander good points.

Regarding
> I was more thinking that a plug-and-play product with a depth-aware filtering pipeline could be useful

So a webcam with background bluring in a sense?

> The main benefit is easier integration for products that don't need much modification or a high-speed response from the host device, but which would benefit from the user being able to tweak some settings.

> An easy example of this would be for PTZ amounts
 
With above mentioned examples the main issue I see is that they might be too specific. As our platform is very versatile, this kind of default behavior might suit very few people. 
And regarding configurability through UVC, as soon as a 3rdparty application is necessary to configure those in runtime, it could as well be DepthAI where all configurability is possible. But I do agree that its sometimes easier to just use existing software to control things.

The way I see it is if we do go forward with this, to expose the "base" of the device, like all camera streams or maybe add depth as well (if possible to consume by certain apps, or maybe even by opencv, ...), with very base configurability through UVC options, that aren't too specific.

In case of having more specific examples running, one could flash (on models with flash onboard) a custom DepthAI application, where they could expose any kind of data and control any kind of mode / settings using the UVC settings and then managing the pipeline nodes by sending configuration messages using a Script node. 

Thoughts?

ES-Alexander, 2021-09-22:
> > I was more thinking that a plug-and-play product with a depth-aware filtering pipeline could be useful
>
> So a webcam with background bluring in a sense?

Yeah sure, that's a good example :-)

> With above mentioned examples the main issue I see is that they might be too specific. As our platform is very versatile, this kind of default behavior might suit very few people.

I'm not recommending all depthai devices get shipped with a set model on them, I'm saying it could be nice if it was a possibility to configure a running model through UVC controls - especially for deployed use-cases where someone makes a model which they load onto a suitable device, then provide that device to someone else who only needs it for that purpose, in which case accessing it as a normal camera but with extra settings for the programmed in extra functionality could be handy :-)

> maybe add depth as well (if possible to consume by certain apps, or maybe even by opencv, ...)

Not what I was going for, but definitely an interesting idea :-)

> In case of having more specific examples running, one could flash (on models with flash onboard) a custom DepthAI application, where they could expose any kind of data and control any kind of mode / [...]

Yeah, this sounds like the kind of thing I was thinking of. Makes sense that it would require a script node to process and handle the UVC settings requests, but more generally I just wasn't sure how feasible it would be to actually get information from inputted UVC controls to somewhere useful outside the UVC node, so thought I'd ask about whether it would be a possibility (hence my original comment) :-)

themarpe, 2021-09-22:
@ES-Alexander I see, you are suggesting basically to have the capability of flashing the device with a UVC enabled DepthAI Application and not one necessarily being supplied by the device itself already. 

In that case, that overlaps nicely - with adding flash onboard both cases could actually be done.

We are discussing internally of adding some small amounts of flash to non IoT devices as well to provide seamless UVC use case as well as the capability of flashing your own. TBD.

> so thought I'd ask about whether it would be a possibility

Yeah, I presume many things could already be done currently (if not counting the general UVC node support), if we send a generic UVC control message from UVC node, and you link that into Script node, parsing it as it seems fit, and issue config messages from there on, like reconfiguring ImageManip for instance, or changing some StereoDepth parameters, etc...