luxonis/depthai #295: Could not be recognize OAK-1 on Docker container

atinfinity, 2021-01-04:
## Description of the bug

I tried to use depthai with OAK-1 on Docker container.
But, OAK-1 could not be recognized.

```python
$ python3
>>> import depthai
>>> device = depthai.Device('', False)
XLink already initialized.
Sending internal device firmware
Failed to boot the device: 1-ma2480, err code 3
depthai: Error initializing xlink
```

## Steps to reproduce the bug
I upload my Dockefile.
https://gist.github.com/atinfinity/c3255077e23164438c19544687e05b84

## System information (version)
### Device
- OAK-1
  - https://store.opencv.ai/products/oak-1 
  - https://github.com/luxonis/depthai-hardware/tree/master/BK1096_OAK-1_Modular

### Host
- Ubuntu 20.04 64bit
- Docker 19.03.13

### Docker Container
- Ubuntu 18.04 64bit
- https://github.com/luxonis/depthai/tree/v0.4.0.0

### Reference
- https://docs.luxonis.com/en/latest/pages/api/?dummy=http://#installing-system-dependencies
- https://github.com/luxonis/depthai-docker

Luxonis-Brandon, 2021-01-04:
Thanks and sorry about the delay.  We have experience getting DepthAI working in Dockers, but I don't think we have written it up yet.  Slack (globally) was down today, which resulted in us not being able to discuss this as normal until now.  

Will circle back.  I think we can get the instructions written up soon.

Thanks again,
Brandon

tomoaki0705, 2021-01-05:
I've hit the same situation, using Docker.
One thing I realized which might help other people is the output of `lsusb`
When running on Docker, I could see the vendor ID and device ID, but not the device name
```
$ lsusb
Bus 001 Device 005: ID 03e7:2485
```

Afterwards, I tried new setup on non-Docker environment, and the output changed
```
$ lsusb
Bus 001 Device 004: ID 03e7:2485 Intel Movidius MyriadX
```

At least, this might help for diagnosis, and I hope this help isolating the cause.
Best

Luxonis-Brandon, 2021-01-06:
Thanks @tomoaki0705 .  And we are still discussing now how to do this the right way.  Sorry about the delay.

Luxonis-Brandon, 2021-01-06:
OK, we think we have summarized what we had iterated on internally and with a customer WRT getting Docker to work in a similar (same?) situation.  We are now testing it.  But in the meantime:

We think it boils down to: (For Linux as far as I know)

1. compiling libusb without udev inside the cointainer
2. replacing default libusb to be loaded with the newly compiled (I guess prepending build directory to PATH or LD_LIBRARY_PATH, will do)
3. Adding the following parameter to docker --device-cgroup-rule='c 189:* rmw

The following should configure, compile and install libusb without udev:
```
wget https://github.com/libusb/libusb/releases/download/v1.0.24/libusb-1.0.24.tar.bz2
tar xf libusb-1.0.24.tar.bz2
cd libusb-1.0.24
./configure --disable-udev --prefix="$PWD/install_dir" 
make -j
make install
export LD_LIBRARY_PATH="$PWD/install_dir/lib":$LD_LIBRARY_PATH
```
And at the end set the LD_LIBRARY_PATH to the installation directory.

We're not exactly sure if that export will persist from docker build to docker usage - might rather replace with
`ENV LD_LIBRARY_PATH="$PWD/install_dir/lib":$LD_LIBRARY_PATH`.

@xtechnicianx is going through and testing/characterizing this.

xtechnicianx, 2021-01-09:
Sorry for the late follow up, was out of office most of Wednesday, and a good portion of Thursday. Took some time to look at this earlier today, but was not successful yet. Whether I update the Dockerfile to build the libusb driver when building the container, or doing it after, it does not seem to work with the above mentioned steps. 

Most of my testing has been done using Ubuntu 20.04.1. But I've also tried with a variety of host OS (including Kali 2020.4, macOS, Windows 10, and virtualized Ubuntu). The outcome is the same across all. 

My testing was strictly with the demo file. Running `depthai_demo.py -dev list`, the output is: 
```
XLink initialized.
Detected 1 device(s):
  3-ma2480     on USB port: 3
```

When trying to interact with the hardware I see two errors. If I don't build libusb without udev, then I get err no 5, then I see err no 3. When I build libusb I seem to only get err no 3. 

```
XLink initialized.
Sending internal device firmware
Failed to boot the device: 3-ma2480, err code 3
depthai: Error initializing xlink
['metaout', 'previewout']
Available streams: []
device is not initialized
Pipeline is not created.
```

I'm utilizing the cgroup rule to properly detect hardware across reconnects (or the USB2/USB3 switch) when the camera inits. The above cgroup rule has a minor typo, it's missing the closing ', it should be `--device-cgroup-rule='c 189:* rmw'`.

I tried running docker with and without `privileged` mode, which should give it full access to the host.

I plan to dig in more, but wanted to provide an update on what I've tried so far.

xtechnicianx, 2021-01-09:
Oh, here's the Dockerfile I used (extension added to satisfy GitHub reqs). And sorry I didn't include the full output from err no 5, as I'm not in front of that computer. Working from notes I made earlier. 
[Dockerfile.txt](https://github.com/luxonis/depthai/files/5790104/Dockerfile.txt)

yoneken, 2021-01-09:
I got the same error in the non-Docker environment.
Ubuntu 18.04.5 / Intel(R) Celeron(R) CPU N3450
[image]

alex-luxonis, 2021-01-09:
@yoneken It may be possible that the `udev` rules that would allow access to the USB device in user mode (without `sudo`) are not installed.

Could you try running these commands taken from our system dependencies install script:
```
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger
```
In addition, please unplug and reinsert the USB cable after running the commands.

https://docs.luxonis.com/en/latest/pages/api/#installing-system-dependencies
https://docs.luxonis.com/en/latest/_static/install_dependencies.sh

yoneken, 2021-01-10:
@alex-luxonis Thank you for your quick reply!
My issue is solved!

Luxonis-Brandon, 2021-01-16:
@atinfinity - so as an update we're making a better Docker support system, you can see https://github.com/luxonis/depthai-docs-website/pull/115 .

Luxonis-Brandon, 2021-01-26:
@atinfinity - we just got this cleanly working so that you can use Docker now (theoretically) without issue.  Sorry about the delay, this one turned out to be tricky!  We will get updated instructions here shortly.  I just saw this PR merge: https://github.com/luxonis/depthai-docs-website/pull/135/files

Luxonis-Brandon, 2021-01-27:
More updates on documenting this here: https://github.com/luxonis/depthai-docs-website/pull/137

atinfinity, 2021-01-28:
@Luxonis-Brandon Thank you for your help! I refered to <https://github.com/luxonis/depthai-python/pull/100>.
And, I could use OAK-1 with depthai-python(gen2_develop) on Docker container. And, I have questions.

- I could not use OAK-1 with non-root user on Docker container. Do I need to make additional settings?
- I used `luxonis/depthai-base` as base image. Is there Dockerfile of this image? I could not find.

Luxonis-Brandon, 2021-01-28:
Thanks for the questions on this.  I think @pamolloy is still in the process of documenting how to use everything.  It will be great to have you try this out so we can refine it.  So thanks for doing so.  Pinging him offline now.

Thanks again,
Brandon

pamolloy, 2021-01-28:
Hi @atinfinity, 

It's great to hear that it worked for you!

>     * I could not use OAK-1 with non-root user on Docker container. Do I need to make additional settings?

Unfortunately, Docker always needs to be run as root. That is a requirement based on how they implemented their container runtime. However, there are other runtimes like [podman](https://podman.io/) that don't have that requirement. One of our customers has successfully used podman with some extra work, but we don't officially support it yet. :crossed_fingers: I'll get to it soon!

>     * I used `luxonis/depthai-base` as base image. Is there Dockerfile of this image? I could not find.

The Dockerfile for the base image is located under [depthai-docs-website/ci/Dockerfile](https://github.com/luxonis/depthai-docs-website/blob/master/ci/Dockerfile).

I've added a link to it on the [depthai-base Docker Hub repository](https://hub.docker.com/r/luxonis/depthai-base).

closed 2021-01-29 by pamolloy

SohailSayed, 2024-01-23:
Still having this issue even after running
```
echo 'SUBSYSTEM=="usb", ATTRS{idVendor}=="03e7", MODE="0666"' | sudo tee /etc/udev/rules.d/80-movidius.rules
sudo udevadm control --reload-rules && sudo udevadm trigger

```