luxonis/depthai #298: Numpy Error when running on Big Sur

esbasti, 2021-01-08:
Hi,

Has anyone experienced issues when trying to run `python3 install_requirements.py` in Big Sur OS.

Seems like the problem is related to the numpy library. However, I was able to install numpy through home brew and did not get the same error.

Happened on Big Sur running on a M1 chip

Here is the error log:

Requirement already satisfied: pip in /opt/homebrew/lib/python3.9/site-packages (20.3.3)
WARNING: Skipping depthai as it is not installed.
Ignoring open3d: markers 'platform_machine != "armv7l" and python_version < "3.9"' don't match your environment
Collecting argcomplete==1.12.1
  Using cached argcomplete-1.12.1-py2.py3-none-any.whl (38 kB)
Collecting depthai==0.4.0.0
  Using cached depthai-0.4.0.0.tar.gz (84 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting numpy==1.19.3
  Using cached numpy-1.19.3.zip (7.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmp60t7dj_s
         cwd: /private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651
    Complete output (217 lines):
    Running from numpy source directory.
    setup.py:480: UserWarning: Unrecognized setuptools command, proceeding with generating Cython sources and expanding templates
      run_build = parse_setuppy_commands()
    Processing numpy/random/_bounded_integers.pxd.in
    Processing numpy/random/_philox.pyx
    Processing numpy/random/_bounded_integers.pyx.in
    Processing numpy/random/_sfc64.pyx
    Processing numpy/random/_mt19937.pyx
    Processing numpy/random/bit_generator.pyx
    Processing numpy/random/mtrand.pyx
    Processing numpy/random/_generator.pyx
    Processing numpy/random/_pcg64.pyx
    Processing numpy/random/_common.pyx
    Cythonizing sources
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    blis_info:
      libraries blis not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    openblas_info:
      libraries openblas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_3_10_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries tatlas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_3_10_blas_info:
      libraries satlas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries ptf77blas,ptcblas,atlas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
[... 159 lines cut ...]
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/__init__.py", line 526, in customize
        linker_so_flags = self.flag_vars.linker_so
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/environment.py", line 37, in __getattr__
        return self._get_var(name, conf_desc)
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/environment.py", line 53, in _get_var
        var = self._hook_handler(name, hook)
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/__init__.py", line 705, in _environment_hook
        return hook()
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/gnu.py", line 346, in get_flags_linker_so
        flags = GnuFCompiler.get_flags_linker_so(self)
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/gnu.py", line 136, in get_flags_linker_so
        os.environ['MACOSX_DEPLOYMENT_TARGET'] = target
      File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/os.py", line 684, in __setitem__
        value = self.encodevalue(value)
      File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/os.py", line 756, in encode
        raise TypeError("str expected, not %s" % type(value).__name__)
    TypeError: str expected, not int
    ----------------------------------------
ERROR: Command errored out with exit status 1: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmp60t7dj_s Check the logs for full command output.
Traceback (most recent call last):
  File "/Users/sebastianvillalobos/Sabana Tech/smartaccess/depthai/install_requirements.py", line 17, in <module>
    subprocess.check_call([*pip_install, "-r", "requirements.txt"])
  File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/opt/homebrew/opt/python@3.9/bin/python3.9', '-m', 'pip', 'install', '--user', '-r', 'requirements.txt']' returned non-zero exit status 1.

Luxonis-Brandon, 2021-01-08:
Hi @esbasti ,

I have not seen this.  But I do have something that might be helpful.  Here is what we did to get development working on M1 Macs:
[https://discuss.luxonis.com/d/69-running-depthai-on-apple-m1-based-macs](https://discuss.luxonis.com/d/69-running-depthai-on-apple-m1-based-macs)

The `why` is that some development tools don't yet run natively on M1 Macs.

Thoughts?

Thanks,
Brandon

alex-luxonis, 2021-01-13:
Closing - duplicate of issue https://github.com/luxonis/depthai/issues/299 (resolved).

closed 2021-01-13 by alex-luxonis