luxonis/depthai #299: Issue in Big Sur related to Numpy

esbasti, 2021-01-08:
Hi,

Has anyone experienced issues when trying to run `python3 install_requirements.py` in Big Sur OS.

Seems like the problem is related to the numpy library. However, I was able to install numpy through home brew and did not get the same error.

Happened on Big Sur running on a M1 chip

Here is the error log:

Requirement already satisfied: pip in /opt/homebrew/lib/python3.9/site-packages (20.3.3)
WARNING: Skipping depthai as it is not installed.
Ignoring open3d: markers 'platform_machine != "armv7l" and python_version < "3.9"' don't match your environment
Collecting argcomplete==1.12.1
  Using cached argcomplete-1.12.1-py2.py3-none-any.whl (38 kB)
Collecting depthai==0.4.0.0
  Using cached depthai-0.4.0.0.tar.gz (84 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Collecting numpy==1.19.3
  Using cached numpy-1.19.3.zip (7.3 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... error
    ERROR: Command errored out with exit status 1:
     command: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmp60t7dj_s
         cwd: /private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651
    Complete output (217 lines):
    Running from numpy source directory.
    setup.py:480: UserWarning: Unrecognized setuptools command, proceeding with generating Cython sources and expanding templates
      run_build = parse_setuppy_commands()
    Processing numpy/random/_bounded_integers.pxd.in
    Processing numpy/random/_philox.pyx
    Processing numpy/random/_bounded_integers.pyx.in
    Processing numpy/random/_sfc64.pyx
    Processing numpy/random/_mt19937.pyx
    Processing numpy/random/bit_generator.pyx
    Processing numpy/random/mtrand.pyx
    Processing numpy/random/_generator.pyx
    Processing numpy/random/_pcg64.pyx
    Processing numpy/random/_common.pyx
    Cythonizing sources
    blas_opt_info:
    blas_mkl_info:
    customize UnixCCompiler
      libraries mkl_rt not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    blis_info:
      libraries blis not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    openblas_info:
      libraries openblas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_3_10_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries tatlas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_3_10_blas_info:
      libraries satlas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
    
    atlas_blas_threads_info:
    Setting PTATLAS=ATLAS
      libraries ptf77blas,ptcblas,atlas not found in ['/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib', '/usr/local/lib', '/usr/lib', '/opt/local/lib']
      NOT AVAILABLE
[... 159 lines cut ...]
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/__init__.py", line 526, in customize
        linker_so_flags = self.flag_vars.linker_so
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/environment.py", line 37, in __getattr__
        return self._get_var(name, conf_desc)
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/environment.py", line 53, in _get_var
        var = self._hook_handler(name, hook)
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/__init__.py", line 705, in _environment_hook
        return hook()
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/gnu.py", line 346, in get_flags_linker_so
        flags = GnuFCompiler.get_flags_linker_so(self)
      File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-tomzolhy/numpy_92985193da7a427c82dd73d9d6d03651/numpy/distutils/fcompiler/gnu.py", line 136, in get_flags_linker_so
        os.environ['MACOSX_DEPLOYMENT_TARGET'] = target
      File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/os.py", line 684, in __setitem__
        value = self.encodevalue(value)
      File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/os.py", line 756, in encode
        raise TypeError("str expected, not %s" % type(value).__name__)
    TypeError: str expected, not int
    ----------------------------------------
ERROR: Command errored out with exit status 1: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py prepare_metadata_for_build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmp60t7dj_s Check the logs for full command output.
Traceback (most recent call last):
  File "/Users/sebastianvillalobos/Sabana Tech/smartaccess/depthai/install_requirements.py", line 17, in <module>
    subprocess.check_call([*pip_install, "-r", "requirements.txt"])
  File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/opt/homebrew/opt/python@3.9/bin/python3.9', '-m', 'pip', 'install', '--user', '-r', 'requirements.txt']' returned non-zero exit status 1.

Luxonis-Brandon, 2021-01-08:
Hi @esbasti ,

I have not seen this. But I do have something that might be helpful. Here is what we did to get development working on M1 Macs:
https://discuss.luxonis.com/d/69-running-depthai-on-apple-m1-based-macs

The why is that some development tools don't yet run natively on M1 Macs.

Thoughts?

Thanks,
Brandon

esbasti, 2021-01-08:
Thank you @Luxonis-Brandon for getting back to me.

I tried running the terminal under Rosetta.

But, unfortunately I'm still getting the same error.

Luxonis-Brandon, 2021-01-08:
Got it.  Just to make sure I'm on the same page, have you already run the following (from [here](https://docs.luxonis.com/en/latest/pages/api/#macos))?
`sudo curl -fL http://docs.luxonis.com/_static/install_dependencies.sh | bash`

If you haven't please do close and re-open the terminal after doing so, and run python3 -m pip install -U pip, which should show something like:
`Requirement already satisfied: pip in /Users/leeroy/Library/Python/3.9/lib/python/site-packages (20.3.3)` if it is already updated.

And I want to share this video in case it's helpful in getting up/going:
[https://www.youtube.com/watch?v=0RGmmjed3Hc&feature=youtu.be](https://www.youtube.com/watch?v=0RGmmjed3Hc&feature=youtu.be)

Unfortunately I don't have an M1 Mac on hand (I really want to get one) to try out, otherwise I could see about reproducing.  @alex-luxonis does have one.

esbasti, 2021-01-08:
Thanks for the video, it was very helpful

I was skipping the `python3 -m pip install depthai`

Notheless I get a new error when running this command.

I think this time the error is caused by home brew. I tried to run homebrew under rosetta 2. But, still got the same error

`Collecting depthai
  Using cached depthai-0.4.0.0.tar.gz (84 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmp8m4hw9u7
       cwd: /private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-gfgg7o6b/depthai_16754c96838549f286e1b6e8543b1e04
  Complete output (58 lines):
  running bdist_wheel
  running build
  running build_ext
  Traceback (most recent call last):
    File "setup.py", line 72, in run
      out = subprocess.check_output(['cmake', '--version'])
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 420, in check_output
      return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 501, in run
      with Popen(*popenargs, **kwargs) as process:
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 947, in __init__
      self._execute_child(args, executable, preexec_fn, close_fds,
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1819, in _execute_child
      raise child_exception_type(errno_num, err_msg, err_filename)
  FileNotFoundError: [Errno 2] No such file or directory: 'cmake'
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "/opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py", line 204, in build_wheel
      return _build_backend().build_wheel(wheel_directory, config_settings,
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-j9ot46yy/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 216, in build_wheel
      return self._build_with_temp_dir(['bdist_wheel'], '.whl',
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-j9ot46yy/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 202, in _build_with_temp_dir
      self.run_setup()
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-j9ot46yy/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 253, in run_setup
      super(_BuildMetaLegacyBackend,
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-j9ot46yy/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 145, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 159, in <module>
      setup(
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-j9ot46yy/overlay/lib/python3.9/site-packages/setuptools/__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-j9ot46yy/overlay/lib/python3.9/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "setup.py", line 74, in run
      raise RuntimeError("CMake must be installed to build the following extensions: " +
  RuntimeError: CMake must be installed to build the following extensions: depthai
  ----------------------------------------
  ERROR: Failed building wheel for depthai
Failed to build depthai
ERROR: Could not build wheels for depthai which use PEP 517 and cannot be installed directly
DarkCloud:~ sebastianvillalobos$ 
`

Luxonis-Brandon, 2021-01-08:
I saw that one once as well.  Can you close the terminal, re-open, and run `python3 -m pip install -U pip` ?  And then try `python3 -m pip install depthai` again?

Luxonis-Brandon, 2021-01-08:
Oh, and sorry.  cmake wasn't installed somehow.

Can you do `brew install cmake`?

alex-luxonis, 2021-01-08:
Hi @esbasti,

This doesn't seem good:
```
Collecting depthai
Using cached depthai-0.4.0.0.tar.gz (84 kB)
```
It tries to build the wheel from sources (`.tar.gz`), while a prebuilt wheel should have been used instead. It might be because the command was ran initially natively (not under Rosetta 2).

Could you open a terminal under Rosetta and confirm that the command `uname -m` returns `x86_64`. Then try:
```
python3 -m pip uninstall depthai -y
python3 -m pip install depthai --no-cache-dir
```

I'm getting this output on Apple M1 with python 3.9.1.5:
```Collecting depthai
  Downloading depthai-0.4.0.0-cp39-cp39-macosx_10_9_x86_64.whl (2.9 MB)
     |████████████████████████████████| 2.9 MB 2.8 MB/s 
Installing collected packages: depthai
Successfully installed depthai-0.4.0.0
```

esbasti, 2021-01-08:
Hi @alex-luxonis thanks for looking into this as well

I can confirm you that after running the command `uname` -m returns `x86_64`

Notheless, when i run `python3 -m pip uninstall depthai -y` got the following warning

WARNING: Skipping depthai as it is not installed.

And when run `python3 -m pip install depthai --no-cache-dir`

Here is the error log 

Collecting depthai
  Downloading depthai-0.4.0.0.tar.gz (84 kB)
     |████████████████████████████████| 84 kB 1.1 MB/s 
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmp1j88zm3u
       cwd: /private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-z9ezgcns/depthai_80f01c8fd1bd4f0e9e1397038bb4394c
  Complete output (58 lines):
  running bdist_wheel
  running build
  running build_ext
  Traceback (most recent call last):
    File "setup.py", line 72, in run
      out = subprocess.check_output(['cmake', '--version'])
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 420, in check_output
      return run(*popenargs, stdout=PIPE, timeout=timeout, check=True,
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 501, in run
      with Popen(*popenargs, **kwargs) as process:
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 947, in __init__
      self._execute_child(args, executable, preexec_fn, close_fds,
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 1819, in _execute_child
      raise child_exception_type(errno_num, err_msg, err_filename)
  FileNotFoundError: [Errno 2] No such file or directory: 'cmake'
  
  During handling of the above exception, another exception occurred:
  
  Traceback (most recent call last):
    File "/opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py", line 280, in <module>
      main()
    File "/opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py", line 263, in main
      json_out['return_val'] = hook(**hook_input['kwargs'])
    File "/opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py", line 204, in build_wheel
      return _build_backend().build_wheel(wheel_directory, config_settings,
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-v7n7tw5c/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 216, in build_wheel
      return self._build_with_temp_dir(['bdist_wheel'], '.whl',
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-v7n7tw5c/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 202, in _build_with_temp_dir
      self.run_setup()
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-v7n7tw5c/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 253, in run_setup
      super(_BuildMetaLegacyBackend,
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-v7n7tw5c/overlay/lib/python3.9/site-packages/setuptools/build_meta.py", line 145, in run_setup
      exec(compile(code, __file__, 'exec'), locals())
    File "setup.py", line 159, in <module>
      setup(
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-v7n7tw5c/overlay/lib/python3.9/site-packages/setuptools/__init__.py", line 153, in setup
      return distutils.core.setup(**attrs)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/core.py", line 148, in setup
      dist.run_commands()
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 966, in run_commands
      self.run_command(cmd)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-v7n7tw5c/overlay/lib/python3.9/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
[... cut ...] sions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "setup.py", line 74, in run
      raise RuntimeError("CMake must be installed to build the following extensions: " +
  RuntimeError: CMake must be installed to build the following extensions: depthai
  ----------------------------------------
  ERROR: Failed building wheel for depthai
Failed to build depthai
ERROR: Could not build wheels for depthai which use PEP 517 and cannot be installed directly

alex-luxonis, 2021-01-09:
@esbasti Thanks for trying out the commands.

So after I installed Homebrew and then the other packages (python3, etc) under Rosetta, later on even when running in a native `arm64` terminal, I still got the prebuilt x86_64 depthai wheel getting downloaded and installed by `pip`.
I also upgraded to the latest Big Sur (11.1), but all was still fine.

I'm reinstalling macOS now, and trying to see if I could get to reproduce your situation, where the build from sources is always attempted. This will take some time...

Just in case, if you haven't already, could you try closing and reopening terminal consoles and maybe also rebooting. Sometimes changes done by installed programs (altering PATH or other environment variables) don't take effect otherwise.

If still no improvement, could you try the command that Brandon suggested `brew install cmake` and see if the build from sources gets to work.

alex-luxonis, 2021-01-09:
@esbasti So I erased and reinstalled Big Sur on my Mac M1, then did these:

1. Opened `Terminal` and executed this dependencies install script from our website:
https://docs.luxonis.com/en/latest/pages/api/#installing-system-dependencies
`sudo curl -fL http://docs.luxonis.com/_static/install_dependencies.sh | bash`
Then ran `python3 -m pip install depthai` and was greeted with the same situation you encountered: it downloaded the `.tar.gz` and attempted the wheel build, failing.

2. **Opened another `Terminal` under Rosetta 2:** 
    - right-click -> `Get Info` -> ticked `Open using Rosetta`. Then trying to open it prompted me to install Rosetta, and after that I was able to open it.
    - tried again `python3 -m pip install depthai` and was prompted with the same situation and error.
    - **ran again the dependencies install: `sudo curl -fL http://docs.luxonis.com/_static/install_dependencies.sh | bash`**
    - tried again `python3 -m pip install depthai` , still with no luck. `brew install cmake` didn't helped.
    - **closed the terminal**
 
3. **Opened again `Terminal` under Rosetta 2:**
    - **ran `python3 -m pip install depthai`** and at this point it worked as expected, downloading the `_x86_64.whl`

Next I cloned this repo, ran `python3 install_requirements.py`, then `python3 depthai_demo.py`, all working well.

I highlighted in **bold** the steps from above that I believe helped. Let us know how it goes on your side.

esbasti, 2021-01-09:
Still not luck.

Tried the steps above

And after running the command `python3 -m pip install depthai` I'm still getting the error below

Also thought it may had to do something with how I installed Brew and Rosetta and try this:

https://stackoverflow.com/questions/64963370/error-cannot-install-in-homebrew-on-arm-processor-in-intel-default-prefix-usr

But it still fails

---------------------------------------------------------------------------------------------------

Collecting depthai
  Using cached depthai-0.4.0.0.tar.gz (84 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/tmpdhhq0zr1
       cwd: /private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-install-3_8q08vt/depthai_77ef299d5c9e4ed49dbfa03c1413b776
  Complete output (143 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is AppleClang 12.0.0.12000032
  -- The CXX compiler identification is AppleClang 12.0.0.12000032
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- [hunter] Calculating Toolchain-SHA1
  -- [hunter] Calculating Config-SHA1
  -- [hunter] HUNTER_ROOT: /Users/sebastianvillalobos/.hunter
  -- [hunter] [ Hunter-ID: 062a19a | Toolchain-ID: 5e27f7a | Config-ID: 2a6efb3 ]
  -- [hunter] NLOHMANN_JSON_ROOT: /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Install (ver.: 3.8.0)
  -- Found nlohmann_json: /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.8.0")
  -- [hunter] NLOHMANN_JSON_SCHEMA_VALIDATOR_ROOT: /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Install (ver.: 2.1.1)
  -- [hunter] XLINK_ROOT: /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Install (ver.: luxonis-2020.2)
  -- [hunter] Building XLink
  loading initial cache file /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/cache.cmake
  loading initial cache file /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Build/XLink/args.cmake
  CMake Error: Could not open file for write in copy operation /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Build/XLink/Build/XLink-Release-prefix/src/XLink-Release-stamp/XLink-Release-urlinfo.txt.tmp
  CMake Error: : System Error: Permission denied
  CMake Error at /opt/homebrew/Cellar/cmake/3.19.2/share/cmake/Modules/ExternalProject.cmake:2661 (configure_file):
    configure_file Problem configuring file
  Call Stack (most recent call first):
    /opt/homebrew/Cellar/cmake/3.19.2/share/cmake/Modules/ExternalProject.cmake:3430 (_ep_add_download_command)
    CMakeLists.txt:152 (ExternalProject_Add)
  
  
  CMake Error: Could not open file for write in copy operation /Users/sebastianvillalobos/.hunter/_Base/062a19a/5e27f7a/2a6efb3/Build/XLink/Build/XLink-Release-prefix/src/XLink-Release-stamp/download-XLink-Release.cmake.tmp
  CMake Error: : System Error: Permission denied
  CMake Error at /opt/homebrew/Cellar/cmake/3.19.2/share/cmake/Modules/ExternalProject.cmake:1448 (configure_file):
    configure_file Problem configuring file
  Call Stack (most recent call first):
    /opt/homebrew/Cellar/cmake/3.19.2/share/cmake/Modules/ExternalProject.cmake:2717 (_ep_write_downloadfile_script)
    /opt/homebrew/Cellar/cmake/3.19.2/share/cmake/Modules/ExternalProject.cmake:3430 (_ep_add_download_command)
    CMakeLists.txt:152 (ExternalProject_Add)
  
  
[... 77 lines cut ...]
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/private/var/folders/45/h1sx4z4164v9hqc36ryjvvfm0000gn/T/pip-build-env-nno8ee5o/overlay/lib/python3.9/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/dist.py", line 985, in run_command
      cmd_obj.run()
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
    File "/opt/homebrew/Cellar/python@3.9/3.9.1_5/Frameworks/Python.framework/Versions/3.9/lib/python3.9/distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
[... cut ...] 9hqc36ryjvvfm0000gn/T/pip-install-3_8q08vt/depthai_77ef299d5c9e4ed49dbfa03c1413b776/build/lib.macosx-11-arm64-3.9/', '-DPYTHON_EXECUTABLE=/opt/homebrew/Cellar/python@3.9/3.9.1_5/bin/python3.9', '-DCMAKE_BUILD_TYPE=Release', '-DHUNTER_CONFIGURATION_TYPES=Release']' returned non-zero exit status 1.
  ----------------------------------------
  ERROR: Failed building wheel for depthai
Failed to build depthai
ERROR: Could not build wheels for depthai which use PEP 517 and cannot be installed directly

alex-luxonis, 2021-01-10:
@esbasti I was able to build the `depthai` wheel from sources under native arm64, after adding the Homebrew install to PATH manually: `export PATH=/opt/homebrew/bin:$PATH`. I also had to `brew install cmake libusb`.

I also see from your logs above that the build was progressing, but reaching some permission issues in `~/.hunter` directory. I believe this happens because the command was ran at some point with sudo (`sudo python3 -m pip install depthai`). Maybe you could resolve the problem by deleting this directory entirely: `sudo rm -rf ~/.hunter` (should be safe to do that, Hunter is a source-code-level package manager we use for builds).

But anyway, this isn't the right direction for fixing the problem (at least for now), as some python libraries are still not supported under native Apple M1. Most prominent is `numpy`, that we depend on for depthai. So for now we need to stay with the x86_64 translation, and thus should get the prebuilt to work (and don't even attempt build from sources):
`python3 -m pip install --only-binary=:all: --no-cache-dir depthai`

To check directly under which platform `python3` runs: `python3 -c "import platform; print(platform.platform())"` :
- `macOS-11.1-arm64-i386-64bit` - bad, native
- `macOS-11.1-x86_64-i386-64bit` - OK, with emulation

In your case it must be something with the Homebrew install. Could you check the PATH env var `echo $PATH` , and if by any chance /opt/homebrew is included, remove it.

And if the problem persists, maybe even remove completely Homebrew, running in a native terminal:
1. `/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/uninstall.sh)"`
    --should remove it from `/opt/homebrew`
2. `arch -x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/uninstall.sh)"` 
    --should remove it from `/usr/local`

And then reinstall with this command under Rosetta:
`sudo curl -fL http://docs.luxonis.com/_static/install_dependencies.sh | bash`
followed by closing and reopening the terminal.

esbasti, 2021-01-12:
Finally I was able to install all dependencies successfully and run the demo.

What ended up working was the following:

And if the problem persists, maybe even remove completely Homebrew, running in a native terminal:

`/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/uninstall.sh)"`
--should remove it from /opt/homebrew
`arch -x86_64 /bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/uninstall.sh)"`
--should remove it from /usr/local

And then reinstall with this command under Rosetta:
`sudo curl -fL http://docs.luxonis.com/_static/install_dependencies.sh | bash`
followed by closing and reopening the terminal.

In addition to running this commands I had to manually delete the Homebrew directory in both locations.

Thank you both for taking the time and troubleshoot for a solution @Luxonis-Brandon @alex-luxonis

Luxonis-Brandon, 2021-01-12:
Glad it's working and thanks for circling back!

alex-luxonis, 2021-01-13:
Thanks! Closing as resolved.

We also [updated](https://github.com/luxonis/depthai-docs-website/pull/109) the install script for macOS, it can be run now on M1 without having to run the terminal under Rosetta (it will automatically prefix certain commands with `arch -x86_64`):
https://docs.luxonis.com/en/latest/pages/api/#macos

closed 2021-01-13 by alex-luxonis

whabilly, 2021-01-13:
The DepthAI API and examples (gen2_develop branch) run fine natively on M1 by the way, once you get native OpenCV for Python installed (I built it from source weeks ago, but it should be possible to install it now using conda via conda-forge) as well as Numpy (also using conda-forge).

alex-luxonis, 2021-01-13:
We were thinking to recommend users to run under Rosetta, until opencv-python, numpy, etc. get to a stable state on M1 arm64, or at least could be built from sources with `pip`. As the performance under x86_64 translation seems pretty good anyway.
But we could look at `conda-forge`. Thanks!

whabilly, 2021-01-13:
I have been using M1-native only development tools since day 1 (early December) and so far Homebrew and Conda-forge have provided all the packages I needed, and in the worse cases building from sources has not been an issue at all.

Luxonis-Brandon, 2021-01-13:
Thanks, both.  And good to know @whabilly .  The M1 chips are so impressive.  Cool to see computer innovation like this.  Before the M1, it seemed computer processors weren't -really- getting any faster in terms of daily user experience.

ikester, 2022-01-20:
Wondering what the status of this is one year later. I just got my OAK-D lite today and wondering if I can install all native arm libraries and binaries or if I need to compile any dependencies.

Luxonis-Brandon, 2022-01-20:
Man how time flies!

Native M1 actually is largely dependent on external tools being converted to M1.  And I'm not sure if those are ported just yet.  Would have to check.  Checking...

That said, install on M1 Mac (M1, M1 Pro, M1 Max) is clean using Rosetta.  So no dependencies are required.  

Will circle back shortly.

ikester, 2022-01-20:
> Man how time flies!

Especially the last couple of years! 2020 and 2021 just blend together for me.

> 
> Native M1 actually is largely dependent on external tools being converted to M1. And I'm not sure if those are ported just yet. Would have to check. Checking...
> 
> That said, install on M1 Mac (M1, M1 Pro, M1 Max) is clean using Rosetta. So no dependencies are required.
> 

I would like to avoid having an Intel (Rosetta) Homebrew install on this machine, hence the question. Now that OAK-D Lite is out maybe more M1 Mac users will come together to find a more seamless solution.

> Will circle back shortly.

Thanks!

Luxonis-Brandon, 2022-01-22:
Thank you!  We can likely integrate this to provide pre-built libraries for Apple M* processors.  Asking team.

themarpe, 2022-01-22:
Hi @whab 
Thanks for the nice writeup!

The reason for not yet having M1 prebuilt wheels is mostly due to GitHub Actions not yet offering M1 macs a Runners. 
However, we might be able to crosscompile M1 wheels on x86_64 macs - if you happen to know an easy way to do so with CMake, we can give it a try.

One option is also custom runners... will circle back if/when we decide to pick any of these options

ikester, 2022-01-23:
Worked like a charm on a MBP M1 Max! Thank you for the detailed write up.

I did get a bunch of warnings at the end of `cmake --build . --parallel`, but everything seems to be working.

@Luxonis-Brandon , @themarpe I know that ideally you'd like to provide pre-built libraries for M1, but it would be great to have this process added to the docs In the meantime. Hopefully this issue will be discoverable as well.

Thanks @whab !

themarpe, 2022-01-23:
Thanks for giving this a try on your end as well @ikester 
Actually, I think the steps could be distilled down to:
```
# Install arm64 brew
/bin/bash -c "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install.sh)"
echo 'eval "$(/opt/homebrew/bin/brew shellenv)"' >> ~/.zprofile
eval "$(/opt/homebrew/bin/brew shellenv)"

# Install dependencies
brew install cmake libusb

# Build wheel from source
python3 -m pip install --no-binary=depthai depthai
```

@ikester would you be able to test this? If it works as expected, we'll add it to the docs.
(Also, the final built wheel has to be checked if its arm64 - I think this could by done by running `readelf -h depthai.dylib`, where `depthai.dylib` points to the location of the built and installed wheel (somewhere in `site`)).

ikester, 2022-01-23:
@themarpe that didn't work for me. I ran the `pip install` command outside a virtual env and got this.

```
~/d/p/Dep/depthai> python3 -m pip install --no-binary=depthai depthai                                                                 ✔
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Collecting depthai
  Using cached depthai-2.14.1.0.tar.gz (325 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (pyproject.toml) ... error
  ERROR: Command errored out with exit status 1:
   command: /opt/homebrew/opt/python@3.9/bin/python3.9 /opt/homebrew/lib/python3.9/site-packages/pip/_vendor/pep517/in_process/_in_process.py build_wheel /var/folders/pb/flklcl_11sl9w1dhmdqz8flr0000gn/T/tmpuvu22c09
       cwd: /private/var/folders/pb/flklcl_11sl9w1dhmdqz8flr0000gn/T/pip-install-xh9m2qju/depthai_1b61178e61be47268f285f613bff32fc
  Complete output (313 lines):
  running bdist_wheel
  running build
  running build_ext
  -- The C compiler identification is AppleClang 13.0.0.13000029
  -- The CXX compiler identification is AppleClang 13.0.0.13000029
  -- Detecting C compiler ABI info
  -- Detecting C compiler ABI info - done
  -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
  -- Detecting C compile features
  -- Detecting C compile features - done
  -- Detecting CXX compiler ABI info
  -- Detecting CXX compiler ABI info - done
  -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
  -- Detecting CXX compile features
  -- Detecting CXX compile features - done
  -- Found Git: /opt/homebrew/bin/git (found version "2.34.1")
  -- [hunter] Calculating Toolchain-SHA1
  -- [hunter] Calculating Config-SHA1
  -- [hunter] HUNTER_ROOT: /Users/ike/.hunter
  -- [hunter] [ Hunter-ID: 9b4e732 | Toolchain-ID: 7322064 | Config-ID: 7005309 ]
  -- [hunter] NLOHMANN_JSON_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 3.9.1)
  -- [hunter] XLINK_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: luxonis-2021.4-master)
  -- [hunter] BZIP2_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.0.8-p0)
  -- [hunter] FP16_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: luxonis-0.0.0)
  -- [hunter] LIBARCHIVE-LUXONIS_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 3.4.2-p2)
  -- [hunter] SPDLOG_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.8.2)
  -- [hunter] ZLIB_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.2.11-p2)
  -- [hunter] BACKWARD_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.6)
  -- [hunter] LIBNOP_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.0-5a66ba9)
  -- Looking for pthread.h
  -- Looking for pthread.h - found
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
  -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
  -- Found Threads: TRUE
  -- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR)
  -- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR)
  -- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR)
  -- Found Backward: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install/lib/backward
  -- Found nlohmann_json: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.9.1")
  -- Performing Test FLAG_-Wall
  -- Performing Test FLAG_-Wall - Success
  -- Performing Test FLAG_-Wextra
  -- Performing Test FLAG_-Wextra - Success
  -- Performing Test FLAG_-Woverloaded-virtual
  -- Performing Test FLAG_-Woverloaded-virtual - Success
  -- Performing Test FLAG_-Wformat=2
  -- Performing Test FLAG_-Wformat=2 - Success
  -- Performing Test FLAG_-Wmisleading-indentation
  -- Performing Test FLAG_-Wmisleading-indentation - Success
  -- Performing Test FLAG_-Wduplicated-cond
  -- Performing Test FLAG_-Wduplicated-cond - Failed
  -- Performing Test FLAG_-Wduplicated-branches
  -- Performing Test FLAG_-Wduplicated-branches - Failed
  -- Performing Test FLAG_-Wnull-dereference
[... 238 lines cut ...]
      cmd_obj.run()
    File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/wheel/bdist_wheel.py", line 299, in run
      self.run_command('build')
    File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/setuptools/_distutils/cmd.py", line 313, in run_command
      self.distribution.run_command(command)
    File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
      cmd_obj.run()
    File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/setuptools/_distutils/command/build.py", line 135, in run
      self.run_command(cmd_name)
[... cut ...]
    File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
      raise CalledProcessError(retcode, cmd)
  subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
  ----------------------------------------
  ERROR: Failed building wheel for depthai
Failed to build depthai
ERROR: Could not build wheels for depthai, which is required to install pyproject.toml-based projects
```

ikester, 2022-01-23:
@whab, so the examples in the `examples` dir work fine, but I was not able to run the depthai_demo.py in the `https://github.com/luxonis/depthai.git` repo. When I try running `python3 install_requirements.py` I get this. Any suggestions for getting that working as well?

```
> python3 install_requirements.py
pip 21.3.1 from /Users/ike/dev/projects/DepthAI/myvenv/lib/python3.9/site-packages/pip (python 3.9)
Requirement already satisfied: pip in /Users/ike/dev/projects/DepthAI/myvenv/lib/python3.9/site-packages (21.3.1)
Found existing installation: opencv-python 4.5.5.62
Uninstalling opencv-python-4.5.5.62:
  Successfully uninstalled opencv-python-4.5.5.62
WARNING: Skipping opencv-contrib-python as it is not installed.
Found existing installation: depthai 2.14.1.0.dev0+df0e2fc71653264dc2c3da6b466228a4b00ece9d
Uninstalling depthai-2.14.1.0.dev0+df0e2fc71653264dc2c3da6b466228a4b00ece9d:
  Successfully uninstalled depthai-2.14.1.0.dev0+df0e2fc71653264dc2c3da6b466228a4b00ece9d
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
Ignoring opencv-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version == "3.10"' don't match your environment
Ignoring opencv-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Ignoring opencv-contrib-python: markers 'platform_machine == "armv6l" or platform_machine == "armv7l"' don't match your environment
Obtaining file:///Users/ike/dev/projects/DepthAI/depthai/depthai_sdk (from -r requirements.txt (line 11))
  Preparing metadata (setup.py) ... done
Collecting requests==2.26.0
  Downloading https://www.piwheels.org/simple/requests/requests-2.26.0-py2.py3-none-any.whl (62 kB)
     |████████████████████████████████| 62 kB 388 kB/s
Collecting pyrebase4==4.5
  Downloading https://www.piwheels.org/simple/pyrebase4/Pyrebase4-4.5.0-py3-none-any.whl (8.9 kB)
Collecting argcomplete==1.12.1
  Downloading https://www.piwheels.org/simple/argcomplete/argcomplete-1.12.1-py2.py3-none-any.whl (38 kB)
Collecting opencv-python==4.5.1.48
  Downloading opencv-python-4.5.1.48.tar.gz (88.3 MB)
     |████████████████████████████████| 88.3 MB 52.6 MB/s
  Installing build dependencies ... error
  ERROR: Command errored out with exit status 1:
   command: /Users/ike/dev/projects/DepthAI/myvenv/bin/python3 /private/var/folders/pb/flklcl_11sl9w1dhmdqz8flr0000gn/T/pip-standalone-pip-tna32jk0/__env_pip__.zip/pip install --ignore-installed --no-user --prefix /private/var/folders/pb/flklcl_11sl9w1dhmdqz8flr0000gn/T/pip-build-env-a42qx42n/overlay --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple --extra-index-url https://www.piwheels.org/simple --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/ --extra-index-url https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/ --prefer-binary -- setuptools wheel scikit-build cmake pip 'numpy==1.13.3; python_version=='"'"'3.6'"'"' and platform_machine != '"'"'aarch64'"'"'' 'numpy==1.19.3; python_version>='"'"'3.6'"'"' and sys_platform == '"'"'linux'"'"' and platform_machine == '"'"'aarch64'"'"'' 'numpy==1.14.5; python_version=='"'"'3.7'"'"' and platform_machine != '"'"'aarch64'"'"'' 'numpy==1.17.3; python_version=='"'"'3.8'"'"' and platform_machine != '"'"'aarch64'"'"'' 'numpy==1.19.3; python_version>='"'"'3.9'"'"' and platform_machine != '"'"'aarch64'"'"''
       cwd: None
  Complete output (533 lines):
  Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple, https://artifacts.luxonis.com/artifactory/luxonis-depthai-data-local/wheels/, https://artifacts.luxonis.com/artifactory/luxonis-python-snapshot-local/
  Ignoring numpy: markers 'python_version == "3.6" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.6" and sys_platform == "linux" and platform_machine == "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8" and platform_machine != "aarch64"' don't match your environment
  Collecting setuptools
    Downloading https://www.piwheels.org/simple/setuptools/setuptools-60.5.0-py3-none-any.whl (958 kB)
  Collecting wheel
    Downloading https://www.piwheels.org/simple/wheel/wheel-0.37.1-py2.py3-none-any.whl (35 kB)
  Collecting scikit-build
    Downloading https://www.piwheels.org/simple/scikit-build/scikit_build-0.12.0-py2.py3-none-any.whl (73 kB)
  Collecting cmake
    Downloading cmake-3.22.1-py2.py3-none-macosx_10_9_universal2.macosx_10_9_x86_64.macosx_11_0_arm64.macosx_11_0_universal2.whl (74.6 MB)
  Collecting pip
    Downloading https://www.piwheels.org/simple/pip/pip-21.3.1-py3-none-any.whl (1.7 MB)
  Collecting numpy==1.19.3
    Downloading numpy-1.19.3.zip (7.3 MB)
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
    Getting requirements to build wheel: started
[... cut ...]
  File "/Users/ike/dev/projects/DepthAI/depthai/install_requirements.py", line 47, in <module>
    subprocess.check_call(pip_package_install + ["-r", "requirements.txt"], cwd=scriptDirectory)
  File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['/Users/ike/dev/projects/DepthAI/myvenv/bin/python3', '-m', 'pip', 'install', '-U', '--prefer-binary', '-r', 'requirements.txt']' returned non-zero exit status 1.
```

themarpe, 2022-01-23:
@ikester 
To address both of the issues:

1. Try running the build for wheel from sources again, but specify the following env var beforehand (we'll fix that in our sdist, to enable it automatically) (will also edit the above comment):
`MACOSX_DEPLOYMENT_TARGET=11.0 python3 -m pip install --no-binary-wheels=depthai depthai`

2. Seems like the specified `numpy` version is too old (transitively specified by `opencv-python`) and doesn't have wheels for arm64 macOS yet. Modify `requirements.txt` the lines with `opencv-python==4.5.1.48` to `opencv-python==4.5.5.62` 

Also, can you run the following snippet on your M1 machine and post back the results:
```
python3 -c "import sys; import platform; from distutils import util; print(f'{sys.platform}, {platform.architecture()}, {platform.processor()}, {platform.machine()}, {util.get_platform()}');"
```

whab, 2022-01-24:
Here are additional step-by-step instructions for running the GUI demo from depthai-3.0.9 (downloaded from https://github.com/luxonis/depthai/releases) on M1 Macs. They need to be applied after the previously provided instructions and in the same virtual environment DepthAIEnv39:

cd depthai-3.0.9

# Edit requirements.txt and keep only the lines specified below (note that opencv-contrib-python==4.5.5.62 was replaced with opencv-contrib-python==4.5.5.62):
nano requirements.txt
	pyrebase4==4.5
	argcomplete==1.12.1
	--extra-index-url https://www.piwheels.org/simple
	opencv-contrib-python==4.5.5.62 ; platform_machine != "aarch64" and platform_machine != "armv6l" and platform_machine != "armv7l" and python_version != "3.10"
	-e ./depthai_sdk

# Edit install_requirements.py:
nano install_requirements.py
    Remove code of block (3 lines) starting with: if thisPlatform == "arm64" and platform.system() == "Darwin":
    Remove the 2 lines starting with: subprocess.check_call(pip_call + ["uninstall"

# Edit depthai_sdk/requirements.txt and keep only the lines specified below:
nano depthai_sdk/requirements.txt
	blobconverter>=1.2.8
	pytube>=11.0.1
	PyTurboJPEG==1.6.4

# Install dependencies
conda install -c anaconda pyqt
python3 install_requirements.py

# Run the GUI demo (the UVC mode, aka Webcam, under the Misc tab also works!)
python3 depthai_demo.py 

# To check the architecture of the compiled module (arm64 should be reported for native M1 code):
lipo -info /opt/homebrew/Caskroom/miniconda/base/envs/DepthAIEnv39/lib/python3.9/site-packages/depthai.cpython-39-darwin.so
	Non-fat file: /opt/homebrew/Caskroom/miniconda/base/envs/DepthAIEnv39/lib/python3.9/site-packages/depthai.cpython-39-darwin.so is architecture: arm64

Tested on a new MacBookPro M1 Pro running macOS Monterey 12.1 with a OAK-D-Lite. Enjoy!

ikester, 2022-01-30:
> @ikester To address both of the issues:
>
> 1. Try running the build for wheel from sources again, but specify the following env var beforehand (we'll fix that in our sdist, to enable it automatically) [...]

After adjusting the command so it can run (replaced `--no-binary-wheels` with `--no-binary`) I still get a not found error:

```
MACOSX_DEPLOYMENT_TARGET=12.0 python3 -m pip install --no-binary=depthai depthai                                                   ✔
DEPRECATION: Configuring installation scheme with distutils config files is deprecated and will no longer work in the near future. If you are using a Homebrew or Linuxbrew Python, please see discussion at https://github.com/Homebrew/homebrew-core/issues/76621
Collecting depthai
  Using cached depthai-2.14.1.0.tar.gz (325 kB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for depthai (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [291 lines of output]
      running bdist_wheel
      running build
      running build_ext
      -- The C compiler identification is AppleClang 13.0.0.13000029
      -- The CXX compiler identification is AppleClang 13.0.0.13000029
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Found Git: /opt/homebrew/bin/git (found version "2.35.1")
      -- [hunter] Calculating Toolchain-SHA1
      -- [hunter] Calculating Config-SHA1
      -- [hunter] HUNTER_ROOT: /Users/ike/.hunter
      -- [hunter] [ Hunter-ID: 9b4e732 | Toolchain-ID: 7322064 | Config-ID: 7005309 ]
      -- [hunter] NLOHMANN_JSON_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 3.9.1)
      -- [hunter] XLINK_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: luxonis-2021.4-master)
      -- [hunter] BZIP2_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.0.8-p0)
      -- [hunter] FP16_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: luxonis-0.0.0)
      -- [hunter] LIBARCHIVE-LUXONIS_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 3.4.2-p2)
      -- [hunter] SPDLOG_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.8.2)
      -- [hunter] ZLIB_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.2.11-p2)
      -- [hunter] BACKWARD_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.6)
      -- [hunter] LIBNOP_ROOT: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install (ver.: 1.0-5a66ba9)
      -- Looking for pthread.h
      -- Looking for pthread.h - found
      -- Performing Test CMAKE_HAVE_LIBC_PTHREAD
      -- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
      -- Found Threads: TRUE
      -- Could NOT find libdw (missing: LIBDW_LIBRARY LIBDW_INCLUDE_DIR)
      -- Could NOT find libbfd (missing: LIBBFD_LIBRARY LIBBFD_INCLUDE_DIR)
      -- Could NOT find libdwarf (missing: LIBDWARF_LIBRARY LIBDWARF_INCLUDE_DIR LIBELF_LIBRARY LIBELF_INCLUDE_DIR)
      -- Found Backward: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install/lib/backward
      -- Found nlohmann_json: /Users/ike/.hunter/_Base/9b4e732/7322064/7005309/Install/lib/cmake/nlohmann_json/nlohmann_jsonConfig.cmake (found version "3.9.1")
      -- Performing Test FLAG_-Wall
      -- Performing Test FLAG_-Wall - Success
      -- Performing Test FLAG_-Wextra
      -- Performing Test FLAG_-Wextra - Success
      -- Performing Test FLAG_-Woverloaded-virtual
      -- Performing Test FLAG_-Woverloaded-virtual - Success
      -- Performing Test FLAG_-Wformat=2
      -- Performing Test FLAG_-Wformat=2 - Success
      -- Performing Test FLAG_-Wmisleading-indentation
      -- Performing Test FLAG_-Wmisleading-indentation - Success
[... 194 lines cut ...]
        File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/site-packages/setuptools/_distutils/dist.py", line 986, in run_command
          cmd_obj.run()
        File "setup.py", line 86, in run
          self.build_extension(ext)
        File "setup.py", line 174, in build_extension
          subprocess.check_call(['cmake', '--build', '.'] + build_args, cwd=self.build_temp)
        File "/opt/homebrew/Cellar/python@3.9/3.9.10/Frameworks/Python.framework/Versions/3.9/lib/python3.9/subprocess.py", line 373, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for depthai
Failed to build depthai
ERROR: Could not build wheels for depthai, which is required to install pyproject.toml-based projects
```

> Also, can you run the following snippet on your M1 machine and post back the results:
> 
> ```
> python3 -c "import sys; import platform; from distutils import util; print(f'{sys.platform}, {platform.architecture()}, {platform.processor()}, {platform.machine()}, {util.get_platform()}');"
> ```

I get:
`darwin, ('64bit', ''), arm, arm64, macosx-12-arm64`

ikester, 2022-01-30:
> Here are additional step-by-step instructions for running the GUI demo from depthai-3.0.9 (downloaded from https://github.com/luxonis/depthai/releases) on M1 Macs. They need to be applied after the previously provided instructions and in the same virtual environment DepthAIEnv39:
...
> Tested on a new MacBookPro M1 Pro running macOS Monterey 12.1 with a OAK-D-Lite. Enjoy!

Thanks @whab. This worked for me, using my self-built wheel. Not ideal given all the steps and how error-prone this is. Really looking forward to a streamlined M1 experience.

themarpe, 2022-07-08:
Just for reference: https://github.com/luxonis/depthai-python/pull/604

Checking out that branch and running: `python3 examples/install_requirements.py` should already install the m1 wheels.

We are in progress of also making a release with m1 wheels included, but are still resolving some intermitten issues with the builders. 

Thanks for all the discussion on pre-official support for m1!

presstoexit, 2023-04-12:
Thanks for all the people putting effort into this.

Trying @whab's recipe from https://github.com/luxonis/depthai/issues/299#issuecomment-1019158365, it was all going very well until I got to the `python3 -m pip wheel . -w wheelhouse` command (apologies for very long paste, not sure what to crop):

```
▶ python3 -m pip wheel . -w wheelhouse
Processing /Users/pr2e/DepthAI/depthai-python
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: depthai
  Building wheel for depthai (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for depthai (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [2619 lines of output]
      /private/var/folders/y7/gddpm8ys0892ztpvq78knn2m0000gq/T/pip-build-env-6iqnxxl_/overlay/lib/python3.9/site-packages/setuptools/dist.py:547: UserWarning: Normalizing '2.21.2.0.dev+bfe65785b69c99f5ad96762ee480a7f30fa46075' to '2.21.2.0.dev0+bfe65785b69c99f5ad96762ee480a7f30fa46075'
        warnings.warn(tmpl.format(**locals()))
      running bdist_wheel
      running build
      running build_ext
      -- The C compiler identification is AppleClang 13.1.6.13160021
      -- The CXX compiler identification is AppleClang 13.1.6.13160021
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Detecting CXX compiler ABI info
      -- Detecting CXX compiler ABI info - done
      -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      -- Generating new toolchain...
      -- Using toolchain file: /Users/pr2e/DepthAI/depthai-python/build/temp.macosx-11.0-arm64-cpython-39/depthai-core/generated/toolchain.cmake
      -- Found Git: /usr/local/bin/git (found version "2.40.0")
      -- [hunter] Calculating Toolchain-SHA1
      -- [hunter] Calculating Config-SHA1
      -- [hunter] HUNTER_ROOT: /Users/pr2e/.hunter
      -- [hunter] [ Hunter-ID: cb0ea1f | Toolchain-ID: ca965da | Config-ID: 7fdaa4a ]
      -- [hunter] NLOHMANN_JSON_ROOT: /Users/pr2e/.hunter/_Base/cb0ea1f/ca965da/7fdaa4a/Install (ver.: 3.9.1)
      -- [hunter] Building nlohmann_json
      loading initial cache file /Users/pr2e/.hunter/_Base/cb0ea1f/ca965da/7fdaa4a/cache.cmake
      loading initial cache file /Users/pr2e/.hunter/_Base/cb0ea1f/ca965da/7fdaa4a/Build/nlohmann_json/args.cmake
      -- The C compiler identification is AppleClang 13.1.6.13160021
      -- The CXX compiler identification is AppleClang 13.1.6.13160021
      -- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      -- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ - skipped
      -- Detecting CXX compile features
      -- Detecting CXX compile features - done
      CMake Warning (dev) at /private/var/folders/y7/gddpm8ys0892ztpvq78knn2m0000gq/T/pip-build-env-6iqnxxl_/overlay/lib/python3.9/site-packages/cmake/data/share/cmake-3.25/Modules/ExternalProject.cmake:3075 (message):
        The DOWNLOAD_EXTRACT_TIMESTAMP option was not given and policy CMP0135 is
        not set.  The policy's OLD behavior will be used.  When using a URL
        download, the timestamps of extracted files should preferably be that of
        the time of extraction, otherwise code that depends on the extracted
        contents might not be rebuilt if the URL changes.  The OLD behavior
        preserves the timestamps from the archive instead, but this is usually not
        what you want.  Update your project to the NEW behavior or specify the
        DOWNLOAD_EXTRACT_TIMESTAMP option with a value of true to avoid this
        robustness issue.
      Call Stack (most recent call first):
        /private/var/folders/y7/gddpm8ys0892ztpvq78knn2m0000gq/T/pip-build-env-6iqnxxl_/overlay/lib/python3.9/site-packages/cmake/data/share/cmake-3.25/Modules/ExternalProject.cmake:4185 (_ep_add_download_command)
        CMakeLists.txt:152 (ExternalProject_Add)
      This warning is for project developers.  Use -Wno-dev to suppress it.
      
      -- Configuring done
      -- Generating done
[... 2492 lines cut ...]
      [ 97%] Building CXX object CMakeFiles/depthai.dir/src/pipeline/datatype/SpatialLocationCalculatorDataBindings.cpp.o
      [ 97%] Building CXX object CMakeFiles/depthai.dir/src/pipeline/datatype/StereoDepthConfigBindings.cpp.o
      [ 98%] Building CXX object CMakeFiles/depthai.dir/src/pipeline/datatype/SystemInformationBindings.cpp.o
      [ 99%] Building CXX object CMakeFiles/depthai.dir/src/pipeline/datatype/TrackedFeaturesBindings.cpp.o
      [100%] Building CXX object CMakeFiles/depthai.dir/src/pipeline/datatype/TrackletsBindings.cpp.o
      [100%] Linking CXX shared module /Users/pr2e/DepthAI/depthai-python/build/lib.macosx-11.0-arm64-cpython-39/depthai.cpython-39-darwin.so
[... cut ...]
      FileNotFoundError: [Errno 2] No such file or directory: '/Users/pr2e/DepthAI/depthai-python/build/lib.macosx-11.0-arm64-cpython-39/depthai/py.typed'
      gmake[2]: *** [CMakeFiles/depthai.dir/build.make:988: /Users/pr2e/DepthAI/depthai-python/build/lib.macosx-11.0-arm64-cpython-39/depthai.cpython-39-darwin.so] Error 1
      gmake[2]: *** Deleting file '/Users/pr2e/DepthAI/depthai-python/build/lib.macosx-11.0-arm64-cpython-39/depthai.cpython-39-darwin.so'
      gmake[1]: *** [CMakeFiles/Makefile2:182: CMakeFiles/depthai.dir/all] Error 2
      gmake: *** [Makefile:1

themarpe, 2023-04-13:
@presstoexit 

Official M1 wheels are available since: https://github.com/luxonis/depthai/issues/299#issuecomment-1179345431

Please follow that installtion.

If you are still hitting issues - double check your installation of python3, to see if its arm64 arch or x86