luxonis/depthai #300: depthai_demo.py failed with xLink

JinFree, 2021-01-13:
H/W: Raspberry Pi 4 Model B Rev 1.2 4GB
OS: Linux raspberrypi 5.4.79-v7l+ #1373 SMP Mon Nov 23 13:27:40 GMT 2020 armv7l GNU/Linux

I bought OAK-1 and OAK-D.
At first, I want to use these in Nvidia Jetson board. But because of open3d, depthai_demo.py couldn't execute on any Jetson board.

And I get RPi 4 to use cameras, but depthai_demo.py fails again.

I fixed some problems with model_compiler.py.

At first,  I installed openvino-2020.04 and changed supported_openvino_version = '2020.01' to supported_openvino_version = 'openvino' so that makes openvino local compile.
Second, I changed myriad_compile_path, deployment_tools/inference_engine/lib/**intel**/myriad_compile to deployment_tools/inference_engine/lib/**armv7l**/myriad_compile
Third, I changed shaves=4 and cmx_slices=4 in myriad_compile_model_local function.
So I makes RPi can compile mobilenet-ssd.

But after that, I got json Runtime error and Device get data failed:7

Below are code and logs.

```
pi@raspberrypi:~/depthai $ python3 depthai_demo.py 
No calibration file. Using Calibration Defaults.
Using depthai module from:  /home/pi/.local/lib/python3.7/site-packages/depthai.cpython-37m-arm-linux-gnueabihf.so
Depthai version installed:  0.4.0.0
Depthai version required:   0.4.0.0
Compiled mode found: compiled for 14 shaves, 14 cmx_slices and 1 NN_engines 
XLink initialized.
Sending internal device firmware
Successfully connected to device.
Usb speed : Super/5000Mbps
Mx serial id : 14442C107138BFD200
Loading config file
Attempting to open stream config_d2h
watchdog started 
Successfully opened stream config_d2h with ID #0!
Closing stream config_d2h: ...
Closing stream config_d2h: DONE.
EEPROM data: valid (v5)
  Board name     : BW1098OBC
  Board rev      : R0M0E0
  HFOV L/R       : 71.86 deg
  HFOV RGB       : 68.7938 deg
  L-R   distance : 7.5 cm
  L-RGB distance : 3.75 cm
  L/R swapped    : yes
  L/R crop region: center
  Rectification Rotation R1 (left):
    0.999990,    0.002084,    0.003869,
   -0.002092,    0.999996,    0.002123,
   -0.003865,   -0.002131,    0.999990,
  Rectification Rotation R2 (right):
    0.999925,   -0.006061,   -0.010594,
    0.006038,    0.999979,   -0.002159,
    0.010607,    0.002095,    0.999942,
  Calibration intrinsic matrix M1 (left):
  861.622253,    0.000000,  636.557739,
    0.000000,  861.885376,  408.870361,
    0.000000,    0.000000,    1.000000,
  Calibration intrinsic matrix M2 (right):
  862.803101,    0.000000,  641.687683,
    0.000000,  863.294922,  408.929321,
    0.000000,    0.000000,    1.000000,
  Calibration rotation matrix R:
    0.999862,    0.008099,    0.014488,
   -0.008161,    0.999958,    0.004194,
   -0.014454,   -0.004312,    0.999886,
  Calibration translation matrix T:
   -7.482183,
    0.045350,
    0.079270,
  Calibration Distortion Coeff d1 (Left):
   -4.627060,   13.011962,    0.000697,    0.001788,  -11.515502,   -4.678207,   13.232531,
  -11.731359,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
  Calibration Distortion Coeff d2 (Right):
   -5.720683,   18.420849,   -0.000076,   -0.000203,  -20.248404,   -5.768956,   18.637218,
  -20.481133,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,    0.000000,
['metaout', 'previewout']
Available streams: ['meta_d2h', 'left', 'right', 'disparity', 'depth', 'metaout', 'previewout', 'jpegout', 'video', 'object_tracker', 'rectified_left', 'rectified_right', 'color']
/home/pi/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.json
depthai: Calibration file is not specified, will use default setting;
config_h2d json:
{"_board":{"calib_data":[0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0,0.0],"mesh_left":[0.0],"mesh_right":[0.0]},"_load_inBlob":true,"_pipeline":{"_streams":[{"name":"metaout"},{"name":"previewout"}]},"ai":{"NCEs":1,"NN_config":{"NN_family":"mobilenet","confidence_threshold":0.5,"output_format":"detection"},"blob0_size":14533056,"blob1_size":0,"calc_dist_to_bb":true,"camera_input":"rgb","cmx_slices":14,"keep_aspect_ratio":true,"num_stages":1,"shaves":14},"app":{"enable_reconfig":true,"sync_sequence_numbers":false,"sync_video_meta_streams":false,"usb_chunk_KiB":64},"board":{"clear-eeprom":false,"left_fov_deg":71.86000061035156,"left_to_rgb_distance_m":0.019999999552965164,"left_to_right_distance_m":0.09000000357627869,"name":"","override-eeprom":false,"revision":"","rgb_fov_deg":68.7938003540039,"stereo_center_crop":false,"store-to-eeprom":false,"swap-left-and-right-cameras":true},"camera":{"mono":{"fps":30.0,"resolution_h":720,"resolution_w":1280},"rgb":{"fps":30.0,"resolution_h":1080,"resolution_w":1920}},"depth":{"depth_limit_mm":10000,"lr_check":false,"median_kernel_size":7,"padding_factor":0.30000001192092896,"warp_rectify":{"edge_fill_color":0,"mirror_frame":true,"use_mesh":false}},"ot":{"confidence_threshold":0.5,"max_tracklets":20}}
size of input string json_config_obj to config_h2d is ->Attempting to open stream config_h2d
Successfully opened stream config_h2d with ID #1!
Writing 1048576 bytes to config_h2d
!!! XLink write successful: config_h2d (1048576)
Closing stream config_h2d: ...
Closing stream config_h2d: DONE.
Creating observer stream host_capture: ...
Attempting to open stream host_capture
Successfully opened stream host_capture with ID #0!
Creating observer stream host_capture: DONE.
Attempting to open stream inBlob
Successfully opened stream inBlob with ID #1!
Writing 14533056 bytes to inBlob
!!! XLink write successful: inBlob (14533056)
Closing stream inBlob: ...
Closing stream inBlob: DONE.
depthai: done sending Blob file /home/pi/depthai/resources/nn/mobilenet-ssd/mobilenet-ssd.blob.sh14cmx14NCE1
Attempting to open stream outBlob
Successfully opened stream outBlob with ID #2!
E: [global] [    571145] [python3] addEvent:264	Condition failed: event->header.flags.bitField.ack != 1
E: [global] [    571145] [python3] addEventWithPerf:276	 addEvent(event) method call failed with an error: 3
E: [global] [    571145] [python3] XLinkReadData:156	Condition failed: (addEventWithPerf(&event, &opTime))
Device get data failed: 7
Closing stream outBlob: ...
E: [global] [    571146] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_CLOSE_STREAM_REQ

E: [xLink] [    571146] [Scheduler00Thr] sendEvents:1027	Event sending failed
Closing stream outBlob: DONE.
Traceback (most recent call last):
  File "depthai_demo.py", line 404, in <module>
    dai.startLoop()
  File "depthai_demo.py", line 82, in startLoop
    p = self.device.create_pipeline(config=config)
RuntimeError: [json.exception.parse_error.101] parse error at line 1, column 1: syntax error while parsing value - unexpected end of input; expected '[', '{', or a literal
=== New data in observer stream host_capture, size: 4
Writing 4 bytes to host_capture
E: [global] [    571210] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_WRITE_REQ

E: [xLink] [    571210] [Scheduler00Thr] sendEvents:1027	Event sending failed
!!! XLink write successful: host_capture (4)
Stopping threads: ...
Stopping threads: DONE 0.000s.
Closing all observer streams: ...
Closing stream host_capture: ...
Closing stream host_capture: DONE.
Closing all observer streams: DONE.
Reseting device: 0.
E: [global] [    571210] [Scheduler00Thr] dispatcherEventSend:53	Write failed (header) (err -4) | event XLINK_RESET_REQ

E: [xLink] [    571210] [Scheduler00Thr] sendEvents:1027	Event sending failed
E: [xLink] [    571210] [python3] DispatcherWaitEventComplete:367	Assertion Failed: curr != NULL 

E: [global] [    571210] [python3] XLinkResetRemote:263	Condition failed: DispatcherWaitEventComplete(&link->deviceHandle)
Reseting: DONE.
```
How can I fix these error?

JinFree, 2021-01-13:
I fixed it.
Install with openvino 2020.1.023 from 
`https://download.01.org/opencv/2020/openvinotoolkit/2020.1/`

closed 2021-01-13 by JinFree

Luxonis-Brandon, 2021-01-13:
(Thanks for circling back with the resolution!)